Ysurac/openmptcprouter #1260: More than one default VPN is selected

jpossetti, 2020-09-29:
Can't user another VPN than Glorytuh

When I can enable MLVPN and disable Glorytun the server trhows an error: More than one default VPN is selected

## Steps to Reproduce the Problem

  1.Disable Glorytun
  2.Enable MLVPN
  3.Apply
  4. Roboot (or not)

Iam trying to choice the best VPN for point to point video streaming over 4g networks. Cant test MLVPN

## Specifications

  - OpenMPTCProuter version: v0.55.3 r0+14198-da98603597
  - OpenMPTCProuter VPS version: 0.1017 5.4.52-mptcp
  - OpenMPTCProuter platform: RPI3 x64

Ysurac, 2020-09-29:
VPN must be set using the wizard. MLVPN is not available in wizard VPN list because it's not working on current release (this will work in next release).

jpossetti, 2020-09-29:
Ok, If I disable Glorytun system says "No VPN running" or something like that but bonding is still working. What VPN do you recomment for video streaming traffic and fault tolerance and FEC?

Ysurac, 2020-09-29:
VPN is used by default for anything that is not TCP. If no VPN are running the status is correct.
If streaming is done over TCP then no problem as it's only use shadowsock, else if it's over UDP than you may have to test each and check what is the best for your case.

jpossetti, 2020-09-29:
Thanks a lot! Best regards

closed 2021-01-04 by github-actions[bot]