Ysurac/openmptcprouter #1261: Restore does not restore the prot forwarding rules

Merrick28, 2020-09-30:
## Expected Behavior
I hoped the backup restoration from the VPS would restore my specific port forwarding rules.

## Current Behavior
My SD Card burnt yesterday, so I installed a new one : flash with BalenaEtcher, boot on the rpi, then add the two Wan and the VPS information. After that I tried to restore a backup from the server. There has been no issue (at least nothing I've seen), but the spécific port forwarding rules were not restored, event after a reboot.

## Steps to Reproduce the Problem

  1. Add specific forwarding rules in the GUI
  2. Prepare a new sd card and boot on it
  3. Add WAN and VPS info
  4. Restore a backup from the server
  5. See what's missing....

Current configuration example : 
[image]

## Specifications

  - OpenMPTCProuter version: v0.55.3 r0+14198-da98603597
  - OpenMPTCProuter VPS version: OpenMPTCProuter VPS 0.1017
  - OpenMPTCProuter platform: RPI3B

Ysurac, 2020-09-30:
Port forwarding rules are not visible on router side or not enabled on VPS side ?

Merrick28, 2020-09-30:
There were not visible on the routeur gui. I created them by hand after testing the restoration.
I don't know if they where enables on the vps side, and as I created them again, I guess it's toot late to check ?

 (we can continue in french if you prefer)

jrvanator, 2020-10-01:
> There were not visible on the routeur gui. I created them by hand after testing the restoration.
> I don't know if they where enables on the vps side, and as I created them again, I guess it's toot late to check ?
> 
> (we can continue in french if you prefer)

If you created them on the command line using iptables, they only existed in memory.  iptables itself does not commit anything to disk.  If you use the GUI, it updates both iptables and the /etc/config/firewall file so the changes persist past reboots.

Ysurac, 2020-10-01:
There is a bug in restore backup from VPS. This will be fixed in next release.

Merrick28, 2020-10-01:
> If you created them on the command line using iptables, they only existed in memory. iptables itself does not commit anything to disk. If you use the GUI, it updates both iptables and the /etc/config/firewall file so the changes persist past reboots.

I created them using the GUI

Thanks @Ysurac for your answer, no rush for me, I've only 4 rules that I can create quickly in case of another crash.

closed 2021-01-06 by github-actions[bot]