Ysurac/openmptcprouter #1262: Comment activer Netflix?

JeedHome44, 2020-09-30:
Bonjour,

Avant de brancher le Pi, je voudrais savoir comment dire à Netflix de ne pas passer par le VPS.
De passer par ma 4G en priorité ou mon ADSL si la 4G tombe.
On regarde principalement Netflix sur notre TV du salon en ethernet sur le réseau mais aussi sur smartphone, tablette.

J'ai vu des issues là dessus mais pas compris.
Possible de me dire comment faire?

Je ne trouve pas le menu pour configurer ca sur le Pi

Merci

kevin39, 2020-09-30:
Salut,
Services -> OMR-Bypass

JeedHome44, 2020-09-30:
Merci mais après? Quoi faire?
As tu un tuto a me donner pour mettre Netflix sur 4G et basculer sur ADSL si 4G hs?

philoun8, 2020-11-01:
Bonsoir

Même en activant netflix dans Services -> OMR-Bypass, certaines fois des films ou series ne fonctionnent pas en affichant le message  "Il semble que vous utilisiez un débloqueur ou un proxy".

Ysurac, 2020-11-01:
Il faut être certain de ne pas utiliser un autre serveur DNS que l'IP d'OpenMPTCProuter. Ca expliquerait pourquoi ça marche de temps en temps.

philoun8, 2020-11-01:
tu veux dire qu'il ne faut rien entrer dans la section serveur dns des pc ou, le cas present pour mon netflix sur un mibox3. 
je mets toujours 8.8.8.8 et 8.8.4.4, les dns de google

Ysurac, 2020-11-01:
Il faut mettre l'IP d'OpenMPTCProuter en tant que DNS, si c'est pas en DHCP, et pas ceux de google.

philoun8, 2020-11-01:
si si, tout est en dhcp. mais je rajoutais juste les dns de google.

philoun8, 2020-11-01:
je signal quand même, car c'est important, que je suis tres content d'OpenMPTCProuteur. tout est beaucoup plus stable a la maison.

closed 2021-02-05 by github-actions[bot]