Ysurac/openmptcprouter #1263: Comment bloquer certains IP?

JeedHome44, 2020-09-30:
Bonjour,

Désolé c'est encore moi.
Je ne trouve pas le menu permettant d'enregistrer des adresse MAC d'appareils qu'on veut bloquer.

J'ai actuellement 2 Pi qui sont d'un côté sur un LAN en ethernet et de l'autre en wifi sur un autre LAN.
En utilisant OMR, je vais les mettre en ethernet sur un seul réseau.

Je veux donc bloquer leur adresse mac respective de leur module wifi le temps que je refasse bien mon wifi chez moi pour éviter qu'ils se connectent en wifi et en ethernet en même temps.

Comment faire?

Merci

JeedHome44, 2020-10-04:
Quelqu'un peut me dire ou se trouve le menu pour bloquer des adresses MAC du réseau? J'ai beau fouiller partout sur OMR, je ne trouve pas.
Merci d'avance.

Adorfer, 2020-10-05:
i would suggest a workaround: define n-th WAN interface, which is NOT connected. 
then make a OMR-bypass for that mac, via the not-connected WAN. (not tested myself, please feedback if it helps)

Ysurac, 2020-10-05:
Same as https://github.com/Ysurac/openmptcprouter/wiki/Block-client-Internet-access but by using mac in advanced settings tab instead of source IP.

JeedHome44, 2020-10-05:
Je suis dans les règles de pare feu et on ne peut choisir que l'IP, pas le MAC.
Comment faire?

Ysurac, 2020-10-05:
Comme je l'ai écrit, dans l'onglet paramètres avancés il est possible de choisir une adresse MAC.

JeedHome44, 2020-10-05:
Super merci!
C'est bien comme ca?

https://www.casimages.com/i/201005095351984815.jpg.html

Ysurac, 2020-10-05:
oui

JeedHome44, 2020-10-05:
Est-ce normal par contre de le voir toujours dans les appareils connectés?
Il faut peut être reboot OMR?
Je ne veux plus qu'OMR lui attribué d'ip.
Merci

closed 2021-01-09 by github-actions[bot]