Ysurac/openmptcprouter #1264: Help!

JeedHome44, 2020-09-30:
Bonjour,

Je me suis lancé.
Après avoir configuré tous les IP fixe en ethernet sur le Pi avec mon PC portable,
J'ai désactivé le DHCP de ma Freebox, celui de mon routeur TP link sur le LAN 4G (mon routeur Huawei a déjà son DHCP de désactivé).
Je fais un reboot de tout ce petit monde avec le Pi OMR branché aussi sur un switch de la Freebox.
Et la rien.
Plus d'accès au réseau 4G. Aucun accès au Pi ORM avec 192.168.100.1

Je suis perdu. Je n'ai plus d'internet chez moi la!

J'ai besoin d'aide!

Mes points d'accès wifi sont tous sur rouge!

Ysurac, 2020-09-30:
Chaque router a une IP fixe configurée ? Ces IPs sont configurés dans OMR ?
Pour OMR il faut voir si il est correctement branché, il intègre un serveur DHCP qui doit donner une adresse IP dans le réseau 192.168.100.x. Si ce n'est pas le cas il est toujours possible de configurer une IP fixe sur le poste client dans ce même réseau.

JeedHome44, 2020-09-30:
Alors, pour résumer:

Jusqu'à aujourd'hui, j'avais un routeur Huawei 4G dans le grenier relié à une CPL gigabit dans le garage.

2 autres CPL ressortent dans la maison.
- Une cote salon avec un switch gigabit ou la Nintendo est branché, la tv...
- Une autre dans le bureau.
Un câble ethernet sort de la CPL pour aller à un routeur TP Link qui me servait pour le DHCP car le routeur Huawei est limité à 8 IP fixe. Sur ce CPL j'ai aussi 1 des ports ethernet de mon NAS.

Ça c'est pour le réseau LAN 4G.

À côté de ça, j'ai une Freebox Mini 4k qui me sert de DHCP car on n'est pas limité au nombre d'appareil et dessus, j'ai un autre routeur TP Link qui lui me sert de point d'accès.
Dessus, j'ai branché l'autre port de mon NAS, Jeedom, Home Assistant...

Donc pour résumer, j'avais d'un côté la 4G et de l'autre l'ADSL.

Ces derniers jours, j'ai attribué l'ip 192.168.0.1 à la Freebox, l'ip 192.168.8.1 au routeur 4G Huawei. Le routeur TP Link sur la 4G servant de DHCP avait l'ip 192.168.1.1.
J'avais donc un réseau en 0.X pour l'ADSL et 1.X pour la 4G.

Ce matin, j'ai donc juste désactivé le DHCP du routeur TP Link blanc 192.168.1.1
J'ai également désactivé le DHCP de la Freebox 192.168.0.1
J'ai branché le Pi OMR sur un port du switch 8 ports branché à la Freebox.

J'ai vu que ça ne marchait pas.

Donc la je viens carrément de débrancher les 2 routeurs TP Link, celui de l'ADSL qui me servait de point d'accès et celui de la 4G qui me servait de routeur.

J'ai branché la CPL du bureau directement sur le switch de la Freebox pour relier les 2 réseaux.

Donc la, il n'y a plus que la Freebox, le routeur 4G du grenier relié par CPL et le Pi OMR.

Mes nouveaux points d'accès Mesh TP link sont tous au rouge aussi alors que le principal est branché à la Freebox et j'ai internet avec mon Pc en ethernet sur la Freebox.

Donc je ne pige pas pourquoi tout est au rouge et pourquoi en tapant 192.168.100.1 sur le Pc ça n'affiche rien (si je branché le Pi en direct sur un Pc portable cet IP marche.
Je peux accéder à ma Freebox avec l'ip 192.168.0.1 mais pas à mon Huawei avec 192.168.8.1.

JeedHome44, 2020-09-30:
Je précise que j'ai fixé les IP de toutes les adresses Mac des appareils de la maison sur OMR via mon Pc portable avant de le brancher sur le réseau.
J'ai attribué des noms personnalisés à chaque MAC dans le DHCP.
Est-ce ça qui bloque tout?
Et j'ai juste activé le by-pass pour Netflix.

Ysurac, 2020-09-30:
Sur le poste client l'IP attribuée est bien dans le réseau 192.168.100.x ? Car ça semble étonnant de pouvoir accéder à la Freebox si c'est le cas.

JeedHome44, 2020-09-30:
Sur le Pc?
Bah quand je vais dans les paramètres réseau, je vois un ethernet actif et un inactif alors que mon PC n'a qu'un port ethernet.
Je ne vois pas où voir l'IP attribué la dessus.
Normalement c'est dans propriété ethernet et IPv4 mais rien ne s'affiche.
Il y a conflit avec la Freebox?

Ysurac, 2020-09-30:
Oui sur le PC. Je suppose que c'est un windows, je ne connais pas windows. Pour voir les IPs il me semble que c'est ```ipconfig /all``` dans une console.
Si tu peux accéder à la Freebox c'est qu'une IP est attribuée ou fixée à ton PC, vu qu'elle n'est pas sur le réseau du RPI c'est certainement pas la bonne IP.

JeedHome44, 2020-09-30:
Ok je vais remettre et désactiver à nouveau le DHCP de la Freebox.
Depuis la dernière mise à jour de Free, elle plante souvent quand je vais dans les paramètres.
Elle n'a peut être pas désactivé complètement le DHCP

JeedHome44, 2020-09-30:
Ah mais en allant dans les paramètres ethernet de mon Pc, je vois que je suis en 192.168.68.X
Que je mets l'ip 192.168.68.1 sur le navigateur, ça m'affiche mon routeur Mesh que j'ai mis en point d'accès. C'est donc lui qui rentre en conflit?
Mais il est en point d'accès. Ça ne devrait pas!

JeedHome44, 2020-09-30:
Et quand je vais sur les paramètres TP link du point d'accès Mesh, je vois des IP en 0.X. Avec DHCP Freebox désactivé.
Je pige plus rien...

Ysurac, 2020-09-30:
Le routeur Mesh a peut être un serveur DHCP qui entre effectivement en conflit.

JeedHome44, 2020-09-30:
Pour la Freebox, il n'y a pas besoin de la mettre en bridge?
Juste désactiver le DHCP?

Ysurac, 2020-09-30:
Oui juste DHCP et pas en bridge.

JeedHome44, 2020-09-30:
Dans réseau local / DHCP, ma Freebox ne veut pas décocher "Assignation fixe par machine"
Par contre serveur DHCP est bien décoché.
Je n'ai accès à la Freebox qu'avec "mafreebox.freebox.fr" si je tape 192.168.0.1 ça ne marche pas.
Tous les appareils affichés dessus n'ont pas d'IP V4. Que des V6.
Et le Pi OMR n'est pas vu par la Freebox.
C'est normal?

Ysurac, 2020-09-30:
Il faudrait désactiver l'IPv6 sur la Freebox, mais pour le moment je pense que c'est pas encore le soucis ;)
Et oui le PI n'a pas à être vu par la Freebox, qui devrait pas voir grand chose. Le PI refuse par défaut les IPv6.

JeedHome44, 2020-09-30:
Mon PC est en Ethernet sur la Freebox.
Si elle ne voit pas le Pi, comment alors peut-on accéder au réseau 100.X du Pi?
Il y a quelque chose à ouvrir comme port sur le PC?

L'autre fois, quand j'ai réussi à mettre le Pi sur le réseau, ça m'a créé un second port ethernet dans mes configuration réseau.
Mais la, le PC affiche que c'est déconnecté.

Ysurac, 2020-09-30:
Tu dois avoir une IP donné par le RPI en 192.168.100.x, si l'IP est autre c'est que quelque chose d'autre donne une IP.

JeedHome44, 2020-09-30:
Alors, je débranche électriquement le Mesh principal.
Le Pc n'a plus d'ipv4 du tout.
Je branche le Mesh en point d'accès, ça me fait un réseau LAN en 68.X
Tout ça avec le Pi OMR branché en direct sur la Freebox.
La Freebox voit bien le routeur Huawei, mon pont Hue...
Mon PC à un IP en 68.X si Mesh actif et rien du tout si Mesh éteint.
Je tourne en rond sans comprendre!
La ça veut bien dire que le DHCP Free est désactivé.
Mais pourquoi aucun IP n'est attribué quand le Mesh est éteint?

Ysurac, 2020-09-30:
Tu peux attribuer une IP fixe à ton PC en 192.168.100.x pour voir si tu arrives à joindre le RPI

JeedHome44, 2020-09-30:
J'ai attribué une IP fixe pour tous les appareils sur OMR en le branchant sur mon portable en direct.
La je me rends compte que le Mesh crée une sorte de réseau LAN de secours vu qu'il ne trouve pas de serveur DHCP sur le réseau car il est bien activé en point d'accès.
Donc, le problème, on dirait que le Pi ne crée pas le serveur DHCP sur le réseau et mon PC hier avait créé un autre port ethernet 3 hier avec TAP-Windows Adapter v9.
Avec ça j'avais accès au réseau 100.X
La il est marqué déconnecté donc le serveur DHCP est comme hors ligne...

JeedHome44, 2020-09-30:
Autre test.
Si je désactive les 2 cartes réseau que je vois sur mon pc.
Si je débranche le Mesh, le pc se met sur le réseau Freebox. Sans IPV4 mais avec IPV6.
Si je branche le Mesh, il bascule sur un réseau LAN du Mesh.
Ça ne serait pas l'ipv6 de la Freebox Lr soucis?

Ysurac, 2020-09-30:
Que la Freebox donne une adresse IPv6 est assez "normal", et ça n'influence pas la partie IPv4 et donc le fait que le RPI doit être joignable. Mais comme je l'ai dis, mais une IP fixe à ton PC et tu verras bien.

JeedHome44, 2020-09-30:
On fait ça comment pour mettre une IP fixe au Pc?

Nicolas BENOIST
06 50 15 74 79

Ysurac, 2020-09-30:
Sous windows aucune idée. Mais il doit y avoir de la documentation sur internet ou auprès de Microsoft.

JeedHome44, 2020-09-30:
Alors on avance lol!
J'ai attribué l'ip fixe du pc par les configurations Windows.
J'ai donc accès maintenant à l'interface OMR

JeedHome44, 2020-09-30:
Bon ça me met ça dans l'état.![Uploading 94685967...]

JeedHome44, 2020-09-30:
Ça me met ça dans état.
J'ai vérifié et l'ip VPS est bonne et j'ai bien mis la clé fournie après l'installation du VPS.

https://www.casimages.com/i/200930024353869779.jpg.html

JeedHome44, 2020-09-30:
Dans le serveur DHCP, il n'y a aucun appareil dans Baux DHCP actifs.
Même pas le PC fixe!

Ysurac, 2020-09-30:
les interfaces sont mal configurées. L'ip semble la même que la passerelle.

JeedHome44, 2020-09-30:
Voici les 2 photos des interfaces.
J'ai juste suivi le site Mo dédié.fr pour mettre ça.

https://www.casimages.com/i/200930024848189204.jpg.html

Ysurac, 2020-09-30:
La configuration est mauvaise. Incrémente de 1 l'adresse IP de chaque interface.

JeedHome44, 2020-09-30:
Ah c'est parce que j'ai mis l'ip de la passerelle pour chaque interface?

JeedHome44, 2020-09-30:
Maintenant ça donne ça.

https://www.casimages.com/i/20093002552014734.jpg.html

Ysurac, 2020-09-30:
oui. Les interfaces ne doivent pas avoir la même IP que la passerelle, sinon il y a conflit.

Ysurac, 2020-09-30:
C'est mieux. Par contre il y a un soucis d'heure soit sur le VPS soit sur le routeur. Ils doivent être à l'heure.

JeedHome44, 2020-09-30:
Par contre, j'ai toujours aucun bail actif en DHCP

JeedHome44, 2020-09-30:
Donc si je comprends bien le schéma la. En haut ça me dit que l'ip de mon Pc n'est pas lié à ce routeur.
Et c'est quoi le output dessous en rouge?

Ysurac, 2020-09-30:
Avant de voir le DHCP, il faudrait déjà que la connexion fonctionne...
Le No output c'est que rien n'arrive à sortir, donc qu'il y a un soucis qui est comme il est écrit et comme je l'ai dit surement un soucis d'heure.

JeedHome44, 2020-09-30:
J'ai reboot le VPS et OMR

JeedHome44, 2020-09-30:
Comment ça d'heure?
J'ai pris un VPS en France avec OVH.
Comment régler le soucis?

Vous voulez mon accès au VPS pour voir?

JeedHome44, 2020-09-30:
J'ai regardé dans Système / Système et le serveur OMR se croyait en août!
J'ai donc actualisé l'heure avec le navigateur mais ça n'a pas l'air d'avoir changé quelque chose.
J'arrive bien à aller sur le VPS avec Putty

Ysurac, 2020-09-30:
Ca doit virer le soucis d'heure indiqué dans la page status.

JeedHome44, 2020-09-30:
J'ai ça maintenant

https://www.casimages.com/i/200930033324401583.jpg.html

JeedHome44, 2020-09-30:
Comment puis je vous faire un don pour tout ce que vous faites pour moi?

JeedHome44, 2020-09-30:
Je pense avoir trouvé pourquoi aucun appareil n'a accès au DHCP.
En bloquant chaque IP, j'ai mis un nom personnalisé pour chaque IP bloqué dans le nom d'hôte.
Si le nom d'hôte ne correspond pas au nom de l'appareil, le système ne va donc pas lui attribuer son IP fixe.
C'est ça?

Je demande ça car je vois aussi un menu Nom d'hôtes

Ysurac, 2020-09-30:
Pour les dons : https://www.paypal.me/ycarus (ou via autre chose sur https://www.openmptcprouter.com/ )
Sinon vu ce que tu as shadowsocks n'arrive pas à monter. Tu peux voir dans Etat->Journal système si il y a un soucis.

Pour le DHCP normalement ça n'a pas de rapport. Mais il est préférable d'avoir la connexion qui fonctionne avant de regarder le DHCP.

JeedHome44, 2020-09-30:
Voilà ce que j'ai fair pour bloquer chaque IP.

https://www.casimages.com/i/200930034209452677.jpg.html

JeedHome44, 2020-09-30:
Voilà son très mérité!

J'espère maintenant y arriver! 😉

Je pense ne pas être loin!

Ysurac, 2020-09-30:
Merci ;)
Essaye déjà de régler les soucis sur la page Status.
Et regarde dans Etat->Journal système les erreurs qu'il peut y avoir.

JeedHome44, 2020-09-30:
Alors je ne sais pas si c'est bon signe mais j'ai été sur degrouptest.com et ça affiche l'ip du VPS.
Donc ça veut dire que ça communique bien?

Ysurac, 2020-09-30:
Oui mais non. Ca passe par le VPN, pas par shadowsocks, donc c'est plus lent.

JeedHome44, 2020-09-30:
Ok je vois oui.
Le test débit change pas beaucoup de la 4G seule.
Vu que ça marche mais pas aussi rapide que voulu.
Est-ce possible de m'aider pour le DHCP?
Car j'aimerai rallumer mes box Domotique la pour les reconfigurer avec leur nouvelle IP.

JeedHome44, 2020-09-30:
Je vois que la méthode de shadow socks sur OMR est différente de ce que j'ai copier à la fin de l'installation du VPS.
C'est en crypté en chacha20 sur le VPs et c'est pas cette méthode de marque sur OMR.

JeedHome44, 2020-09-30:
Par contre sur OMR ça me demande un mot de passe de shadow socks Mais j'ai eu seulement le port, l'encryption et la key.
Je ne mets donc rien dans passwd sur OMR?

Ysurac, 2020-09-30:
Sur un RPI4 ça doit être chacha20.
Toute la configuration se fait via le wizard uniquement. Il y a une case paramètres avancés dans le wizard pour voir les clefs etc...

JeedHome44, 2020-09-30:
J'ai mis chacha20, une key du 34 caractère mais je mets donc rien dans mot de passe?

JeedHome44, 2020-09-30:
Et pour le DHCP?
Car la c'est étonnant...

Ysurac, 2020-09-30:
Il n'y a pas de mot de passe dans le Système->OpenMPTCProuter.
Pour DHCP il faut voir dans Réseau->Interfaces puis sur l'interface LAN l'onglet DHCP serveur.

JeedHome44, 2020-09-30:
J'ai changé shadow socks pour mettre sur chacha20 et ça remet chacha20-ietf-poly1305.
C'est normal?

Ok je vais voir pour le DHCP

Ysurac, 2020-09-30:
Oui, c'est normal.

JeedHome44, 2020-09-30:
Voilà dans interface

https://www.casimages.com/i/200930050157709446.jpg.html

[19 more replies]

closed 2021-01-04 by github-actions[bot]