Ysurac/openmptcprouter #1265: free+routeur 4g+VPS

ozonestrike, 2020-09-30:
Bonsoir,

J'ai pris un vps chez Ovh et j'ai agréger un routeur 4g et l'adsl de chez free !

J'ai bien tout configurer puisque tout est vert ;) Mais le problème, c'est que l'ip de ma freebox revient sans cesse pourtant le dhcp de la 4g et la freebox est désactivé :o auriez-vous une idée de pourquoi l'ip de la freebox revient constamment ? Pour avoir accès à l'ip du vps je dois forcer windows en rentrant l'ip manuellement ...

Merci d'avance.

CDT.

kevin39, 2020-09-30:
Bonsoir,
L'IP WAN ? Quand tu vas sur www.monip.org c'est l'IP de ta freebox ?

Ysurac, 2020-09-30:
Utilisez le template dans la mesure du possible car j'ai besoin notamment des versions, de l'architecture, etc...

Il s'agit de l'IPv4 ou IPv6 ? Et elle revient sans cesse où ? Qu'est-ce qui est forcé sur windows ?

ozonestrike, 2020-09-30:
Oui Kevin39 c'est l'ip wan :(

Ysurac c'est assez drôle parce que sous windows dans les détails de connexion c'est bien l'ip du vps qui est détecté mais sur les sites c'est l'ip de la freebox :o et pour pouvoir accéder enfin à la connexion du vps je dois désactiver la v6 et mettre dans la v4 l'ip du Raspberry 4 (OpenMPTCProuter) et là j'ai enfin l'ip du vps.

kevin39, 2020-09-30:
Je comprends pas comment Windows peut t'afficher ton IP du VPS, il est pas censé l'a connaître à moins que tu utilises un logiciel spécifique ? Du coup je sais pas.

Par contre, si tu as un soucis avec l'IPv6 c'est que tu as du laisser l'ipv6 de la freebox activé.
Il faut que tu désactives l'IPv6 dans la Freebox (en espérant que tu n'ais pas une Freebox Crystal qui n'a pas d'option pour le faire).

ozonestrike, 2020-09-30:
Non tout est désactivé mais c'est pas le pc le problème, puisque j'ai toujours accès à la tv de free (pop) câblé en rj45 ! J'ai pris la delta en freebox ;) Donc je comprends pas c'est quoi le problème en fait

kevin39, 2020-09-30:
Ton PC il est bien en DHCP ? Tu n'a pas mis d'IP fixe dans la configuration ?

ozonestrike, 2020-09-30:
Oui le pc est bien en dhcp ;)

ozonestrike, 2020-09-30:
Pour que ça fonctionne, il faut que je reboot le rasby et pendant un laps de temps j'aurais bien l'ip du vps mais peu de temps après il repart sur l'ip de la freebox !

kevin39, 2020-09-30:
Si tout est au vert dans OMR **et** si on parle bien de l'IPv4, c'est à priori que le DHCP de ta Freebox est encore actif.

ozonestrike, 2020-10-01:
oui mais pourtant le dhcp est désactivé de partout ...

kevin39, 2020-10-01:
Appuies sur la touche Windows+R, cela va ouvrir une fenêtre "Executer".
Tapes : **cmd** et valide

Dans la fenêtre noire qui s'ouvre, tapes : 
ipconfig /all

Montre moi le résultats stp

ozonestrike, 2020-10-01:
Non mais je viens de voir que j'avais bloqué l'ip du pc sur omrbypass sur la freebox -_- 

En fait je voulais débloquer mycanal,netflix,rmc sport avec le vps et je me suis planté ...

Bon bah j'ai bien l'ip du VPS ! Aurais-tu une solution pour débloquer CORRECTEMENT mycanal,rmc sport, netflix et aussi débloquer la tv freebox pop avec le VPS ?

CDT.

kevin39, 2020-10-01:
Dans omr-bypasss, tu peux faire des exceptions par services.

ozonestrike, 2020-10-01:
Oui mais cela ne fonctionne pas ;)

kevin39, 2020-10-01:
Tu es en quelle version ?

ozonestrike, 2020-10-01:
de omr-bypass ?

ozonestrike, 2020-10-01:
"Powered by openmptcprouter v0.55.3 r0+14198-da98603597"

kevin39, 2020-10-01:
Ok, et tu avais mis quelle conf dans OMR-bypass ?

ozonestrike, 2020-10-01:
Bah je voulais débloquer mycanal,netflix etc mais j'ai fais des erreurs et j'ai bloqué mon ip pc sur la freebox ;) j'attends quelques jours pour voir si c'était bien ça.

kevin39, 2020-10-01:
Tu as juste à choisir "Netflix" dans services et choisir le wan pas lequel le service doit sortir.

Par contre pour RMC & Canal j'ai aucune piste pour toi, ils sont pas dans la liste.
Eventuellement il faudrait chercher les IP pour faire des exceptions à la main mais ça risque d'être fastidieux.

ozonestrike, 2020-10-01:
sinon il n'existe pas un vps qui fonctionne pour tout les sites ?

kevin39, 2020-10-01:
Milkywan

ozonestrike, 2020-10-01:
https://www.jeuxvideo.com/ est bloqué aussi lui avec Milkywan ?

ozonestrike, 2020-10-02:
Bonjour,

Je viens de prendre un serv chez Milkywan effectivement ça fonctionne correctement ! Cependant, le débit est faible sur les speedtest une idée ? Sachant qu'eux disent qu'il n'y a pas de bridage.

CDT.

ozonestrike, 2020-10-03:
Un autre serveur sinon qui n'est pas blacklisté ? ;)

Ysurac, 2020-10-03:
A ma connaissance il n'y a qu'eux.
Pour voir pourquoi le débit est faible il faut regarder la charge CPU sur le VPS et le routeur. Et éventuellement prendre un VPS plus puissant si c'est qu'un bidouille.

ozonestrike, 2020-10-03:
Ysurac ;) j'ai pris plus puissant :o mais ça ne change que très peux ! Le gars m'a dit ". Tu devrais dans un premier temps désactiver le chiffrement" comment procéder ?

CDT.

Ysurac, 2020-10-03:
Dans le Système->OpenMPTCProuter, wizard, case "paramètres avancés"

ozonestrike, 2020-10-03:
ShadowSocks settings chiffrement ? il est sur chacha, faut mettre aucun ?

CDT.

Ysurac, 2020-10-03:
oui

ozonestrike, 2020-10-05:
Bon problème résolu, je suis passé sur un serveur avec davantage de CPU.

CDT.