Ysurac/openmptcprouter #1266: Débit download très bas

JeedHome44, 2020-09-30:
Bonjour,

Tout est correctement paramétré enfin.
Par contre, avec ma connexion 4G, j'avais l'habitude d'avoir des téléchargement à 7-8 Mbps en download.
Là je plafonne à 500 kbps.
Y a-t-il une configuration particulière pour avoir un download aussi bien voir meilleur que la 4G seule?

Merci

JeedHome44, 2020-10-02:
Peut on simplement dire à OMR de passer tout le flux sur une seule connexion et si elle coupe, passer par l'autre qui sert de secours?
J'ai l'impression que ça fou un bordel monstre OMR sur mon nouveau système Mesh pour le wifi.
Mes Google Home ne pige plus rien, n'ont plus de connexion...

Ysurac, 2020-10-02:
Il est possible que tu aies plutôt un soucis de DNS sur tes machins google. Il faut qu'ils utilisent le DNS d'OMR, car 8.8.8.8 ou 8.8.4.4 sont utilisés pour tester les connections et ça peut poser un soucis si ils sont utilisés par ailleurs.
Sinon il faut passer une autre connexion en mode backup pour qu'elle soit utilisée uniquement en cas de soucis.

JeedHome44, 2020-10-02:
Comment faire ça? Dans le DHCP ou je bloque chaque IP, je dois tous leur dire de passer par la passerelle 100.1?

Nicolas BENOIST
06 50 15 74 79

JeedHome44, 2020-10-02:
Mais la c'est même mon iPhone qui, voit le wifi des Mesh et quand je clique dessus, il me dit aucune connexion internet. J'ai l'impression qu'OMR n'est pas fait pour marcher avec un système Mesh.
Si?
Car sur le Pc en ethernet, la le test débit est top mais zéro avec les Mesh

Ysurac, 2020-10-02:
OMR est totalement indépendant d'un système Mesh ou autre chose. Le soucis est certainement au niveau de la configuration de ton système, il faut donc vérifier la configuration IP, passerelle et DNS sur ce dernier.

JeedHome44, 2020-10-02:
Ok merci.

Par contre, pour 1fichier, ça ne télécharge qu'à 500 Kbps sur le pc en ethernet. Avant avec la 4G seule ça montait à 7-8 Mbps.
Pourquoi une telle différence?
Quand je teste le débit avec nPerf pourtant je suis à plus de 100 Mbps en download. Avant j'étais à 80 Mbps 4G seule

Ysurac, 2020-10-02:
Il faudrait être plus précis pour pouvoir dire quelque chose. Mais il peut y avoir plein de facteurs qui peuvent être au niveau du VPS aussi.

JeedHome44, 2020-10-04:
> Il faudrait être plus précis pour pouvoir dire quelque chose. Mais il peut y avoir plein de facteurs qui peuvent être au niveau du VPS aussi.

Bonjour,
Aujour'dhui je télécharge autre chose sur internet et je me rends compte que je suis aussi limité à 500 kbps. Pourtant, avant en 4G j'étais bien plus haut.

Ce n'est donc pas forcément 1fichier qui est limité.
Dans quel menu peut on configuré la limite de téléchargement?

Merci

kevin39, 2020-10-04:
Bonjour,
Ton problème est ailleurs, si je ne me trompe pas tu sembles avoir un soucis de lenteur depuis le début.

Dans état tout est au vert ?

JeedHome44, 2020-10-04:
Salut! En fait, au début, j'avais mal configuré. Ce que je trouve bizarre, c'est qu'en étant sur mon Wifi et tester avec Nperf, j'ai des bons débits. Nperf ne testerai pas plutot le débit du VPS et pas ma connexion?

kevin39, 2020-10-04:
> Nperf ne testerai pas plutot le débit du VPS et pas ma connexion?

C'est techniquement impossible.

> J'ai passé ma connexion 4G en Maitre et ca change pas grand chose.

Non ça changera pas le débit. Théoriquement il faut mettre en "Master" la connexion la plus stable (normalement l'ADSL).

Par contre, tu n'as pas répondu.
Est-ce que tout est vert ? Aucune erreur dans l'état d'OMR ?

Aussi, pour tester réellement le débit de chaque WAN et de l'agrégation il faudrait procéder ainsi : 
1. Connecte toi en SSH au routeur (192.168.100.1 si tu n'as pas changé l'IP).
2. Tape "omr-test-speed". Laisse tourner 30 secs et montre nous le résultat.
3. Tape "omr-test-speed wan1". Laisse tourner 30 secs et montre nous le résultat.
4. Tape "omr-test-speed wan2". Laisse tourner 30 secs et montre nous le résultat.

JeedHome44, 2020-10-04:
Excuses moi. Oui tout est en vert sur OMR. Ou voit on les erreurs?
Je ne sais pas si c'est lié mais j'ai 2-3 appareils avec IP fixé qui ne prennent pas le bon IP. J'ai pourtant recopier plusieurs fois le MAC et attendu que l'ip se renouvelle.

root@OpenMPTCProuter:~# omr-test-speed
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  9 10.0G    9  976M    0     0  9988k      0  0:17:29  0:01:40  0:15:49 10.4M

root@OpenMPTCProuter:~# omr-test-speed wan1
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0 10.0G    0 22.3M    0     0   602k      0  4:49:52  0:00:37  4:49:15  626k

root@OpenMPTCProuter:~# omr-test-speed wan2
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  1 10.0G    1  107M    0     0  2531k      0  1:09:01  0:00:43  1:08:18 1937k

LE WAN 2 a l'air de tourner comme d'habitude. Les 2 additionné ca devrait quand même faire plus que 500 kbps.

kevin39, 2020-10-04:
Tu as d'autres WAN ?

JeedHome44, 2020-10-04:
Non que Freebox ADSL et Free mobile 4G illimité.
Est-ce qu'il y aurait une limitation dans le logiciel pour la 4G pour éviter les hors forfait?
Vu que je suis illimité je veux pas limiter la 4G

Ysurac, 2020-10-04:
Il n'y a aucune limitation de ce type.
Par contre je vois pas comment tu peux avoir 9.9Mo en agrégé avec une connexion à 0.6Mo et une à 2.5Mo. Ou alors la connexion 4G est extrêmement variable.
Après quand je demandais plus de précision, c'est par exemple le lien vers un fichier posant soucis.

JeedHome44, 2020-10-04:
Avant, en téléchargeant en 4G, ça montait rapidement jusqu'à environ 8Mbps en download pour n'importe quel fichier sur internet.
La j'ai téléchargé différents fichiers pour tester et toujours à 500 kbps.
Après, c'est sur le pc de bureau en ethernet que j'ai ce débit. Y a-t-il quelque chose à mettre sur l'ip fixe du pc pour permette d'avoir le débit maximum dessus?
Faut que je teste avec mon pc portable en wifi voir si j'ai le même débit.
Donc c'est étonnant d'avoir 9Mbps avec le test et 500kbps sur le PC

Ysurac, 2020-10-04:
Il faut vérifier que la passerelle sur le pc client soit bien l'IP d'OMR.

Ysurac, 2020-10-04:
Et que l'IPv6 n'est pas utilisé/dispo sur le pc client, car il peut utiliser celui donné par l'ADSL.

JeedHome44, 2020-10-04:
Oui c'est bien 192.168.100.1 la passerelle et je l'ai meme mise pour tester sur l'ip bloqué mais ça change rien.

JeedHome44, 2020-10-04:
Ah si l'ipv6 est utilisé sur le PC! C'est ça le soucis?

Ysurac, 2020-10-04:
Si l'IPv6 n'est pas fournis par OpenMPTCProuter et que c'est l'IP d'un fournisseur oui. Tout le trafic IPv6 passe par la.

kevin39, 2020-10-04:
Dans l'immédiat, pour écarter la thèse de l'IPv6 tu peux aller sur les propriétés de la carte réseau (du PC) et décocher "Internet Protocol Version 6"

[image]

JeedHome44, 2020-10-04:
Ah bah c'était bien ça!
Pourtant, l'ipv6 est désactivé sur la Freebox il me semble.
J'ai désactivé l'ipv6 sur le port ethernet du pc et la c'est parfait!
Donc, si je comprends bien, il faut aussi le désactiver sur tous les appareils de la maison?

kevin39, 2020-10-04:
Ca devrait fonctionner mais ce n'est pas la meilleure solution. L'idéal c'est d'avoir un ipv6 fonctionnel :)

JeedHome44, 2020-10-04:
Ok mais pourtant j'avais regardé sur la feeebox et il est marqué comme désactivé et la Huawei ne le propose pas. Je pige pas...
Mais c'est bon à savoir qu'il faut désactiver l'ipv6!
Maintenant il faut que je comprenne pourquoi 2-3 appareils ne prennent pas l'ip fixe attribuée alors que le MAC est le même que celui bloqué. Mon 80 autres se bloquent bien avec leur MAC.

kevin39, 2020-10-04:
C'est quel modèle de Freebox ?
La Crystal ne permet pas le blocage de l'ipv6.

kevin39, 2020-10-04:
Rectification, plus aucune Freebox ne semble proposer l'option : 
https://www.freenews.fr/freenews-edition-nationale-299/freebox-9/il-ne-sera-bientot-plus-possible-de-desactiver-ipv6-sur-freebox

JeedHome44, 2020-10-04:
Mini 4k!
Ah c'est pour ça on ne peut plus le bloquer?
C'est peut être pour ça aussi:
J'ai un projecteur SmartLife qui ne veut pas se mettre sur l'application SmartLife. Quand je l'avais eu, impossible de le mettre sur le LAN ADSL, ça n'allait pas assez vite pour communiquer avec le serveur SmartLife. Par contre, sur la 4G, aucun soucis.
Si un appareil essaie alors de se connecter en IPv6 sur le réseau, il va donc forcément tomber sur la Freebox et éviter le VPS c'est bien ça?
Comment alors obliger un IP particulier sur le réseau à ne pas passer par l'ipv6?

kevin39, 2020-10-04:
> Si un appareil essaie alors de se connecter en IPv6 sur le réseau, il va donc forcément tomber sur la Freebox et éviter le VPS c'est bien ça?

Oui.

> Comment alors obliger un IP particulier sur le réseau à ne pas passer par l'ipv6?

Tu ne peux pas, le fait qu'il soit en ipv6 bypass complètement OMR.
Je ne vois que 2 solutions : 
- Sois tu désactives l'ipv6 sur chaque périphérique (mais certains ne te permettront pas de faire ça)
- Sois tu passes ta Freebox en bridge mais tu vas te lancer dans une config réseau encore + avancée (voir Google)

Ysurac, 2020-10-04:
On peut bloquer l'IPv6 sur la majorité des Freebox en indiquant ```fe80::2``` dans Next Hop, ce qui devrait bloquer le RA IPv6.
Actuellement avec ta configuration, tout appareil recevant une adresse IPv6 distribuée via RA par la Freebox utilise donc la Freebox pour le trafic IPv6. Donc soit blocage niveau freebox indiqué ci dessus, soit blocage au niveau de chaque client dans la configuration, soit utilisation d'un routeur intermédiaire entre la freebox et le switch, soit encore utilisé un routeur supporté par OpenMPTCProuter avec un port dédié par WAN.

JeedHome44, 2020-10-04:
On ne peut pas utiliser l'ipv6 sur OMR et le mettre prioritaire sur la Freebox?

Ysurac, 2020-10-04:
On peut utiliser l'IPv6 sur OMR, mais la Freebox envoie ses messages avec une priorité haute, tout comme OMR, du coup la Freebox est toujours utilisée (car priorité haute et IP publique).

JeedHome44, 2020-10-04:
Et donc, si coupure ADSL, les appareils ne passeront même pas par la 4G ni le VPS?
En gros, l'ipv6 rend obsolète OMR?
C'est pour ça que je trouve le réseau bizarre depuis!
Et vu que je ne peux pas virer l'ipv6 des appareils SmartLife ou autre, c'est pour ça qu'ils sont lents! Ils ne passent que par la Freebox!
Le mode bridge vire l'ipv6?

JeedHome44, 2020-10-04:
Donc solution? Mode bridge?
J'ai jamais fait ça donc je ne sais pas comment le paramétrer.

Ysurac, 2020-10-04:
Encore une fois, si la configuration des Next Hop est possible que la Mini 4k, utilise ça.

L'IPv6 ne rend pas obsolète OMR, c'est lié à une connerie de chez Free c'est tout. Désactivable sur pas mal via Next Hop.

Le mode bridge est possible si l'on a une carte réseau dédiée.

JeedHome44, 2020-10-04:
Je ne vois pas ce qu'est Next Hop.
Y a-t-il un lien pour comprendre?

Ysurac, 2020-10-04:
(une recherche de 2s sur google, m'informe que la mini 4k utilise freebox OS et que la configuration des next hop est possible)

JeedHome44, 2020-10-04:
Ok merci, je vais me renseigner pour comprendre ce que c'est

Ysurac, 2020-10-04:
Cherche dans l'interface la configuration IPv6. Je n'ai pas de box avec Freebox OS, je ne peux pas faire de capture. Mais il y en a sur google image.

Ysurac, 2020-10-04:
D'ailleurs tant que tu y seras, si tu peux faire des capture avec le next hop à fe80::2, ça pourrait aller dans la doc ensuite :)

JeedHome44, 2020-10-04:
Si j'ai bien pigé, Next hop permet d'envoyer tous les paquets Free IPv6 vers OMR c'est ça?

Ysurac, 2020-10-04:
non. Ca dirige vers un lien local non existant, donc ça bloque.

JeedHome44, 2020-10-04:
Je tombe sur des tutos qui explique comment vraiment redirigé les paquets vers un routeur existant sur le LAN.
Je vois pas quoi mettre dans la ligne en fait pour que les paquets soient bloqués.
Je mets n'importe quoi?

Ysurac, 2020-10-04:
Comme déjà indiqué plusieurs fois, dans la ligne Next Hop, met : ```fe80::2```

JeedHome44, 2020-10-04:
Ah excuses moi je pensais que c'était juste un exemple

JeedHome44, 2020-10-04:
Ca me met cet avertissement:

Attention si vous configurez un Next Hop pour le premier subnet, il ne sera plus annoncé par la Freebox sur votre réseau. Cela empêchera les appareils dépendant d’IPv6 de fonctionner sans un routeur tiers qui gère ce subnet.

Je mets donc fe80::2 sur la seconde ligne ou la première? Je préfère être sur pour ne pas faire une connerie ^^

Ysurac, 2020-10-04:
Comme déjà indiqué, je n'ai pas de Freebox avec Freebox OS. Donc je ne sais pas ce que tu peux voir, tout ce que je peux te dire c'est quand il y a Next Hop il faut indiquer ```fe80::2```. Et le message d'avertissement correspond bien à ce que l'on souhaite.

JeedHome44, 2020-10-04:
ok je viens de voir une vidéo avec ce code ou la personne le met sur le 1 donc je fais ca de suite.
je te fais une capture

JeedHome44, 2020-10-04:
Voici la capture: https://www.casimages.com/i/201004101441501774.png.html

JeedHome44, 2020-10-04:
Merci!
C'était donc bien la Freebox!
J'ai mis sur la première ligne ce qu'il fallait et j'ai réactivé l'ipv6 sur le pc de bureau et ça télécharge vite maintenant.
Preuve que la Freebox est bloqué sur IPv6!
Vous êtes au top!

closed 2020-10-05 by Ysurac