Ysurac/openmptcprouter #1267: [question] openmptcprouter-mini packaging

PCRyan, 2020-10-01:
## Expected Behavior
@Ysurac (or anyone else) Could you explain a bit about the motivation behind the mini config added in July versus the regular config? Was it added purely as an option to package a lightweight OMR image? If anything else was involved, I'm curious to understand it.

## Context (Environment)
Inspired by [this comment](https://github.com/Ysurac/openmptcprouter/issues/29#issuecomment-487801652) by @jayanta525, I am working toward generating an OMR image for the TL-WR841N v13 (v13 chosen because it is the only version of the WR841N with 8MB flash instead of 4MB). I have accomplished now all the preliminaries of proving a normal OMR build and the latest OpenWRT for the device and successfully flashing it. I understand I will probably need to slim down the normal OMR build to make it productive on the device. That is the work I am now tackling.

Comments/hints welcome!

I'll be happy to add to this super project any successful outcome and then support it ongoing. Thus far it's been a welcome and overdue deeper education in all all things Linux and open-source. (My background has thus far been heavy in C++/Java/Javascript in the Microsoft and Eclipse IDEs.)

jayanta525, 2020-10-01:
I successfully compiled a build with mptcp enabled kernel for ramips target Netgear R6220. 

Image size was ~7MB without much package.

Disclaimer: The build isn't an OpenMPTCProuter build. Its an mptcp enabled kernel with openwrt rootfs along with shadowsocks.

Ysurac, 2020-10-01:
In most case devices with small flash have also a small CPU and doesn't give good result.
And yes openmptcprouter-mini was added for small devices, but not yet used on an officially supported device.

PCRyan, 2020-10-01:
Hmm. Interesting (and quick) responses, thanks!

@Ysurac The [WR841Nv13](https://openwrt.org/toh/tp-link/tl-wr841nd#hardware) runs at 575 MHz whereas the two supported Linksys WRT devices OMR supports run at 1866 MHz. I take it that's how the T-Link's CPU is small, as you say. Do you feel it's rather pointless to tweak OMR to have it run productively on the WR841Nv13? Or may it be the sort of target that your mini build could advance or inform? I don't care much about getting a huge bandwidth boost so much as having two cheap providers working together here in Austria and so yield seamless failover. (Of course a boost toward 60 Mbps would be a cool extra sweetener. :)

@jayanta525 Do you still see the WR841N as a reasonable target for OMR? Thanks for the link to your interesting build which it looks you accomplished before you posted the comment that got me going.

Clearly a ton must be cut to get the regular OMR image (41MB for the supported WRT32x) down to < 10MB. The default OpenWRT image for my WR841Nv13 (including LuCI) is 7.2MB.

jayanta525, 2020-10-02:
@PCRyan OpenWrt snapshots are getting bigger in size. A build for WR841Nv13 can easily be compiled with MPTCP kernel, but you won't be able to fit many packages. Also considering the processing capabilities of the 575MHz single-core CPU, I won't say WR841Nv13 to be a reasonable target for OMR.

There are various low-cost SBC with really good SoC, some also supporting crypto acceleration. My suggestion would be to go with an SBC with OMR.

If power consumption is your primary concern, check out RK3308 based armv8 ROCK Pi S. You can have OMR installed on it and use the WR841Nv13 as a managed switch and AP with OpenWrt.

PCRyan, 2020-10-02:
> There are various low-cost SBC with really good SoC, some also supporting crypto acceleration. My suggestion would be to go with an SBC with OMR. If power consumption is your primary concern, check [...]

Thanks for the commentary @jayanta525! It seems clear then that I need to give up the dream of OMR on 8MB flash and this CPU. :) Ah well, like I said, it's been a good education.

I will think over the single-board computer approach (SBC) approach as you suggested. Many of the WiFi devices these days seem like overkill for my purposes (non-gamers in apartment, limited streaming by a couple teenagers). I wouldn't say power consumption is my primary concern as much as just getting decent value from whatever I stand up. (I got the WR841N super cheaply, so I'm fine with liquidating it.) I do find appeal in having one box taking care of things, though, for sure. Do you (or anyone else) have a favored single device that OMR would work well on? I will be combining two LTE-based internet providers to start with (SIM chips).

Corrado56, 2020-10-02:
I doesn't understand the quest for a "mini" setup. 
I live in Italy in a place without a good fixed internet access. I use openMPTCP router to aggregate 3 LTE connections from 3 different providers (Vodafone, Wind, Iliad) and 3 towers.
With this configuration I reach 100Mbps during the day and more than 160 during the night.
The Ping go from native 25/30 ms to 40/50 after aggregation.
LTE connections are not really stables, but mptcp do his work, and I have the failover as an added bonus.

I work from home, and I am very happy with the current setup, BUT:

The ongoing costs are:
- 3 LTE subscription about 90 €/month
- VPS rental about 15 €/month
- Energy for: 3 LTE Routers, OpenMptcp router, a fifth apparate to distribute wifi, and some little switch's

Upfront costs:
- 3 LTE routers (2 Huawei B535 + 1Mikrotik Sxt-Lte6) about € 330 
- 2 Directional External antennas (I don't have a good signal) € 260 
- Linksys 3200 (openmptcp router) € 180
- Wifi router € 150 
- Other hardware ...

If after this investment, I reduce the consumption of only 1 element (openMptcp router) do I have a real change ?
For me the quest is for better performances, less overhead, better hardware, efficient VPS. 
Ongoing cost is not relevant.

Just an opinion.

ascomputer, 2020-10-19:
Hi Corrado I'm also in Italy, can I ask you where did you rent the vps? i can't find a good price ... thanks antonio

Corrado56, 2020-10-19:
Hi, I made some test with Aruba (cloud.it) and Hetzner.com. Both could be configured to initially  use Debian 10, then upgraded to use MPTCP.
Hetzner is Ok from CPU and memory sizing, but using germany servers, Ping never go under 40, usually between 40 and 50. As a side problem, receiving an Intra-EEC invoice involves additional administrative work.
With Aruba i initially buy the cheapest offer, ping is better 30/35, and the size is sufficient to run a VPS. but I found sometimes an initial response lag (i have no proof of that), so I upgraded to the "Large" size, that is the cheapest size with 2 cpu.
I've been using this sizing for three months now, without any problem. Note: I create the server in the IT3 server farm near Milan, IT1 and IT2 are near Arezzo.
Regards
Corrado
[image]

[image]

ascomputer, 2020-10-20:
Hi Corrado, thanks for the reply! aruba is certainly one of the best in Italy but the problem is that it gives you 12TB of traffic per month I honestly don't know if it's enough for a month but I would try it. I am currently using seflow.it dedicated server with unlimited traffic and bandwidth up to 1G / bits with 40% theoretically guaranteed about 400M / bits guaranteed at a cost of 27 euros + VAT, as you well know, vps are virtual machines that could be under stress and consequently do not perform much (I would like to try Aruba anyway). Do you also use Raspberry pi 4b?
Anyway thanks again for the reply.

Adorfer, 2020-10-20:
I do not understand how the selection of VPS relates the the "mini packaging" OMR build. 
(in other words: Please stay focussed: 1) do not derail threads 2) if discussing other topics either a) find an existing issue/thread or b) open a new one with a RELEVANT topic)

jayanta525, 2020-10-23:
@Corrado56 I guess, @PCRyan isn't looking at spending nearly € 900 as upfront costs just in hardwares.

> I doesn't understand the quest for a "mini" setup.

I agree with you at this. But, considering an user, with multiple ADSL connections with a fixed bandwidth, a minimal selection of packages is sufficient for its use-case. The core packages of openmptcprouter along with sqm, should suffice.

There are lots of sub $30 MT7621 devices with dual core 880Mhz CPUs supporting OpenWrt. If OMR-mini could be installed in these selection of devices, it would be a huge savings in upfront costs.

Ysurac, 2020-10-23:
Many MT7621 devices have only a 16Mo flash. OpenMPTCProuter with the web interface on 16Mo seems to be really difficult and I'm not sure this would be really usable.

jayanta525, 2020-10-23:
@Ysurac Sure, may i ask which device you have?

Ysurac, 2020-10-23:
It's a WE3526.
I need to find how to debrick it now :)

PCRyan, 2020-10-23:
Gentlemen, I appreciate very much your musings. I have spent these last weeks educating myself further on the technologies involved and my options, from higher-powered like the tantalizing UniElec U7623 and the solid Netgear R7800 on down to the challenging but promising Rock Pi and BananaPi R2. I'm still weighing everything, and your suggestion of a MT7621 middle ground points to a perhaps better-balanced middle ground (2-core 880 MHz). I see a good number of options there in the OpenWRT hardware table, many with substantially more than 16MB flash. Thanks. :)

Will keep you posted. It ought not to take much longer now to be ready to post a reasoned proposal here for feedback.

Adorfer, 2020-10-24:
if it's "value for money": AVM Fritzbox7362sl is EOL, but there are plenty of refurbished units on the market, since it had been part of a "standard package" for VDSL-provider "1&1" for years in germany -> prices are around 10-25€/unit including shipping. 

https://openwrt.org/toh/hwdata/avm/avm_fritz_box_7362sl
(dualcore, 128MB flash, 128MB RAM, and just ignore/not use the integrated g.17a-vectoring-modem, unless you like to dive into  lantiq-blob-license debates.)

closed 2021-01-28 by github-actions[bot]