Ysurac/openmptcprouter #1268: [question] is it possible to monitor each client bandwith ?

Merrick28, 2020-10-01:
## Context (Environment)
I've installed zabbix agent on my rpi, so I can monitor bandwith from every network card (to see bandwith on vdsl-Wan1 or 4G-Wan2)

I would like to know if I can monitor the bandiwth used by every connected client ? (for example I would like to know how bandwith is used by teenager's phone ?)

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.55.3 r0+14198-da98603597
  - OpenMPTCProuter VPS version: 0.1017 5.4.52-mptcp
  - OpenMPTCProuter platform: RPI3B

Ysurac, 2020-10-01:
maybe with luci-app-nlbwmon

Merrick28, 2020-10-01:
It's installed, but nobody's home so I can't fully test it, I will try this evening and let you know.

gwarf, 2020-10-01:
Interested to learn about this, would also like to be able to understand if video conf are lagging due to kids playing/updating games or if it's just something else :)

Merrick28, 2020-10-01:
It seems to work : 
[image]
I will try to find a way to get real time stats now.

Merrick28, 2020-10-01:
It seems that with luci-app-nlbwmon comes nlbw, so you can use in cli :
```
root@OpenMPTCProuter:~# nlbw -c show
  Fam            Host (    MAC )      Layer7      Conn.   > Downld. ( > Pkts. )      Upload (   Pkts. )
IPv4    192.168.1.216 (aa:76:b5)       other    44.38 K     6.13 GB (   4.98 M)   155.49 MB (   2.37 M)
IPv4  192.168.100.206 (3b:0a:1a)       HTTPS     3.69 K     5.41 GB (   1.59 M)   103.83 MB (   1.76 M)
IPv4    192.168.0.216 (8d:95:08)       other    67.07 K     5.19 GB (   5.22 M)   370.85 MB (   3.37 M)
IPv4  192.168.100.231 (c2:73:a1)       other       594      4.44 GB ( 951.92 K)    23.01 MB ( 427.70 K)
IPv4  192.168.100.192 (8f:11:21)       HTTPS     2.58 K   321.32 MB (  65.99 K)     3.66 MB (  47.26 K)
IPv4  192.168.100.209 (09:e2:e2)        HTTP       136    128.08 MB (  24.26 K)   703.30 KB (  13.41 K)
IPv4  192.168.100.156 (18:52:85)       HTTPS     5.89 K    79.25 MB (  54.00 K)    12.41 MB (  53.47 K)
IPv4  192.168.100.209 (09:e2:e2)       HTTPS    12.61 K    47.39 MB (  35.25 K)    12.58 MB (  44.08 K)
IPv4  192.168.100.187 (df:18:91)       HTTPS     3.02 K    22.80 MB (  13.08 K)     4.99 MB (  18.29 K)
IPv4  192.168.100.231 (c2:73:a1)        HTTP       288     16.85 MB (   3.46 K)   401.82 KB (   7.37 K)
IPv4  192.168.100.123 (21:f4:45)       other       174     11.46 MB (  41.67 K)     7.94 MB (  45.33 K)
IPv4    192.168.100.1 (d7:10:06)       other   118.16 K     8.13 MB ( 141.21 K)     6.96 MB ( 122.67 K)
IPv4     10.255.255.2 (00:00:00)         DNS    34.86 K     2.80 MB (  12.12 K)   996.31 KB (  12.21 K)
IPv4  192.168.100.123 (21:f4:45)       HTTPS     1.27 K     2.05 MB (   4.18 K)     1.36 MB (   4.53 K)
[...]
```
I will try to create a parse script that will allow me to use it in zabbix, I will let you know

Merrick28, 2020-10-02:
Finally, I used the command
```nlbw -c show -n -g host``` to get the data I wanted. I managed to create a discovery rule for zabbix to get the infos I wanted, but it's not very secure (I had to install sudo, and make zabbix user use sudo without password to make it work).
I create a dns entry for machines without dhcp.

** CAUTION : quick and dirty, use at you own risks ! **

/etc/steph/network_discovery.sh
```
!/bin/sh
IFS=$'\n'
echo {\"data\":[
for ligne in $(sudo /usr/sbin/nlbw -c show -n -g host|grep -v 10.255|grep -v 00:00:00|tail -n+2)
do
ip=$(echo $ligne|awk '{print $1}')
hostname=$(dig -x $ip +short)
echo {
echo   \"{#NETWORK_CLIENT}\": \"$hostname\",
echo   \"{#NETWORK_IP}\": \"$ip\"
echo }, 
done
echo {
echo   \"{#NETWORK_CLIENT}\": \"test\",
echo   \"{#NETWORK_IP}\": \"test2\"
echo }
echo ]}
```

/etc/steph/download.sh
```
#!/bin/sh
IFS=$'\n'
for ligne in $(sudo /usr/sbin/nlbw -c show -n -g host|grep -v 10.255|grep -v 00:00:00|tail -n+2)
do
ip=$(echo $ligne|awk '{print $1}')
hostname=$(dig -x $ip +short)
download=$(echo $ligne|awk '{print $4}'|awk -F'B' '{print $1}')
if [ "$hostname" == "$1" ]; then
  echo $download 
fi
done
```

/etc/steph/upload.sh
```
#!/bin/sh
IFS=$'\n'
for ligne in $(sudo /usr/sbin/nlbw -c show -n -g host|grep -v 10.255|grep -v 00:00:00|tail -n+2)
do
ip=$(echo $ligne|awk '{print $1}')
hostname=$(dig -x $ip +short)
upload=$(echo $ligne|awk -F ')' '{print $3}' | awk '{print $1}'|awk -F'B' '{print $1}')
if [ "$hostname" == "$1" ]; then
  echo $upload
fi
done
```

/etc/zabbix_agentd.conf.d/monitor.conf
```
UserParameter=custom.network.discovery, /etc/steph/network_discovery.sh
UserParameter=custom.network.download[*], /etc/steph/download.sh $1 
UserParameter=custom.network.upload[*], /etc/steph/upload.sh $1 
```

And then you can create a zabbix template with the discovery. I use grafana to create graphs : 
[image]

gwarf, 2020-10-02:
Looks quite cool, so now I've got to deploy some zabbix and grafana somewhere... 😅 
A native/easy way to get this (or even a bit simpler) overview directly in openmptcprouter would be wonderful.
Or even (with) something running on the VPS as for pihole?

Merrick28, 2020-10-03:
Less quick and dirty : https://github.com/Merrick28/zabbix-Openwrt-Clients-monitor

@Ysurac, I think we can close ?

gwarf, 2020-10-05:
It looks great, it may be nice to have a place in the wiki where this kind of nice stuff could be documented (or at least listed with a short description) so that others could discover them for using and potentially help improving them.

closed 2021-01-09 by github-actions[bot]