Ysurac/openmptcprouter #1269: Configuration router down

AndroFlo, 2020-10-01:
Hi,
I have a Huawei B726 4g router and a Adsl router.
I have configured like the instructions with ip etc.

My rapserry p3 is pluged in ethernet on my adsl router and the 4g box also to the adsl router.

But my connection only work on Adsl Box like the screenshot say this [screenshot](https://ibb.co/0XLHx7Q) the 4g box is down.

If i connect my rasperry to the 4g box its the Adsl Box Down ([screenshot](https://ibb.co/gjySLRy))

Ipv6 is disabled and dhcp router also on the 2 router.

Does someone have an idea ?

Thank's

Ysurac, 2020-10-01:
Please use the template when opening an issue. I always need versions, architecture,...

In your case the 4G box can't be reached in first configuration and adsl box in second config. This seems to be a physical connection problem. Both box are ethernet with a switch on the back ? Check connections and cables.

AndroFlo, 2020-10-02:
@Ysurac thnak's for your response and sorry for the template.
I changed the ethernet cable and it was the problem.

Thank's

closed 2020-10-02 by AndroFlo