Ysurac/openmptcprouter #1270: Iphone USB Sharing

KaylFr, 2020-10-01:
## Expected Behavior
Connect iPhone USB Sharing connection

## Current Behavior
Connect an iphone by adding connection

## Context (Environment)
I followed the info from:
https://github.com/Ysurac/openmptcprouter/issues/606#issuecomment-523109105
On the iPhone at the top right I have the icon going to blue but it displays Gateway down in OMB

[image]
[image]

## Specifications

  - OpenMPTCProuter version: v0.55.3 r0+14198-da98603597
  - OpenMPTCProuter VPS version: 0.1017 5.4.52
  - OpenMPTCProuter platform: Raspberry  Pi 4

KaylFr, 2020-10-01:
how to know the ip?

Ysurac, 2020-10-01:
No idea, I don't have an iPhone.

Mutanter, 2020-10-01:
double check your hotspot is active and the device is in trusted state
change the subnet to /28 (255.255.255.240)
reconnect your phone

closed 2021-01-05 by github-actions[bot]