Ysurac/openmptcprouter #1271: How to setup redundant connection? not aggregation.

openaspace, 2020-10-01:
HI need to use the system for live streaming events using LTE usb key.

From my tests  the  connection switch from one to another data sim or cable connection it's not without interruption. 
Some packets are lost and some times it employ from 5 to 10 seconds to make the switch... this mean that it's not working in redundant mode, but in aggregation mode.

I need to have the clone of the same signal on all connections in the the tunnel , to not loose in any case the connection.

I'm not understanding how to set this in explicit way. 
In the control panel the only parameter I see is in the "Master interface selection" with "balance" "Dynamic changes" etc.. 

But how to set a real realtime backup without loosing 1ms of pachets?

Really thank you.

Ysurac, 2020-10-01:
Please use template when submitting an issue.

Ysurac, 2020-10-01:
And for redundant mode, it's in Network->MPTCP.

openaspace, 2020-10-01:
Really thank you.
But also in redundant mode and creating 2 subflow instead of 1 .. when a carrier go down , there is a little fraction of second where the global connection go down.
[https://www.dropbox.com/s/pv9hvys3c17tsrn/Schermata%202020-10-01%20alle%2021.52.21.png?dl=0](url)

openaspace, 2020-10-02:
Also with software streaming like OBS and VMIX, when a carrier disconnect and internet connection come back, are not enable to reconnect and I need to close and reopen the streaming software.... it's like the computer client, after the disconnection can't find the route...

Ysurac, 2020-10-02:
You shouldn't create more than 1 subflow.
You can increase Multipath TCP SYN retries and change Congestion Control in Network->MPTCP.

openaspace, 2020-10-02:
I have done a lot of test and configuration.. but also when a carrier go down and I see that the ping command don't produce any lost packet... the streaming connection using OBS or an external hardware encoder, go completely down.

Also, if for example I'm working with a simple file upload, when the wan switch happen , the transfer continue to works correctly without problems. Therefore the real problem is how they manage different kind of connection...

**There are some settings that I can test to avoid that a live video broadcast to a RTMP server go down when there is a wan switch?**

Ysurac, 2020-10-02:
TCP use shadowsocks, but UDP and ICMP use the VPN. RTMP is over TCP, so not the same path as ICMP.

In State->System log you can check what problem can have ss-redir.
Redundant mode for master interface selection should also have no downtime. But you can also check in log if there is no problem with this.

You should also check OBS log if there is any to know why it goes down and maybe increase some timeout.

openaspace, 2020-10-02:
This my configuration: 
https://www.dropbox.com/s/5snq9kteii3crz1/OpenMPTCProuter-Interfaces-LuCI.png?dl=0
https://www.dropbox.com/s/ie6a7jvvkum80ru/OpenMPTCProuter-LuCI.png?dl=0
https://www.dropbox.com/s/vd89ly0qogibh2q/OpenMPTCProuter-MPTCP-LuCI.png?dl=0

I have also tried to disable Glorytun and use MLVPN, but the system says me that there two default vpn ..but no other vpn are set as autostart or enabled..

Ysurac, 2020-10-02:
VPN must be modified only via the wizard using advanced settings checkbox.

openaspace, 2020-10-02:
Looking other similar system I see that exist the "**Multipath broadcast**" where for example with 3 wan sending simultaneously the same data in the vpn... exist this settings.. because I have in any case a small downtime..

If exit a downtime i can use any lte router with failover..

openaspace, 2020-10-02:
> VPN must be modified only via the wizard using advanced settings checkbox.

Ok. I have modified but whitout significant results...

Ysurac, 2020-10-02:
It's what do redundant mode I think. I never used any similar system.
Did you check log ?

openaspace, 2020-10-02:
`Fri Oct  2 14:27:19 2020 daemon.notice netifd: Network device 'eth1' link is down
Fri Oct  2 14:27:19 2020 daemon.notice netifd: Interface 'wan1' has link connectivity loss
Fri Oct  2 14:27:19 2020 kern.info kernel: [ 1459.857648] r8152 2-3:1.0 eth1: carrier off
Fri Oct  2 14:27:29 2020 daemon.err /usr/bin/ss-local[17142]: getpeername: Socket not connected
Fri Oct  2 14:27:29 2020 daemon.notice netifd: Network device 'eth1' link is up
Fri Oct  2 14:27:29 2020 daemon.notice netifd: Interface 'wan1' has link connectivity
Fri Oct  2 14:27:29 2020 kern.info kernel: [ 1469.797525] r8152 2-3:1.0 eth1: carrier on
Fri Oct  2 14:27:30 2020 user.notice post-tracking-post-tracking: wan1 (eth1) switched off because gateway down and ping from 192.168.20.150 error (8.8.8.8,8.8.4.4)
Fri Oct  2 14:27:30 2020 user.notice post-tracking-post-tracking: Delete default route via 192.168.20.1 dev eth1
Fri Oct  2 14:27:30 2020 user.notice post-tracking-post-tracking: wan1 down. Replace default route by 192.168.20.1 dev eth2
Fri Oct  2 14:27:39 2020 daemon.notice netifd: Network device 'eth2' link is down
Fri Oct  2 14:27:39 2020 daemon.notice netifd: Interface 'wan2' has link connectivity loss
Fri Oct  2 14:27:40 2020 user.notice post-tracking-post-tracking: wan1 (eth1) switched up
Fri Oct  2 14:27:40 2020 user.notice post-tracking-post-tracking: Set server vps (144.76.151.62) default route via 192.168.20.1
Fri Oct  2 14:27:41 2020 daemon.err /usr/bin/ss-redir[17144]: remote recv: Operation timed out
Fri Oct  2 14:27:41 2020 daemon.err glorytun[7886]: read: Operation timed out
Fri Oct  2 14:27:41 2020 daemon.info glorytun[7886]: STOPPED tun0
Fri Oct  2 14:27:41 2020 daemon.err glorytun[7886]: 144.76.151.62.65001: connected
Fri Oct  2 14:27:42 2020 user.notice post-tracking-post-tracking: Set server vps (144.76.151.62) route via 192.168.20.1 metric 3
Fri Oct  2 14:27:45 2020 daemon.err /usr/bin/ss-redir[17144]: remote recv: Operation timed out
Fri Oct  2 14:27:47 2020 daemon.err omr-tracker[6736]: Failed to send request: Operation not permitted
Fri Oct  2 14:27:48 2020 daemon.err /usr/bin/ss-redir[17144]: remote recv: Operation timed out
Fri Oct  2 14:27:51 2020 daemon.err glorytun[7886]: 144.76.151.62.65001: key exchange failed
Fri Oct  2 14:27:51 2020 daemon.info glorytun[7886]: STOPPED tun0
Fri Oct  2 14:27:52 2020 daemon.err glorytun[7886]: 144.76.151.62.65001: connected
Fri Oct  2 14:27:53 2020 daemon.err /usr/bin/ss-redir[17143]: remote recv: Operation timed out
Fri Oct  2 14:27:53 2020 user.notice post-tracking-post-tracking: wan2 (eth2) switched off because gateway down and ping from 192.168.20.151 error (8.8.4.4,9.9.9.9)
Fri Oct  2 14:27:53 2020 user.notice post-tracking-post-tracking: Delete default route via 192.168.20.1 dev eth2
Fri Oct  2 14:27:56 2020 user.notice post-tracking-post-tracking: omrvpn down. Replace default route by 192.168.20.1 dev eth1
Fri Oct  2 14:27:56 2020 user.notice post-tracking-post-tracking: omrvpn down because gateway down
Fri Oct  2 14:27:56 2020 user.notice post-tracking-post-tracking: Tunnel down use ShadowSocks for UDP
Fri Oct  2 14:27:56 2020 user.notice Shadowsocks: Rules DOWN
Fri Oct  2 14:27:57 2020 user.notice Shadowsocks: Rules UP
Fri Oct  2 14:27:57 2020 daemon.info glorytun[7886]: STARTED tun0
Fri Oct  2 14:27:57 2020 user.notice Shadowsocks: Reload omr-bypass rules
Fri Oct  2 14:27:57 2020 user.notice omr-bypass: Starting OMR-ByPass...
Fri Oct  2 14:27:57 2020 user.notice omr-bypass: Reload dnsmasq...
Fri Oct  2 14:27:57 2020 daemon.info dnsmasq[31976]: read /etc/hosts - 4 addresses
Fri Oct  2 14:27:57 2020 daemon.info dnsmasq[31976]: read /tmp/hosts/dhcp.cfg01411c - 2 addresses
Fri Oct  2 14:27:57 2020 daemon.info dnsmasq-dhcp[31976]: read /etc/ethers - 0 addresses
Fri Oct  2 14:27:57 2020 user.notice omr-bypass: OMR-ByPass is running
Fri Oct  2 14:28:03 2020 user.notice post-tracking-post-tracking: Tunnel up : Replace default route by 10.255.255.1 dev tun0
Fri Oct  2 14:28:05 2020 user.notice post-tracking-post-tracking: Tunnel up disable use of ShadowSocks for UDP
Fri Oct  2 14:28:05 2020 user.notice Shadowsocks: Rules DOWN
Fri Oct  2 14:28:05 2020 user.notice Shadowsocks: Rules UP
Fri Oct  2 14:28:05 2020 user.notice Shadowsocks: Reload omr-bypass rules
Fri Oct  2 14:28:05 2020 user.notice omr-bypass: Starting OMR-ByPass...
Fri Oct  2 14:28:06 2020 user.notice omr-bypass: Reload dnsmasq...
Fri Oct  2 14:28:06 2020 daemon.info dnsmasq[31976]: read /etc/hosts - 4 addresses
Fri Oct  2 14:28:06 2020 daemon.info dnsmasq[31976]: read /tmp/hosts/dhcp.cfg01411c - 2 addresses
Fri Oct  2 14:28:06 2020 daemon.info dnsmasq-dhcp[31976]: read /etc/ethers - 0 addresses
Fri Oct  2 14:28:06 2020 user.notice omr-bypass: OMR-ByPass is running
Fri Oct  2 14:28:09 2020 daemon.notice netifd: Network device 'eth2' link is up
Fri Oct  2 14:28:09 2020 daemon.notice netifd: Interface 'wan2' has link connectivity
Fri Oct  2 14:28:09 2020 kern.info kernel: [ 1509.623854] asix 1-6:1.0 eth2: link up, 100Mbps, full-duplex, lpa 0xCDE1
Fri Oct  2 14:28:16 2020 user.notice post-tracking-post-tracking: wan2 (eth2) switched up
Fri Oct  2 14:28:16 2020 user.notice post-tracking-post-tracking: Set server vps (144.76.151.62) route via 192.168.20.1 metric 4
Fri Oct  2 14:28:17 2020 daemon.notice netifd: Network device 'eth1' link is down
Fri Oct  2 14:28:17 2020 daemon.notice netifd: Interface 'wan1' has link connectivity loss
[... cut ...]
Fri Oct  2 14:28:20 2020 daemon.err /usr/bin/ss-redir[17143]: remote recv: Host is unreachable
Fri Oct  2 14:28:22 2020 daemon.err /usr/bin/ss-redir[17144]: remote recv: Operation timed out
Fri Oct  2 14:28:22 2020 daemon.err /usr/bin/ss-redir[17144]: remote recv: Operation timed out
Fri Oct  2 14:28:23 2020 daemon.err /usr/bin/ss-redir[17143]: remote recv: Host is unreachable
Fri Oct  2 14:28:23 2020 daemon.err /usr/bin/ss-redir[17144]: remote recv: Host is unreachable
Fri Oct  2 14:28:23 2020 daemon.err /usr/bin/ss-redir[17144]: remote recv: Host is unreachable`

openaspace, 2020-10-02:
> It's what do redundant mode I think.
Here you can find the "broadcast" mode
https://zerotier.atlassian.net/wiki/spaces/SD/pages/568459265/Multipath

openaspace, 2020-10-02:
Selecting " Force retrieve all keys from server." the system appear more reactive..
But a disconnection till happen.. and I see that can't connect to remote server.. can't retrieve scripts....

openaspace, 2020-10-02:
How the "broadcast" diagram show, I need to obtain exactly the maximum stability with top redundancy over 3 Lte usb keys

Ysurac, 2020-10-02:
It's a bonding mode, not related to MPTCP and not really efficient with links that don't have same latency.
Force retrieve all keys from server, retrieve all keys/settings from the server when there is an issue, so it's not related at all.
I don't understand what you mean by "can't connect to remote server.. can't retrieve scripts...", all is green in status page ?

In log I can see problem with 2 interfaces, but what I need is log when you have an issue only.

openaspace, 2020-10-02:
Where are the deep logs in the system?
The problem with the two interfaces is the simulation of manual link down detaching the ethernet.

.. When  link 1 go down, in the main interfaces .. the system report both interfaces down.. but only one is without cable. .. and when unlink ethernet 2 the openmtcrouter section report that can't connect server admin script. (i'm making tests with 1gbps fiber connection with 5ms latency)

Please look the two screenshot.
https://www.dropbox.com/s/xkzh47o673xdeq8/OpenMPTCProuter-Stato-LuCI%20%281%29.png?dl=0
https://www.dropbox.com/s/jofw77oz2is12vm/OpenMPTCProuter-Stato-LuCI.png?dl=0

Ok, not related to MPTCP .. but how to set a different bonding logic in the preferences??

thank you for the support.

Mutanter, 2020-10-02:
Ping (ICMP) != RTMP Stream (TCP)
Your OBS behavior is also not MPTCP related.
Just aggregate the links and your problems are gone?

Ysurac, 2020-10-02:
Deep log ? All logs are in State->System log and State->Kernel log.

Detaching the ethernet is not the same as link down.
Redundant mode can't work when an interface is set as backup (this seems normal...).

Screenshot without any info doesn't help. What did you do to get them ?

Please can you give me what I asked you: the log when there is the issue.

Ysurac, 2020-10-02:
Detaching cable give an info to the router.

I need a test done without backup mode and the result in system log.

openaspace, 2020-10-02:
Without backup: 
[ 2779.845792] asix 1-6:1.0 eth2: link up, 100Mbps, full-duplex, lpa 0xCDE1
[ 2787.047082] r8152 2-3:1.0 eth1: carrier off
[ 2799.195392] r8152 2-3:1.0 eth1: carrier on
[ 3033.353440] r8152 2-3:1.0 eth1: carrier off
[ 3039.418097] r8152 2-3:1.0 eth1: carrier on
[ 3045.025038] asix 1-6:1.0 eth2: link up, 100Mbps, full-duplex, lpa 0xCDE1
[ 3089.050936] asix 1-6:1.0 eth2: link up, 100Mbps, full-duplex, lpa 0xCDE1
[ 3096.412954] r8152 2-3:1.0 eth1: carrier off
[ 3102.321378] r8152 2-3:1.0 eth1: carrier on
[ 3161.811888] r8152 2-3:1.0 eth1: carrier off
[ 3165.576616] r8152 2-3:1.0 eth1: carrier on
[ 3171.858099] asix 1-6:1.0 eth2: link up, 100Mbps, full-duplex, lpa 0xCDE1

When the cable disconnect , ping timeout.
In the backup mode, sometimes the no time out , the ms increase to 900ms for two ping but no disconnections.

openaspace, 2020-10-02:
if in the backup mode: 
icmp_seq=281 ttl=54 time=29.305 ms
icmp_seq=282 ttl=54 time=29.546 ms
**icmp_seq=285 ttl=54 time=2703.376 ms**
**icmp_seq=286 ttl=54 time=1698.446 ms**
**icmp_seq=287 ttl=54 time=693.803 ms**
icmp_seq=288 ttl=54 time=28.901 ms
icmp_seq=289 ttl=54 time=29.200 ms

 3250.855547] r8152 2-3:1.0 eth1: carrier off
[ 3254.364429] r8152 2-3:1.0 eth1: carrier on
[ 3268.610049] asix 1-6:1.0 eth2: link up, 100Mbps, full-duplex, lpa 0xCDE1
[ 3329.657587] asix 1-6:1.0 eth2: link up, 100Mbps, full-duplex, lpa 0xCDE1
[ 3337.594774] r8152 2-3:1.0 eth1: carrier off
[ 3345.455679] r8152 2-3:1.0 eth1: carrier on
[ 3352.570198] r8152 2-3:1.0 eth1: carrier off
[ 3358.701851] r8152 2-3:1.0 eth1: carrier on
[ 3369.844017] asix 1-6:1.0 eth2: link up, 100Mbps, full-duplex, lpa 0xCDE1

openaspace, 2020-10-02:
Could be more easy if you connect directly to my desk? :) 🥇

Ysurac, 2020-10-02:
I need system log, not kernel log.

openaspace, 2020-10-02:
From system log of the web control panel
Fri Oct  2 19:00:29 2020 daemon.notice netifd: Network device 'eth2' link is down
Fri Oct  2 19:00:29 2020 daemon.notice netifd: Interface 'wan2' has link connectivity loss
Fri Oct  2 19:00:31 2020 user.notice post-tracking-post-tracking: Set server vps (xxx.xxx.xxx.xxx) default route xxx.xxx.xxx.xxx nexthop via 192.168.20.1 dev eth1 weight 1 nexthop via 192.168.20.1 dev eth2 weight 10
Fri Oct  2 19:00:33 2020 daemon.err /usr/bin/ss-local[17470]: getpeername: Socket not connected
Fri Oct  2 19:00:35 2020 user.notice post-tracking-post-tracking: Set server vps (xxx.xxx.xxx.xxx) default route xxx.xxx.xxx.xxx nexthop via 192.168.20.1 dev eth1 weight 1 nexthop via 192.168.20.1 dev eth2 weight 10
Fri Oct  2 19:00:35 2020 daemon.notice netifd: Network device 'eth2' link is up
Fri Oct  2 19:00:35 2020 daemon.notice netifd: Interface 'wan2' has link connectivity
Fri Oct  2 19:00:35 2020 kern.info kernel: [ 1069.490931] asix 1-6:1.0 eth2: link up, 100Mbps, full-duplex, lpa 0xCDE1
Fri Oct  2 19:00:36 2020 daemon.err /usr/bin/ss-redir[17471]: remote recv: Host is unreachable
Fri Oct  2 19:00:36 2020 daemon.err /usr/bin/ss-redir[17471]: remote recv: Operation timed out
Fri Oct  2 19:00:37 2020 daemon.err /usr/bin/ss-redir[17471]: remote recv: Operation timed out

Ysurac, 2020-10-02:
Both WAN are on same network with same gateway, this can't work well.

openaspace, 2020-10-03:
Hello.
I have switched to two lte connection using usb thetering... but using redundant or round-robin the problem is the same like the previous configuration with both wan with the same gateway.

I observe that randomly .. can be that I'm streaming with obs a 1000kbps and on the two phone I see 500k upload connection for each one.. and this is when it works, because i detach one connection and all split to the other looking upload bandwidth at 1000kbps... with no OBS disconnection.. therefore it's not a tcp connection of the vpn and icmp protocol(?)

but...

When I reconnect the disconnected connection, I see that the data continue to go only on the previous active connection, and the reconnected connection remain idle with some kb random of data under 10k...

So, I have set to round-robin... and the same.. set to redundant.. and it's the same...

Ok, I have decided to wait 2/3 minutes to see if the algorithm recreate the bonding correctly.. but this don't happen..

finally disconnecting the single connection that is trasfering all the data,....the obs connection go down and after some second the bandwidth is splitted to the second one...

This is a clear bad configuration?

Please let me know.
Really thank you for your job and support!

openaspace, 2020-10-03:
And also I observe that when a new connection start it's redirected some times only to only one wan , without split the traffic for both devices.. 
therefore can be that I have the terminal active with a ping command.. and obs streaming...

I disconnect one connection and streaming go down, but ping go on....
or ping go down and streaming remaining on.

This a clear bad configuration of bonding redundancy

openaspace, 2020-10-03:
So.. I think to have found the problem!

I have set congestion control to CUBIC  and REDUNDAT.

I start the stream on OBS and on the two phones I see 1000kbps upload on each one!! 👍 

... At this point all is working and I disconnect one phone connection, no downtime, obs is correctly streaming without problems! yeah!.... but .. I reconnect the disconnected phone.. and the tcp data continue to go only on the always on phone connection, without restart the REDUNDANT on the reconnected phone!

To have newly the redundancy , I need to stop the STREAM and start it NEWLY!
...... so how to correct this problem?

Ysurac, 2020-10-04:
As always I need system log when it's not working.

openaspace, 2020-10-04:
Hello, thank you!
Here the complete Multipath Debug, where 2 tethering lte device are connected, an obs stream stream in redundancy, both devices are working correctly, one device is disconnected and reactivated.. without restart the redundant connecttion.

Sun Oct  4 06:59:32 2020 kern.err kernel: [  701.716755] mptcp_alloc_mpcb: created mpcb with token 0xa7a7919
Sun Oct  4 06:59:32 2020 kern.err kernel: [  701.716818] mptcp_add_sock: token 0xa7a7919 pi 1, src_addr:192.168.42.117:62053 dst_addr:vps_server: xxx.xxx.xxx.xxx:65101
Sun Oct  4 06:59:32 2020 kern.err kernel: [  701.797492] mptcp_add_sock: token 0xa7a7919 pi 2, src_addr:0.0.0.0:0 dst_addr:0.0.0.0:0
Sun Oct  4 06:59:32 2020 kern.err kernel: [  701.797589] __mptcp_init4_subsockets: token 0xa7a7919 pi 2 src_addr:192.168.42.29:0 dst_addr:vps_server: xxx.xxx.xxx.xxx:65101 ifidx: 29
Sun Oct  4 06:59:43 2020 kern.err kernel: [  712.964001] mptcp_close: Close of meta_sk with tok 0x55cf11ed
Sun Oct  4 06:59:43 2020 kern.err kernel: [  712.985605] mptcp_del_sock: Removing subsock tok 0x55cf11ed pi:1 state 7 is_meta? 0
Sun Oct  4 06:59:43 2020 kern.err kernel: [  713.028685] mptcp_del_sock: Removing subsock tok 0x55cf11ed pi:2 state 7 is_meta? 0
Sun Oct  4 06:59:43 2020 kern.err kernel: [  713.028759] mptcp_sock_destruct destroying meta-sk token 0x55cf11ed
Sun Oct  4 06:59:46 2020 kern.err kernel: [  715.430367] mptcp_alloc_mpcb: created mpcb with token 0xa7475052
Sun Oct  4 06:59:46 2020 kern.err kernel: [  715.430428] mptcp_add_sock: token 0xa7475052 pi 1, src_addr:192.168.42.117:62065 dst_addr:vps_server: xxx.xxx.xxx.xxx:65101
Sun Oct  4 06:59:46 2020 kern.err kernel: [  715.517400] mptcp_add_sock: token 0xa7475052 pi 2, src_addr:0.0.0.0:0 dst_addr:0.0.0.0:0
Sun Oct  4 06:59:46 2020 kern.err kernel: [  715.517493] __mptcp_init4_subsockets: token 0xa7475052 pi 2 src_addr:192.168.42.29:0 dst_addr:vps_server: xxx.xxx.xxx.xxx:65101 ifidx: 29
Sun Oct  4 06:59:46 2020 kern.err kernel: [  715.521144] mptcp_alloc_mpcb: created mpcb with token 0xaaac7d9
Sun Oct  4 06:59:46 2020 kern.err kernel: [  715.521219] mptcp_add_sock: token 0xaaac7d9 pi 1, src_addr:192.168.42.117:62071 dst_addr:vps_server: xxx.xxx.xxx.xxx:65101
Sun Oct  4 06:59:46 2020 kern.err kernel: [  715.610101] mptcp_add_sock: token 0xaaac7d9 pi 2, src_addr:0.0.0.0:0 dst_addr:0.0.0.0:0
Sun Oct  4 06:59:46 2020 kern.err kernel: [  715.610178] __mptcp_init4_subsockets: token 0xaaac7d9 pi 2 src_addr:192.168.42.29:0 dst_addr:vps_server: xxx.xxx.xxx.xxx:65101 ifidx: 29
Sun Oct  4 06:59:49 2020 kern.err kernel: [  719.245446] mptcp_close: Close of meta_sk with tok 0x2789d009
Sun Oct  4 06:59:49 2020 kern.err kernel: [  719.365309] mptcp_del_sock: Removing subsock tok 0x2789d009 pi:2 state 7 is_meta? 0
Sun Oct  4 06:59:49 2020 kern.err kernel: [  719.365494] mptcp_sock_destruct destroying meta-sk token 0x2789d009
Sun Oct  4 06:59:57 2020 kern.err kernel: [  726.997520] mptcp_del_sock: Removing subsock tok 0x4912f299 pi:1 state 7 is_meta? 0
Sun Oct  4 07:00:02 2020 kern.err kernel: [  732.252048] mptcp_close: Close of meta_sk with tok 0xaaac7d9
Sun Oct  4 07:00:02 2020 kern.err kernel: [  732.274330] mptcp_del_sock: Removing subsock tok 0xaaac7d9 pi:2 state 7 is_meta? 0
Sun Oct  4 07:00:03 2020 kern.err kernel: [  732.420643] mptcp_del_sock: Removing subsock tok 0xaaac7d9 pi:1 state 7 is_meta? 0
Sun Oct  4 07:00:03 2020 kern.err kernel: [  732.420726] mptcp_sock_destruct destroying meta-sk token 0xaaac7d9
Sun Oct  4 07:00:05 2020 kern.err kernel: [  734.637745] mptcp_close: Close of meta_sk with tok 0xa7a7919
Sun Oct  4 07:00:05 2020 kern.info kernel: [  734.678929] usb 1-1.4: USB disconnect, device number 13
Sun Oct  4 07:00:05 2020 daemon.notice netifd: Network device 'usb0' link is down
Sun Oct  4 07:00:05 2020 daemon.notice netifd: Interface 'wan2' has link connectivity loss
Sun Oct  4 07:00:05 2020 kern.info kernel: [  734.680659] rndis_host 1-1.4:1.0 usb0: unregister 'rndis_host' usb-0000:00:12.2-1.4, RNDIS device
Sun Oct  4 07:00:05 2020 kern.err kernel: [  734.683795] addr4_event_handler created event for 192.168.42.117, code 2 prio 0 idx 30
Sun Oct  4 07:00:05 2020 daemon.debug dnsmasq[1758]: stopped listening on usb0(#30): 192.168.42.117 port 53
Sun Oct  4 07:00:05 2020 kern.err kernel: [  734.685070] addr4_event_handler created event for 192.168.42.117, code 2 prio 0 idx 30
Sun Oct  4 07:00:05 2020 kern.err kernel: [  734.685142] add_pm_event del old_code 2
Sun Oct  4 07:00:05 2020 user.notice ModemManager: hotplug: remove network interface usb0: event processed
Sun Oct  4 07:00:05 2020 user.notice ModemManager: hotplug: event reported: action=remove, name=usb0, subsystem=net
Sun Oct  4 07:00:05 2020 kern.err kernel: [  734.725606] mptcp_del_sock: Removing subsock tok 0xa7a7919 pi:2 state 7 is_meta? 0
Sun Oct  4 07:00:05 2020 daemon.notice netifd: wan2 (30605): udhcpc: SIOCGIFINDEX: No such device
Sun Oct  4 07:00:05 2020 daemon.notice netifd: wan2 (30605): udhcpc: received SIGTERM
Sun Oct  4 07:00:05 2020 daemon.notice netifd: wan2 (30605): udhcpc: unicasting a release of 192.168.42.117 to 192.168.42.129
Sun Oct  4 07:00:05 2020 daemon.notice netifd: wan2 (30605): udhcpc: sending release
Sun Oct  4 07:00:05 2020 daemon.notice netifd: wan2 (30605): udhcpc: bind(UDP): Address not available
Sun Oct  4 07:00:05 2020 daemon.notice netifd: wan2 (30605): udhcpc: entering released state
Sun Oct  4 07:00:05 2020 daemon.notice netifd: wan2 (30605): Command failed: Permission denied
[... cut ...]
Sun Oct  4 07:00:33 2020 kern.err kernel: [  762.797699] mptcp_alloc_mpcb: created mpcb with token 0x5d094b1c
Sun Oct  4 07:00:33 2020 kern.err kernel: [  762.797759] mptcp_add_sock: token 0x5d094b1c pi 1, src_addr:192.168.42.117:62189 dst_addr:vps_server: xxx.xxx.xxx.xxx:65101
Sun Oct  4 07:00:33 2020 kern.err kernel: [  762.884224] mptcp_add_sock: token 0x5d094b1c pi 2, src_addr:0.0.0.0:0 dst_addr:0.0.0.0:0
Sun Oct  4 07:00:33 2020 kern.err kernel: [  762.884307] __mptcp_init4_subsockets: token 0x5d094b1c pi 2 src_addr:192.168.42.29:0 dst_addr:vps_server: xxx.xxx.xxx.xxx:65101 ifidx: 29

openaspace, 2020-10-04:
**UPDATE**
Configuration: 2 LTE tethering phones
Problem: when one phones disconnect and reconnect, no redundancy is newly established.
**NEWS: When the reconnected phone don't restart to transmit redundancy data after reconnection and I connect an ethernet device, on the reconnected phone restart the redundancy**

Sun Oct  4 07:15:34 2020 user.notice post-tracking-post-tracking: wan1 (eth1) switched up
Sun Oct  4 07:15:34 2020 user.notice post-tracking-post-tracking: Set server vps (vps_server: xxx.xxx.xxx.xxx) default route vps_server: xxx.xxx.xxx.xxx nexthop via 192.168.42.129 dev usb0 weight 1 nexthop via 192.168.42.129 dev usb1 weight 1 nexthop via 192.168.20.1 dev eth1 weight 1
Sun Oct  4 07:15:34 2020 daemon.err omr-tracker[12615]: sendto: Operation not permitted
Sun Oct  4 07:15:34 2020 daemon.err omr-tracker[12615]: sendto: Operation not permitted
Sun Oct  4 07:15:34 2020 kern.err kernel: [ 1663.821761] mptcp_alloc_mpcb: created mpcb with token 0x5e3d85a6
Sun Oct  4 07:15:34 2020 kern.err kernel: [ 1663.821822] mptcp_add_sock: token 0x5e3d85a6 pi 1, src_addr:192.168.42.117:62437 dst_addr:vps_server: xxx.xxx.xxx.xxx:65500
Sun Oct  4 07:15:34 2020 kern.err kernel: [ 1663.902489] mptcp_add_sock: token 0x5e3d85a6 pi 2, src_addr:0.0.0.0:0 dst_addr:0.0.0.0:0
Sun Oct  4 07:15:34 2020 kern.err kernel: [ 1663.902609] __mptcp_init4_subsockets: token 0x5e3d85a6 pi 2 src_addr:192.168.42.29:0 dst_addr:vps_server: xxx.xxx.xxx.xxx:65500 ifidx: 41
Sun Oct  4 07:15:34 2020 kern.err kernel: [ 1663.902721] mptcp_add_sock: token 0x5e3d85a6 pi 3, src_addr:0.0.0.0:0 dst_addr:0.0.0.0:0
Sun Oct  4 07:15:34 2020 kern.err kernel: [ 1663.902789] __mptcp_init4_subsockets: token 0x5e3d85a6 pi 3 src_addr:192.168.20.243:0 dst_addr:vps_server: xxx.xxx.xxx.xxx:65500 ifidx: 11
Sun Oct  4 07:15:34 2020 kern.err kernel: [ 1663.973246] mptcp_close: Close of meta_sk with tok 0x5e3d85a6
Sun Oct  4 07:15:34 2020 kern.err kernel: [ 1663.992557] mptcp_del_sock: Removing subsock tok 0x5e3d85a6 pi:3 state 7 is_meta? 0
Sun Oct  4 07:15:34 2020 kern.err kernel: [ 1664.044583] mptcp_del_sock: Removing subsock tok 0x5e3d85a6 pi:1 state 7 is_meta? 0
Sun Oct  4 07:15:34 2020 kern.err kernel: [ 1664.054188] mptcp_del_sock: Removing subsock tok 0x5e3d85a6 pi:2 state 7 is_meta? 0
Sun Oct  4 07:15:34 2020 kern.err kernel: [ 1664.054256] mptcp_sock_destruct destroying meta-sk token 0x5e3d85a6
Sun Oct  4 07:15:34 2020 kern.err kernel: [ 1664.061588] mptcp_alloc_mpcb: created mpcb with token 0x74b20d82
Sun Oct  4 07:15:34 2020 kern.err kernel: [ 1664.061643] mptcp_add_sock: token 0x74b20d82 pi 1, src_addr:192.168.42.117:62439 dst_addr:vps_server: xxx.xxx.xxx.xxx:65500
Sun Oct  4 07:15:34 2020 kern.err kernel: [ 1664.154721] mptcp_add_sock: token 0x74b20d82 pi 2, src_addr:0.0.0.0:0 dst_addr:0.0.0.0:0
Sun Oct  4 07:15:34 2020 kern.err kernel: [ 1664.154815] __mptcp_init4_subsockets: token 0x74b20d82 pi 2 src_addr:192.168.42.29:0 dst_addr:vps_server: xxx.xxx.xxx.xxx:65500 ifidx: 41
Sun Oct  4 07:15:34 2020 kern.err kernel: [ 1664.154945] mptcp_add_sock: token 0x74b20d82 pi 3, src_addr:0.0.0.0:0 dst_addr:0.0.0.0:0
Sun Oct  4 07:15:34 2020 kern.err kernel: [ 1664.155016] __mptcp_init4_subsockets: token 0x74b20d82 pi 3 src_addr:192.168.20.243:0 dst_addr:vps_server: xxx.xxx.xxx.xxx:65500 ifidx: 11
Sun Oct  4 07:15:34 2020 kern.err kernel: [ 1664.235469] mptcp_close: Close of meta_sk with tok 0x74b20d82
Sun Oct  4 07:15:34 2020 kern.err kernel: [ 1664.255267] mptcp_del_sock: Removing subsock tok 0x74b20d82 pi:3 state 7 is_meta? 0
Sun Oct  4 07:15:34 2020 kern.err kernel: [ 1664.314671] mptcp_del_sock: Removing subsock tok 0x74b20d82 pi:1 state 7 is_meta? 0
Sun Oct  4 07:15:34 2020 kern.err kernel: [ 1664.319542] mptcp_alloc_mpcb: created mpcb with token 0x74eebd4a
Sun Oct  4 07:15:34 2020 kern.err kernel: [ 1664.319596] mptcp_add_sock: token 0x74eebd4a pi 1, src_addr:192.168.42.117:62441 dst_addr:vps_server: xxx.xxx.xxx.xxx:65500
Sun Oct  4 07:15:34 2020 kern.err kernel: [ 1664.333275] mptcp_del_sock: Removing subsock tok 0x74b20d82 pi:2 state 7 is_meta? 0
Sun Oct  4 07:15:34 2020 kern.err kernel: [ 1664.333351] mptcp_sock_destruct destroying meta-sk token 0x74b20d82
Sun Oct  4 07:15:35 2020 kern.err kernel: [ 1664.407953] mptcp_add_sock: token 0x74eebd4a pi 2, src_addr:0.0.0.0:0 dst_addr:0.0.0.0:0
Sun Oct  4 07:15:35 2020 kern.err kernel: [ 1664.408049] __mptcp_init4_subsockets: token 0x74eebd4a pi 2 src_addr:192.168.42.29:0 dst_addr:vps_server: xxx.xxx.xxx.xxx:65500 ifidx: 41
Sun Oct  4 07:15:35 2020 kern.err kernel: [ 1664.408246] mptcp_add_sock: token 0x74eebd4a pi 3, src_addr:0.0.0.0:0 dst_addr:0.0.0.0:0
Sun Oct  4 07:15:35 2020 kern.err kernel: [ 1664.408320] __mptcp_init4_subsockets: token 0x74eebd4a pi 3 src_addr:192.168.20.243:0 dst_addr:vps_server: xxx.xxx.xxx.xxx:65500 ifidx: 11
Sun Oct  4 07:15:35 2020 kern.err kernel: [ 1664.490715] mptcp_close: Close of meta_sk with tok 0x74eebd4a
Sun Oct  4 07:15:35 2020 kern.err kernel: [ 1664.510240] mptcp_del_sock: Removing subsock tok 0x74eebd4a pi:3 state 7 is_meta? 0
Sun Oct  4 07:15:35 2020 kern.err kernel: [ 1664.510320] mptcp_del_sock: Removing subsock tok 0x74eebd4a pi:2 state 7 is_meta? 0
Sun Oct  4 07:15:35 2020 kern.err kernel: [ 1664.639526] mptcp_del_sock: Removing subsock tok 0x74eebd4a pi:1 state 7 is_meta? 0
Sun Oct  4 07:15:35 2020 kern.err kernel: [ 1664.639603] mptcp_sock_destruct destroying meta-sk token 0x74eebd4a
[... cut ...]
Sun Oct  4 07:16:45 2020 kern.err kernel: [ 1734.416478] mptcp_del_sock: Removing subsock tok 0x821a8ab1 pi:1 state 7 is_meta? 0
Sun Oct  4 07:16:45 2020 kern.err kernel: [ 1734.453247] mptcp_del_sock: Removing subsock tok 0xcb5f3a54 pi:3 state 7 is_meta? 0
Sun Oct  4 07:16:45 2020 kern.err kernel: [ 1734.453329] mptcp_sock_destruct destroying meta-sk token 0xcb5f3a54
Sun Oct  4 07:16:45 2020 kern.err kernel: [ 1734.457358] mptcp_del_sock: Removing subsock tok 0x821a8ab1 pi:3 state 7 is_meta? 0
Sun Oct  4 07:16:45 2020 kern.err kernel: [ 1734.457427] mptcp_sock_destruct destroying meta-sk token 0x821a8ab1

Mutanter, 2020-10-04:
Without having analyzed the logs, do you monitor the OMR status page in parallel to the errors? 
This sounds like similar problems I have encountered.
I've also done some tests in different constellations and had that feeling of randomness. 
As far as I can tell, the status of the network interface does matter and relate to the described behaviour (gateway down, no IP ( link ), cant reach dns etc.). Perhaps Ysurac has some more details on this.

openaspace, 2020-10-04:
> Without having analyzed the logs, do you monitor the OMR status page in parallel to the errors?
> This sounds like similar problems I have encountered.
> I've also done some tests in different [...]

Hello, thank you.
I will try to monitor in parallel OMR.

**The only relevant informations I see on the last logs are relative to the stopped vpn connection tun0  **
Sun Oct 4 07:00:09 2020 user.notice post-tracking-post-tracking: omrvpn down because gateway down
Sun Oct 4 07:00:09 2020 user.notice post-tracking-post-tracking: Tunnel down use ShadowSocks for UDP
Sun Oct 4 07:00:09 2020 user.notice Shadowsocks: Rules DOWN
Sun Oct 4 07:00:09 2020 daemon.err glorytun[7682]: write: Operation timed out
Sun Oct 4 07:00:09 2020 daemon.info glorytun[7682]: STOPPED tun0

°°°°°°°°°°°°°°°°°°°°°°°°°°°
Sun Oct 4 07:00:09 2020 daemon.err omr-tracker[12617]: Another app is currently holding the xtables lock. Perhaps you want to use the -w option?
Sun Oct 4 07:00:09 2020 user.notice Shadowsocks: Rules UP
Sun Oct 4 07:00:09 2020 kern.err kernel: [ 739.223244] mptcp_alloc_mpcb: created mpcb with token 0xa96a2273
Sun Oct 4 07:00:09 2020 kern.err kernel: [ 739.223305] mptcp_add_sock: token 0xa96a2273 pi 1, src_addr:192.168.42.29:15905 dst_addr:vps_server: xxx.xxx.xxx.xxx:65500
°°°°°°°°°°°°°°°°°°°°°°°°°°°

openaspace, 2020-10-04:
> Without having analyzed the logs, do you monitor the OMR status page in parallel to the errors?
> This sounds like similar problems I have encountered.
> I've also done some tests in different [...]

Also.. I'm thinking to change the laptop computer host where is running the openmptcrouter , testing it on a certified hardware server like an hold DELL T310 with no usb/ethernet adapter ....only for test.. because a dual xeon cpu is a little power expensive for a system like this.

openaspace, 2020-10-05:
> Without having analyzed the logs, do you monitor the OMR status page in parallel to the errors?
> This sounds like similar problems I have encountered.
> I've also done some tests in different [...]

I'm becoming crazy, where to se the OMR status page?

I'm experiencing really random redundant upload connection.. i disconnect and reconnect a device..it sometimes restart redundancy or not..  or could be.. if not restart.. that I connect the ethernet cable... and it start the redundancy on the ethernet and old one where was not starting....

completely random!

Ysurac, 2020-10-05:
As I already said, same gateway IP for usb1 and usb2 is not a good idea.

openaspace, 2020-10-05:
Hello, as indicated in the logs, i'm using 3 different gateways.
2 LTE with different operators and 1 lan cable.

Ysurac, 2020-10-05:
So there is a problem somewhere: ```Set server vps (vps_server: xxx.xxx.xxx.xxx) default route vps_server: xxx.xxx.xxx.xxx nexthop via 192.168.42.129 dev usb0 weight 1 nexthop via 192.168.42.129 dev usb1 weight 1 nexthop via 192.168.20.1 dev eth1 weight 1```

openaspace, 2020-10-05:
> So there is a problem somewhere: `Set server vps (vps_server: xxx.xxx.xxx.xxx) default route vps_server: xxx.xxx.xxx.xxx nexthop via 192.168.42.129 dev usb0 weight 1 nexthop via 192.168.42.129 dev usb1 weight 1 nexthop via 192.168.20.1 dev eth1 weight 1`

obviously xxx.xxx.xxx.xxx is my datacenter server ip obfuscated...

Ysurac, 2020-10-05:
It's why I'm talking about gateway...

openaspace, 2020-10-05:
> It's why I'm talking about gateway...

What you mean? 
SIM 1 - gateway assigned by the mobile isp
SIM 2 - gateway assigned by the mobile isp
Ethernet wan 1 - Assigned by the router (192.168.20.1)

Ysurac, 2020-10-05:
As you can see, usb0 and usb1 use same 192.168.42.129 gateway, you said you use 3 differents gateway.
Also route configuration is strange, this is like if there is no defined master connection: weight shoudldn't be all the same.

openaspace, 2020-10-05:
The only observation I can do , it that the two LTE phones connected as usb tethering from android , are receiving as default, the same usb/lan tethering subnet/broadcast, but the WAN IP is different.

root@OpenMPTCProuter:~# ifconfig
eth0      Link encap:Ethernet  HWaddr C8:0A:A9:2B:DE:0C  
          inet addr:192.168.100.1  Bcast:192.168.100.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:505868 errors:0 dropped:0 overruns:0 frame:0
          TX packets:267502 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:601462300 (573.5 MiB)  TX bytes:28222804 (26.9 MiB)

eth1      Link encap:Ethernet  HWaddr 00:E0:4C:35:7A:A5  
          inet addr:192.168.20.244  Bcast:192.168.20.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:372142 errors:0 dropped:0 overruns:0 frame:0
          TX packets:517761 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:100 
          RX bytes:30036305 (28.6 MiB)  TX bytes:412525260 (393.4 MiB)

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:549732 errors:0 dropped:0 overruns:0 frame:0
          TX packets:549732 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:613246742 (584.8 MiB)  TX bytes:613246742 (584.8 MiB)

tun0      Link encap:UNSPEC  HWaddr 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  
          inet addr:10.255.255.2  P-t-P:10.255.255.1  Mask:255.255.255.255
          UP POINTOPOINT RUNNING NOARP MULTICAST  MTU:1500  Metric:1
          RX packets:175 errors:0 dropped:0 overruns:0 frame:0
          TX packets:156 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:33653 (32.8 KiB)  TX bytes:12810 (12.5 KiB)

usb0      Link encap:Ethernet  HWaddr 6A:CD:45:95:B7:E7  
          inet addr:192.168.42.117  Bcast:192.168.42.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:189364 errors:0 dropped:0 overruns:0 frame:0
          TX packets:356977 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:100 
          RX bytes:15374525 (14.6 MiB)  TX bytes:306489665 (292.2 MiB)

usb1      Link encap:Ethernet  HWaddr 32:CE:82:29:56:22  
          inet addr:192.168.42.29  Bcast:192.168.42.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1400  Metric:1
          RX packets:197878 errors:28 dropped:0 overruns:0 frame:28
          TX packets:338181 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:100 
          RX bytes:17524690 (16.7 MiB)  TX bytes:289634251 (276.2 MiB)

openaspace, 2020-10-05:
> As you can see, usb0 and usb1 use same 192.168.42.129 gateway, you said you use 3 differents gateway.
> Also route configuration is strange, this is like if there is no defined master connection: weight shoudldn't be all the same.

This could be a really bed problem:

"Unfortunately, there is no way to change the IP address range for the builtin tethering support, even in CyanogenMod. The Wi-Fi hotspot IP is hardcoded in android.net.wifi.WifiStateMachine.startTethering(); the USB tethering IP is hardcoded in com.android.server.connectivity.Tethering"

Ysurac, 2020-10-05:
ok, but I had no idea that this was tethering and you said gateway was not the same.
You can still try with only 2 connections (only one tethering) and check if same issue. Also set one connection as master as I said.

openaspace, 2020-10-05:
> ok, but I had no idea that this was tethering and you said gateway was not the same.
> You can still try with only 2 connections (only one tethering) and check if same issue. Also set one connection as master as I said.

With only 1 lte tethering and wan cable appear to work... but my question is.. 
what happen with LTE usb key that are working with embed virtual lan that usually are all on 192.168.1.0/24 networks?

This is a really big limit don't distinguish that a same subnet is coming from different devices...

Ysurac, 2020-10-05:
This should work with USB dongle, there is no such limit. Only a limit if it's same physical network.

openaspace, 2020-10-05:
...Connected a USB dongle at the system boot.. The router replaced the previous wan usb device with the lan of the USB stick automatically...

Ysurac, 2020-10-05:
It's not the same issue.

openaspace, 2020-10-05:
> This should work with USB dongle, there is no such limit. Only a limit if it's same physical network.

For Live events on the road the usb tethering with mobile phones it's a salvation... there is really no way to solve the problem that iphones and android devices are using all the same lan subnet?

Ysurac, 2020-10-05:
As I don't know of such problem, I don't know.

openaspace, 2020-10-05:
> As I don't know of such problem, I don't know.

The solution could be to isolate each device within a dedicated different subnet.

For each device the system will create a different subnet where to receive the device subnet, in this way each device net will be managed and isolated from others.

Mutanter, 2020-10-05:
Ok, bear in mind and just to clarify : 
If you connect your tethered Phone, you'll get a new ETH interface in OMR / OpenWRT.
For example: new Interface for your phone is ETH2
If you disconnect this Phone your ETH2 Interface will be "deleted" but it is still visible in OMR Status page.
Suppose we now connect a new interface (either a USB LTE stick or another cell phone or another USB to LAN interface etc.). ) is automatically assigned to the device ETH 2 and this is exactly what causes problems.
OMR still "hold" the "old" ETH2 configuration while OpenWRT / linux Kernel re-assigned ETH2 to the "new" device.

openaspace, 2020-10-05:
Really thank you.
I'm understanding that will be better to buy 4 LTE router , each one on different subnet  and connect each one to the router bonding.... it's only sane way to operate...

Malaga82, 2020-10-05:
probably the best you can buy with budget is the Huawei B535, at least is 2CA.

openaspace, 2020-10-07:
> ok, but I had no idea that this was tethering and you said gateway was not the same.
> You can still try with only 2 connections (only one tethering) and check if same issue. Also set one connection as master as I said.

After the last the system was used and working perfectly, now without change nothing, with 1 LTE tethering and 1 Lan device, the redundancy restart only on the tether device and not on the lan , after a disconnection...

.....

openaspace, 2020-10-07:
The problem is always the same.. 
1) both wan are online
2) I start the streaming and both wan I see are transferring data at the same speed
3) detach cable from wan1  and the stream works over wan 2
4) reattach wan 1 and the stream continue to be transfered only over wan 2

This replicate wan1 to wan2

Ysurac, 2020-10-07:
I'm not able to reproduce the issue.
Can you try to use ```omr-test-speed``` and do a plug off/plug on ? And check speed and MPTCP realtime traffic ?
i'm really not sure that MPTCP redundant mode really work well in all case...

[30 more replies]

closed 2021-03-11 by github-actions[bot]