Ysurac/openmptcprouter #1272: mlvpn not starting : getaddrinfo error: Unknown error

pquan, 2020-10-02:
## Expected Behavior

MLVPN tunnel working when enabled from LUCI

## Current Behavior

MLVPN tunnel does not start. There are errors in the log files as follows:

```
logread -l 100 |grep mlvpn
Fri Oct  2 09:39:47 2020 daemon.warn mlvpn[4983]: 8 getaddrinfo error: Unknown error
Fri Oct  2 09:39:47 2020 daemon.warn mlvpn[4983]: 7 getaddrinfo error: Unknown error
Fri Oct  2 09:39:49 2020 daemon.warn mlvpn[4983]: 8 getaddrinfo error: Unknown error
Fri Oct  2 09:39:49 2020 daemon.warn mlvpn[4983]: 7 getaddrinfo error: Unknown error
Fri Oct  2 09:39:53 2020 daemon.warn mlvpn[4983]: 7 getaddrinfo error: Unknown error
Fri Oct  2 09:39:53 2020 daemon.warn mlvpn[4983]: 8 getaddrinfo error: Unknown error
```

## Possible Solution

Not sure. Looks like a problem in the resolver or configuration issue. Does not seem realted the the conf file format but more resolver library / cross-compilation difference in lib versions etc.

## Steps to Reproduce the Problem

  1. Install openmptcp router with default settings.
  2. Disable all VPN (glorutun, dsvpn etc)
  3. Enable only MLVPN
  4. logread -l 1000 -f |grep mlvpn

## Context (Environment)

I'm trying to use mlvpn in place of other vpn solutions (which have other issues) for a DSL+LTE backup link (same latency actually and similar speeds), while passing all traffic in one of the two vpn.

## Specifications

  - OpenMPTCProuter version:  openmptcprouter v0.55.3 r0+14198-da98603597 / LuCI Master git-20.219.57394-35e2735
  - OpenMPTCProuter VPS version:  Don't really know / does not apply (?)
  - OpenMPTCProuter platform:  Raspberry Pi 4 Model B Rev 1.1

Ysurac, 2020-10-02:
It's a know issue in current release, it's why MLVPN is not available in wizard, fixed in next release.

pquan, 2020-10-02:
This is a super quick response! 
If/when you have a commit, I can compile locally and test before the release. 

PS: I'm mostly interested in preventing data leakage between the router and the vpn, hence the usage of mptcp over vpn.

Ysurac, 2020-10-02:
it's already available in develop branch and in snapshots image.

MPTCP over VPN is not related to MLVPN. MLVPN is a VPN, like OpenVPN, glorytun, DSVPN,... MPTCP over VPN is made when an ISP block MPTCP, then MPTCP is done over OpenVPN.

By default everything is encrypted, via shadowsocks for TCP, and by VPN (default glorytun TCP) for UDP and ICMP. Default encryption is chacha20 or AES if router and VPS support AES hardware encryption.

pquan, 2020-10-03:
Oh I see. I'm still looking at how the various pieces are glued togherer :)
MLVPN comes up now on the develop branch. 

A side note:  When trying to install some packages, opkg is unable to download the package feeds from the dev server as wget is missing https support and there's a redirect from http->https there. ca-certificates is present but wget complains about missing libustream-ssl/tls:
```
wget https://download.openmptcprouter.com/develop/5.4/rpi4/packages/aarch64_cortex-a72/base/Packages.gz
wget: SSL support not available, please install one of the libustream-.*[ssl|tls] packages as well as the ca-bundle and ca-certificates packages.
```
libustream does not seem compiled in the current develop branch.

closed 2020-10-03 by pquan