Ysurac/openmptcprouter #1274: QUESTION:  use certain links more than others

tlkelley70, 2020-10-02:
I have searched and I cannot find an answer or a setting that could do this.  Basically I have two links that are limited and two that are not.  I would like to balance better the use of the two unlimited links so I retain the use of the two limited links longer as less data per day go over them.  I would give up some speed to ensure that I have this level of redundancy.  Is this possible?  So lets say I wanted 33% on one link 33% on another link, and 16.5% per the last two links each just as an example.

Ysurac, 2020-10-03:
Links are used based on latency, so it's not possible to do what you want.

tlkelley70, 2020-10-03:
Thanks for clarifying.

closed 2021-01-06 by github-actions[bot]