Ysurac/openmptcprouter #1275: IPv6 Forwarding

iamfreaky, 2020-10-02:
## Expected Behavior
IPv6 Port Opening should work, but it doesnt. 

Notation, This is with a Subnet of /48 on the VPS. (Online.net)
OpenMPTCP Gets /56

## Current Behavior
IPv6 Outgoing is working Properly
IPv6 Ingoing/Forwarding is not Working

## Steps to Reproduce the Problem

  1. Set Firewall Rules in Traffic Rules. 
Setup used for the Rule
config rule
        option family 'ipv6'
        list proto 'tcp'
        option src 'vpn'
        option dest 'lan'
        list dest_ip 'IPv6_Address'
        option dest_port '80'
        option target 'ACCEPT'

## Context (Environment)
A Webserver Should be reachable under IPv6, under IPv4 this isnt a Problem, only under IPv6.

## Specifications

  - OpenMPTCProuter version: 0.55.3
  - OpenMPTCProuter VPS version: 0.1017
  - OpenMPTCProuter platform: ESXi x86_64

Ysurac, 2020-10-03:
Can you check on the VPS, that rules are available in ```/etc/shorewall6/rules``` ?
If yes, I need the result of ```ip -6 r``` on the VPS.

Ysurac, 2020-10-03:
I found the issue, I will fix that.

Ysurac, 2020-10-03:
I need also the contain of ```/etc/shorewall6/params.vpn```

iamfreaky, 2020-10-03:
`/etc/shorewall6/rules`
`#
# Shorewall version 4.0 - Sample Rules File for two-interface configuration.
# Copyright (C) 2006-2014,2007 by the Shorewall Team
#
# This library is free software; you can redistribute it and/or
# modify it under the terms of the GNU Lesser General Public
# License as published by the Free Software Foundation; either
# version 2.1 of the License, or (at your option) any later version.
#
# See the file README.txt for further details.
#------------------------------------------------------------------------------
# For information about entries in this file, type "man shorewall-rules"
######################################################################################################################################################################################################
#ACTION         SOURCE          DEST            PROTO   DEST    SOURCE          ORIGINAL        RATE            USER/   MARK    CONNLIMIT       TIME            HEADERS         SWITCH          HELPER
#                                                       PORT    PORT(S)         DEST            LIMIT           GROUP
?SECTION ALL
?SECTION ESTABLISHED
?SECTION RELATED
?SECTION INVALID
?SECTION UNTRACKED
?SECTION NEW

#       Don't allow connection pickup from the net
#
Invalid(DROP)   net             all             tcp
#
#       Accept DNS connections from the firewall to the network
#
DNS(ACCEPT)     $FW             net
#
#       Allow Ping from/to the VPN
#
ACCEPT          vpn             $FW             ipv6-icmp
ACCEPT          vpn             net             ipv6-icmp
ACCEPT          $FW             vpn             ipv6-icmp
#
#       Allow Ping from the firewall to the network
#
ACCEPT          $FW             net             ipv6-icmp
#
# Drop Ping from the "bad" net zone.. and prevent your log from being flooded..
#
#DROP           net             $FW             ipv6-icmp
ACCEPT          net             $FW             ipv6-icmp
#
#       Accept connection from port > 65000 for shadowsocks and glorytun on the firewall
#
ACCEPT          net             $FW             tcp     65000-65535
ACCEPT          net             $FW             udp     65000-65535
#
#       Accept connection from SSH to the firewall
#
ACCEPT          net             $FW             tcp     65222
#
#       DHCP forward to the VPN from the firewall
#
ACCEPT          $FW             vpn             udp     53
ACCEPT          vpn             net             udp     53
#
#       Redirect all port from 1 to 64999 to the VPN client from the network
#
#DNAT           net             vpn:10.0.0.2    tcp     1-64999`

`ip -6 r`

`::1 dev lo proto kernel metric 256 pref medium
64:ff9b::/96 dev dsvpn0 proto kernel metric 256 pref medium
IPv6_ADDRESS::/56 via fe80::a00:2 dev omr-6in4-user0 metric 1024 pref medium
IPv6_ADDRESS::/48 dev enp1s0 proto kernel metric 256 pref medium
fe80::a00:0/126 dev omr-6in4-user0 proto kernel metric 256 pref medium
fe80::/64 dev tun1 proto kernel metric 256 pref medium
fe80::/64 dev tun0 proto kernel metric 256 pref medium
fe80::/64 dev mlvpn0 proto kernel metric 256 pref medium
fe80::/64 dev gt-udp-tun0 proto kernel metric 256 pref medium
fe80::/64 dev dsvpn0 proto kernel metric 256 pref medium
fe80::/64 dev gt-tun0 proto kernel metric 256 pref medium
fe80::/64 dev enp1s0 proto kernel metric 256 pref medium
fe80::/64 dev omr-6in4-user0 proto kernel metric 256 pref medium
default via fe80::a293:51ff:feb7:93d5 dev enp1s0 metric 1024 onlink pref medium`

`/etc/shorewall6/params.vpn`
`OMR_ADDR=fe80::a00:2/126`

iamfreaky, 2020-10-03:
//ACTION         SOURCE          DEST            PROTO   DEST    SOURCE          ORIGINAL        RATE            USER/   MARK    CONNLIMIT       TIME            HEADERS         SWITCH          HELPER
                                                       PORT    PORT(S)         DEST            LIMIT           GROUP
?SECTION ALL
?SECTION ESTABLISHED
?SECTION RELATED
?SECTION INVALID
?SECTION UNTRACKED
?SECTION NEW

       Don't allow connection pickup from the net

Invalid(DROP)   net             all             tcp

//       Accept DNS connections from the firewall to the network

DNS(ACCEPT)     $FW             net

//       Allow Ping from/to the VPN

ACCEPT          vpn             $FW             ipv6-icmp
ACCEPT          vpn             net             ipv6-icmp
ACCEPT          $FW             vpn             ipv6-icmp

//       Allow Ping from the firewall to the network

ACCEPT          $FW             net             ipv6-icmp

// Drop Ping from the "bad" net zone.. and prevent your log from being flooded..

//DROP           net             $FW             ipv6-icmp
ACCEPT          net             $FW             ipv6-icmp

//       Accept connection from port > 65000 for shadowsocks and glorytun on the firewall

ACCEPT          net             $FW             tcp     65000-65535
ACCEPT          net             $FW             udp     65000-65535

//      Accept connection from SSH to the firewall

ACCEPT          net             $FW             tcp     65222

//       DHCP forward to the VPN from the firewall

ACCEPT          $FW             vpn             udp     53
ACCEPT          vpn             net             udp     53

//      Redirect all port from 1 to 64999 to the VPN client from the network

//DNAT           net             vpn:10.0.0.2    tcp     1-64999

Ysurac, 2020-10-03:
Dirty paste but I don't see the redirection in shorewall config. It was saved ?

iamfreaky, 2020-10-03:
I hope that this is better. And no i cant see my Firewall rule from OpenMPTCP on the VPS side, 

[shorewall6-rules.txt](https://github.com/Ysurac/openmptcprouter/files/5321639/shorewall6-rules.txt)

Ysurac, 2020-10-03:
I need the result of ```uci show firewall``` from the router. And I will make some tests :)

iamfreaky, 2020-10-03:
[ucishowfirewall.txt](https://github.com/Ysurac/openmptcprouter/files/5321733/ucishowfirewall.txt)

iamfreaky, 2020-10-03:
[ucishowfirewall.txt](https://github.com/Ysurac/openmptcprouter/files/5321734/ucishowfirewall.txt)

Ysurac, 2020-10-04:
I will fix that in next release. I only add forward rules to firewall, but there is no forward rules for IPv6 :)

iamfreaky, 2020-10-04:
A great to know

closed 2021-01-08 by github-actions[bot]