Ysurac/openmptcprouter #1277: Problème avec un serveur iptv

skyfr54, 2020-10-04:
Bonjour, depuis que j'ai mis openmptcprouter, avec deuxième abonnement iptv le flux video ne passe plus (il accepte l'abonnement, affiche les listes de chaines et l'epg fonctionne aussi)...Le premier serveur qui marche est http://rv.therevenge.co:80 il fonctionne parfaitement avec iptvsmarters pro mais le deuxième qui est http://s.goldysat.com:8880. J'ai essayé une redirection de ports mais je ne sais pas si je l'ai fait correctement....Pouvez-vous m'aider à trouver une solution? Je pense que le port 8880 doit etre fermé.

kevin39, 2020-10-04:
Bonjour,
L'IPtv étant un service illégal je t'invite à supprimer les 2 lignes de ton post afin d'obtenir de l'aide sinon ton post risque d'être supprimé.

Ysurac, 2020-10-04:
Merci d'utiliser le template lors de l'ouverture d'une issue.
Sinon le flux est certainement bloqué par IP, donc il faut utiliser omr-bypass pour contourner.

Il faut également éviter de mettre des liens vers du contenu qui a de grande chance d'être illégal (même si le lien est je pense insuffisant pour avoir quelque chose).

closed 2020-10-04 by Ysurac