Ysurac/openmptcprouter #1278: Random name assignment of Usb tethering / LTE usb on disconnect and reconnect

openaspace, 2020-10-04:
## Expected Behavior
When a Usb device was set with specified name , on disconnect and reconnect need to maintain the same association name/device in the configuration. If a device is disconnected for example to change the SIM card, when reconnect need to maintain the same association hardware/name

## Current Behavior
If I set 2 LTE SIM modem device and I disconnect both and reconnect newly, in the preference the name of the saved device is mixed.. exchanging the name / device

## Possible Solution
I don't know. Can absolutely help generally, a led pulsation or other kind of signal to identify the device (attached) that was go off... For any reason.

## Steps to Reproduce the Problem

  1. Connect two LTE devices
  2. Assign name to each device
  3. Disconnect all LTE devices
4. Reconnect both devices
5. Name are randomly inverted in the configuration

## Context (Environment)
In a live streaming contest with a lot of LTE Sim cards, could easily occur that some SIM can go offline for some reason.. for example no signal.. or exhausted the prepay data included..
And is really important to identify which device is off, to detach, change SIM card and reattach to reconnect.

Ysurac, 2020-10-05:
Please use the all the issue template. I need versions used.

Mutanter, 2020-10-05:
As far as i can tell:
The interfaces are reboot stable, unless you change the configuration or switch / change the USB port(s) of the LTE stick(s).

pbapba512, 2020-10-06:
Unfortunatelly its not true, at least with the Huawei modems. I think we have to wait for the next release for the better 4G USB dongle handling.

openaspace, 2020-10-07:
I have decide to not use usb dongle. Only lte routers with external antenna.. and have all connection on the openmptcprouter managed as ethernet connection. It's more safe in critical stress live environment

openaspace, 2020-10-07:
raspberry it's no good choice , look for a system like this at 12volt: https://it.aliexpress.com/item/4000663839406.html?spm=a2g0o.cart.0.0.59cb3c00F6Xau8&mp=1

openaspace, 2020-10-07:
> As far as i can tell:
> The interfaces are reboot stable, unless you change the configuration or switch / change the USB port(s) of the LTE stick(s).
> Just "label" your sticks with its Gateway IP and you're good to go...

Can't be a rule to follow in high stress live applications .. where you can have 15 data SIM , all different operators with different APN settings .. if you use the SIM of another brand on the X USB DONGLE will not connect.. it's crazy for a live application of 8 hours.

Ysurac, 2020-10-07:
Only 8 WANs can be used simultaneously.
You can try latest beta, this may solve the issue.

openaspace, 2020-10-07:
> Only 8 WANs can be used simultaneously.
> You can try latest beta, this may solve the issue.

Thank you. 15 SIM over 8 hours , 4 sim every 3 hours to be changed

closed 2021-01-10 by github-actions[bot]