Ysurac/openmptcprouter #1280: Pb de débit avec le vps d'ovh...

skyfr54, 2020-10-05:
## Expected Behavior

Jai toujours mon souci de débit avec le openmptcprouter mais apparemment ça vient d'ovh car ils ont une option speedtest (nperf) sur leur site et pour une bande passante annoncée à 1gb, le speedtest ne dépasse pas les 350 Mb et down et en up (mesuré avec la livebox 5 2Gb) donc apparemment leur 1Gb c'est du pipeau et ce pour plus de 22€ par mois. Je vais donc changer de fournisseur de vps dans 15j. A ce propos est-ce que qqn en connait un qui fait du 1Gb réel, avec combien de datas (illimité serait le mieux, mais ovh me fournit 4To et ça a l'air de suffire, reste à voir, je suis un peu plus qu'à la moitié du mois), de l'ipv6, la localisation du serveur (le plus proche de Nancy) et bien sur le tarif....Merci d'avance! 
Ps : Je parle parfaitement anglais pour avoir vécu 17 ans aux usa mais beaucoup de personnes donnent leur issue en français et de plus une bonne partie de mon pb concerne la France et particulièrement le grand est comme location...merci de votre compréhension!

## Possible Solution

## Steps to Reproduce the Problem

  1.
  2.
  3.

## Context (Environment)

## Specifications

  - OpenMPTCProuter version: 
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter platform:

xabix, 2020-10-05:
Hello,

Moi je suis chez PulseHeberg et j'en suis content.

Ysurac, 2020-10-05:
Chez OVH, qu'elle est l'offre utilisée ? Il faudrait voir la charge CPU également.
Un test de téléchargement sur le VPS d'un fichier depuis un serveur ayant une connexion non surchargée à 10Gb/s peut aussi permettre de vérifier la vitesse réelle. Il y a des fichiers téléchargeable ici : https://testdebit.info/

closed 2021-01-09 by github-actions[bot]

Adorfer, 2021-01-10:
note: if you want to have good SIP and/or go into the scale of above 500MBit/s: You need a VPS with dedicated CPU cores (if not dedicated bare metal server). otherwise you will repeatedly find yourself in "becoming slow" problems after a burst of the first 1-5 minutes.