Ysurac/openmptcprouter #1281: Soucis concernant les soucis de connexion aux API de Discord, npmjs, etc..

AdrienDssy, 2020-10-05:
## Current Behavior
Il ce peux que la plupart du temps j'obtiens une erreur côté Discord, NPMJS, etc...
 request to https://discordapp.com/api/v7/failed, reason: getaddrinfo EAI_AGAIN discordapp.com
request to https://discord.com/api/v7/failed, reason: getaddrinfo EAI_AGAIN discord.com, etc...

Faut-il configurer un truc en particulier sur le routeur car sur mon home lab je n'ai pas de soucis point de vu réseau, j'ai tout vérifier.
## Specifications

- OpenMPTCProuter version: (0.55.3)
- OpenMPTCProuter VPS version:  (0.1017 5.4.52-mptcp)   
- OpenMPTCProuter platform: (x86_64)

Ysurac, 2020-10-05:
Ca ressemble à un soucis DNS. Donc il faut vérifier la configuration DNS au niveau du poste client et si jamais elle a été modifiée sur le routeur.

AdrienDssy, 2020-10-06:
Visiblement rien ne change et les DNS n'ont pas été touchés côté routeur 🤔🤔

kevin39, 2020-10-06:
Ouvre une invite de commande et tape : 
> nslookup discord.com
> nslookup discordapp.com

Tu obtiens quoi ?

AdrienDssy, 2020-10-08:
Serveur :   OpenMPTCProuter.lan
Address:  192.168.100.1

Réponse ne faisant pas autorité :
Nom :    discord.com
Addresses:  162.159.138.232
          162.159.135.232
          162.159.137.232
          162.159.136.232
          162.159.128.233

**nslookup discordapp.com**

Serveur :   OpenMPTCProuter.lan
Address:  192.168.100.1

Réponse ne faisant pas autorité :
Nom :    discordapp.com
Addresses:  162.159.130.233
          162.159.129.233
          162.159.133.233
          162.159.134.233
          162.159.135.233

AdrienDssy, 2020-10-08:
J'ai essayer avec des interfaces différentes (carte réseau interne, carte réseau externe et carte réseau PCI-E)

kevin39, 2020-10-08:
Tu as essayé d'un autre PC et/ou d'un autre navigateur ?

AdrienDssy, 2020-10-08:
ça à l'air de venir quelques part sur le routeur car mon père aussi à des soucis des fois 🤔

Ysurac, 2020-10-08:
Chez qui est le VPS ?
Que donne la charge CPU du VPS (via top ou htop par exemple) ?
Le soucis est sporadique ou toujours sur les mêmes sites ?

AdrienDssy, 2020-10-08:
Google, anciennement OVH mais le soucis restais le même
La charge CPU est d'environ 10%
Le soucis est sporadique

Ysurac, 2020-10-08:
Les connexions sont de quel type ? LTE, ADSL, FTTH, autre ?
Pas de wifi quelque part ?

Il faudrait aussi voir dans Etat->Journal Système ce qu'il peut y avoir.

AdrienDssy, 2020-10-08:
LTE,ADSL 

> Thu Oct  8 19:46:47 2020 daemon.err openvpn(wan3)[12415]: TLS Error: reading acknowledgement record from packet
Thu Oct  8 19:46:47 2020 daemon.err openvpn(wan3)[12415]: TLS Error: reading acknowledgement record from packet
Thu Oct  8 19:46:47 2020 daemon.err openvpn(wan3)[12415]: TLS Error: reading acknowledgement record from packet
Thu Oct  8 19:46:47 2020 daemon.err openvpn(wan3)[12415]: TLS Error: reading acknowledgement record from packet
Thu Oct  8 19:46:47 2020 daemon.err openvpn(wan3)[12415]: TLS Error: reading acknowledgement record from packet
Thu Oct  8 19:46:47 2020 daemon.err openvpn(wan3)[12415]: TLS Error: reading acknowledgement record from packet
Thu Oct  8 19:46:47 2020 daemon.err openvpn(wan3)[12415]: TLS Error: reading acknowledgement record from packet
Thu Oct  8 19:46:47 2020 daemon.err openvpn(wan3)[12415]: TLS Error: Unroutable control packet received from [AF_INET]54.38.188.175:65301 (si=3 op=P_ACK_V1)
Thu Oct  8 19:46:47 2020 daemon.err openvpn(wan3)[12415]: VERIFY ERROR: depth=1, error=self signed certificate in certificate chain: CN=ChangeMe
Thu Oct  8 19:46:47 2020 daemon.err openvpn(wan3)[12415]: OpenSSL: error:1416F086:SSL routines:tls_process_server_certificate:certificate verify failed
Thu Oct  8 19:46:47 2020 daemon.err openvpn(wan3)[12415]: TLS_ERROR: BIO read tls_read_plaintext error
Thu Oct  8 19:46:47 2020 daemon.err openvpn(wan3)[12415]: TLS Error: TLS object -> incoming plaintext read error
Thu Oct  8 19:46:47 2020 daemon.err openvpn(wan3)[12415]: TLS Error: TLS handshake failed
Thu Oct  8 19:46:47 2020 daemon.err openvpn(wan3)[12415]: TLS Error: Unroutable control packet received from [AF_INET]54.38.188.175:65301 (si=3 op=P_CONTROL_V1)
Thu Oct  8 19:46:49 2020 daemon.err openvpn(wan3)[12415]: TLS Error: TLS key negotiation failed to occur within 60 seconds (check your network connectivity)
Thu Oct  8 19:46:49 2020 daemon.err openvpn(wan3)[12415]: TLS Error: TLS handshake failed
Thu Oct  8 19:46:49 2020 daemon.notice openvpn(wan3)[12415]: SIGUSR1[soft,tls-error] received, process restarting
Thu Oct  8 19:46:50 2020 user.notice post-tracking-post-tracking: Tunnel up : Replace default route by 10.255.255.1 dev tun0
Thu Oct  8 19:46:50 2020 daemon.err /usr/bin/ss-redir[15648]: remote recv: Connection reset by peer
Thu Oct  8 19:46:52 2020 user.notice post-tracking-post-tracking: wan1 (eth4) switched up
Thu Oct  8 19:46:52 2020 user.notice post-tracking-post-tracking: Set server VPS (35.210.18.10) default route via 192.168.10.1
Thu Oct  8 19:46:52 2020 daemon.err /usr/bin/ss-redir[15647]: remote recv: Connection reset by peer
Thu Oct  8 19:46:52 2020 daemon.err /usr/bin/ss-redir[15648]: remote recv: Connection reset by peer
Thu Oct  8 19:46:52 2020 user.notice OMR-VPS: Can't get token, try later
Thu Oct  8 19:46:52 2020 daemon.err /usr/bin/ss-redir[15647]: remote recv: Connection reset by peer
Thu Oct  8 19:46:52 2020 daemon.err /usr/bin/ss-redir[15648]: remote recv: Connection reset by peer
Thu Oct  8 19:46:52 2020 daemon.err /usr/bin/ss-redir[15648]: remote recv: Connection reset by peer
Thu Oct  8 19:46:52 2020 daemon.err /usr/bin/ss-redir[15648]: remote recv: Connection reset by peer
Thu Oct  8 19:46:54 2020 user.notice post-tracking-post-tracking: Tunnel up disable use of ShadowSocks for UDP
Thu Oct  8 19:46:55 2020 user.notice Shadowsocks: Rules DOWN
Thu Oct  8 19:46:55 2020 user.notice Shadowsocks: Rules UP
Thu Oct  8 19:46:55 2020 user.notice Shadowsocks: Reload omr-bypass rules
Thu Oct  8 19:46:55 2020 user.notice omr-bypass: Starting OMR-ByPass...
Thu Oct  8 19:46:55 2020 user.notice omr-bypass: Reload dnsmasq...
Thu Oct  8 19:46:55 2020 daemon.info dnsmasq[29711]: read /etc/hosts - 4 addresses
Thu Oct  8 19:46:55 2020 daemon.info dnsmasq[29711]: read /tmp/hosts/dhcp.cfg01411c - 3 addresses
Thu Oct  8 19:46:55 2020 daemon.info dnsmasq-dhcp[29711]: read /etc/ethers - 0 addresses
Thu Oct  8 19:46:55 2020 user.notice omr-bypass: OMR-ByPass is running
Thu Oct  8 19:46:59 2020 daemon.warn openvpn(wan3)[12415]: WARNING: No server certificate verification method has been enabled.  See http://openvpn.net/howto.html#mitm for more info.
Thu Oct  8 19:46:59 2020 daemon.warn openvpn(wan3)[12415]: NOTE: the current --script-security setting may allow this configuration to call user-defined scripts
Thu Oct  8 19:46:59 2020 daemon.notice openvpn(wan3)[12415]: TCP/UDP: Preserving recently used remote address: [AF_INET]54.38.188.175:65301
Thu Oct  8 19:46:59 2020 daemon.notice openvpn(wan3)[12415]: UDP link local (bound): [AF_INET][undef]:65301
Thu Oct  8 19:46:59 2020 daemon.notice openvpn(wan3)[12415]: UDP link remote: [AF_INET]54.38.188.175:65301
Thu Oct  8 19:46:59 2020 user.notice post-tracking-post-tracking: Set server VPS (35.210.18.10) route via 192.168.10.1 metric 3
Thu Oct  8 19:46:59 2020 daemon.err openvpn(wan3)[12415]: TLS Error: Unroutable control packet received from [AF_INET]54.38.188.175:65301 (si=3 op=P_ACK_V1)
Thu Oct  8 19:47:01 2020 daemon.err /usr/bin/ss-redir[15647]: remote recv: Connection reset by peer
Thu Oct  8 19:47:01 2020 daemon.err openvpn(wan3)[12415]: TLS Error: Unroutable control packet received from [AF_INET]54.38.188.175:65301 (si=3 op=P_ACK_V1)
Thu Oct  8 19:47:03 2020 daemon.err openvpn(wan3)[12415]: VERIFY ERROR: depth=1, error=self signed certificate in certificate chain: CN=ChangeMe
[... cut ...]
Thu Oct  8 21:23:04 2020 daemon.err openvpn(wan3)[12415]: OpenSSL: error:1416F086:SSL routines:tls_process_server_certificate:certificate verify failed
Thu Oct  8 21:23:04 2020 daemon.err openvpn(wan3)[12415]: TLS_ERROR: BIO read tls_read_plaintext error
Thu Oct  8 21:23:04 2020 daemon.err openvpn(wan3)[12415]: TLS Error: TLS object -> incoming plaintext read error
Thu Oct  8 21:23:04 2020 daemon.err openvpn(wan3)[12415]: TLS Error: TLS handshake failed
Thu Oct  8 21:23:04 2020 daemon.notice openvpn(wan3)[12415]: SIGUSR1[soft,tls-error] received, process restarting

AdrienDssy, 2020-10-08:
JE tiens à ajouter que WAN3 n'existe pas

Ysurac, 2020-10-08:
Il y a donc un soucis de configuration. On dirait un MPTCP  over VPN qui a été activé peut être sur une interface supprimé, il va falloir que j'ajoute un check pour ce genre de cas.

AdrienDssy, 2020-10-08:
Comment je peux le retirer vu que c'est pas visible ?

Ysurac, 2020-10-08:
Il doit être visible dans VPN->OpenVPN

AdrienDssy, 2020-10-08:
Effectivement

AdrienDssy, 2020-10-08:
Mais en tout cas les soucis de timeout sont toujours présent

Ysurac, 2020-10-08:
Il faut regarder encore le journal système voir si il n'y a pas autre chose.

AdrienDssy, 2020-10-08:
Non rien de plus mise à part une requête DHCP pour un autre équipement

AdrienDssy, 2020-10-09:
Je suis retourner chez OVH et c'est bien mieux, j'ignore pourquoi ça merdais côté Google alors que tout les ports étaient ouvert.

ça me fait souvent ce soucis de VPN côté OVH alors que maintenant j'atteint les 30-70mbits (selon l'heure dans la journée)

>Fri Oct  9 14:39:45 2020 user.notice omr-tracker: Launching...
Fri Oct  9 14:39:45 2020 user.notice omr-tracker: Launched
Fri Oct  9 14:39:45 2020 daemon.info glorytun: starting glorytun vpn instance vpn
Fri Oct  9 14:39:45 2020 daemon.err glorytun[20155]: 54.38.188.175.65001: connected
Fri Oct  9 14:39:45 2020 daemon.err glorytun[20155]: 54.38.188.175.65001: key exchange failed
Fri Oct  9 14:39:45 2020 daemon.info glorytun[20155]: STOPPED tun0
Fri Oct  9 14:39:46 2020 user.notice omr-bypass: Starting OMR-ByPass...
Fri Oct  9 14:39:46 2020 user.notice omr-bypass: Restart dnsmasq...
Fri Oct  9 14:39:46 2020 daemon.info dnsmasq[23271]: exiting on receipt of SIGTERM
Fri Oct  9 14:39:46 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Fri Oct  9 14:39:46 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Fri Oct  9 14:39:46 2020 user.notice dnsmasq: Allowing RFC1918 responses for domain plex.direct
Fri Oct  9 14:39:46 2020 daemon.info dnsmasq[26298]: started, version 2.82 cachesize 150
Fri Oct  9 14:39:46 2020 daemon.info dnsmasq[26298]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC no-ID loop-detect inotify dumpfile
Fri Oct  9 14:39:46 2020 daemon.info dnsmasq-dhcp[26298]: DHCP, IP range 192.168.100.100 -- 192.168.100.249, lease time 12h
Fri Oct  9 14:39:46 2020 daemon.info dnsmasq[26298]: using only locally-known addresses for domain test
Fri Oct  9 14:39:46 2020 daemon.info dnsmasq[26298]: using only locally-known addresses for domain onion
Fri Oct  9 14:39:46 2020 daemon.info dnsmasq[26298]: using only locally-known addresses for domain localhost
Fri Oct  9 14:39:46 2020 daemon.info dnsmasq[26298]: using only locally-known addresses for domain local
Fri Oct  9 14:39:46 2020 daemon.info dnsmasq[26298]: using only locally-known addresses for domain invalid
Fri Oct  9 14:39:46 2020 daemon.info dnsmasq[26298]: using only locally-known addresses for domain bind
Fri Oct  9 14:39:46 2020 daemon.info dnsmasq[26298]: using nameserver 8.4.4.8#53
Fri Oct  9 14:39:46 2020 daemon.info dnsmasq[26298]: using nameserver 8.8.8.8#53
Fri Oct  9 14:39:46 2020 daemon.info dnsmasq[26298]: using nameserver 1.0.0.1#53
Fri Oct  9 14:39:46 2020 daemon.info dnsmasq[26298]: using nameserver 1.1.1.1#53
Fri Oct  9 14:39:46 2020 daemon.info dnsmasq[26298]: using only locally-known addresses for domain use-application-dns.net
Fri Oct  9 14:39:46 2020 daemon.info dnsmasq[26298]: using only locally-known addresses for domain lan
Fri Oct  9 14:39:46 2020 daemon.info dnsmasq[26298]: using 1 more local addresses
Fri Oct  9 14:39:46 2020 daemon.info dnsmasq[26298]: read /etc/hosts - 4 addresses
Fri Oct  9 14:39:46 2020 daemon.info dnsmasq[26298]: read /tmp/hosts/dhcp.cfg01411c - 3 addresses
Fri Oct  9 14:39:46 2020 daemon.info dnsmasq-dhcp[26298]: read /etc/ethers - 0 addresses
Fri Oct  9 14:39:46 2020 user.notice omr-bypass: OMR-ByPass is running
Fri Oct  9 14:39:47 2020 daemon.err glorytun[20155]: 54.38.188.175.65001: connected
Fri Oct  9 14:39:47 2020 daemon.err glorytun[20155]: 54.38.188.175.65001: key exchange failed
Fri Oct  9 14:39:47 2020 daemon.info glorytun[20155]: STOPPED tun0
Fri Oct  9 14:39:47 2020 daemon.err glorytun[20155]: 54.38.188.175.65001: connected
Fri Oct  9 14:39:48 2020 daemon.err glorytun[20155]: 54.38.188.175.65001: key exchange failed
Fri Oct  9 14:39:48 2020 daemon.info glorytun[20155]: STOPPED tun0
Fri Oct  9 14:39:48 2020 daemon.err glorytun[20155]: 54.38.188.175.65001: connected
Fri Oct  9 14:39:49 2020 daemon.err glorytun[20155]: 54.38.188.175.65001: key exchange failed
Fri Oct  9 14:39:49 2020 daemon.info glorytun[20155]: STOPPED tun0
Fri Oct  9 14:39:50 2020 daemon.err glorytun[20155]: 54.38.188.175.65001: connected
Fri Oct  9 14:39:51 2020 daemon.err glorytun[20155]: 54.38.188.175.65001: key exchange failed
Fri Oct  9 14:39:51 2020 daemon.info glorytun[20155]: STOPPED tun0
Fri Oct  9 14:39:59 2020 user.notice post-tracking-post-tracking: omrvpn down because gateway down
Fri Oct  9 14:39:59 2020 user.notice post-tracking-post-tracking: omrvpn down. Replace default route by 192.168.10.1 dev eth0
Fri Oct  9 14:40:02 2020 daemon.err glorytun[20155]: 54.38.188.175.65001: connected
Fri Oct  9 14:40:02 2020 daemon.err glorytun[20155]: 54.38.188.175.65001: key exchange failed
Fri Oct  9 14:40:02 2020 daemon.info glorytun[20155]: STOPPED tun0
Fri Oct  9 14:40:14 2020 daemon.err glorytun[20155]: 54.38.188.175.65001: connected
Fri Oct  9 14:40:14 2020 daemon.err glorytun[20155]: 54.38.188.175.65001: key exchange failed
Fri Oct  9 14:40:14 2020 daemon.info glorytun[20155]: STOPPED tun0
Fri Oct  9 14:40:14 2020 daemon.err glorytun[20155]: 54.38.188.175.65001: connected
Fri Oct  9 14:40:15 2020 daemon.err glorytun[20155]: 54.38.188.175.65001: key exchange failed
Fri Oct  9 14:40:15 2020 daemon.info glorytun[20155]: STOPPED tun0
Fri Oct  9 14:40:19 2020 daemon.err glorytun[20155]: 54.38.188.175.65001: connected
Fri Oct  9 14:40:20 2020 daemon.err glorytun[20155]: 54.38.188.175.65001: key exchange failed
Fri Oct  9 14:40:20 2020 daemon.info glorytun[20155]: STOPPED tun0
[... cut ...]
Fri Oct  9 14:41:40 2020 daemon.err glorytun[20155]: 54.38.188.175.65001: connected
Fri Oct  9 14:41:40 2020 daemon.err glorytun[20155]: 54.38.188.175.65001: key exchange failed
Fri Oct  9 14:41:40 2020 daemon.info glorytun[20155]: STOPPED tun0
Fri Oct  9 14:41:41 2020 daemon.err glorytun[20155]: 54.38.188.175.65001: connected
Fri Oct  9 14:41:41 2020 daemon.err glorytun[20155]: 54.38.188.175.65001: key exchange failed
Fri Oct  9 14:41:41 2020 daemon.info glorytun[20155]: STOPPED tun0
Fri Oct  9 14:41:42 2020 daemon.err glorytun[20155]: 54.38.188.175.65001: connected

closed 2020-10-09 by AdrienDssy