Ysurac/openmptcprouter #1282: Agréger des connexions sur iOS ou Android

JeedHome44, 2020-10-05:
Bonjour,

Je suis en train de me demander s'il est possible d'agréger les connexions de smartphones présents en wifi sur le LAN OMR?

Bien entendu, vu que ces forfaits ont une limite de data, y a-t-il moyen de récupérer par exemple sur le compte Bouygues de chaque numéro la data utilisée déjà et donc, par exemple, si le 20 du mois il en reste beaucoup, utiliser la data du mobile en plus et laisser 2 Go pour la fin du mois?
Il faudrait donc pouvoir vérifier en temps réel la data restante dés que le smartphone se connecte au réseau LAN.

Ça serait intéressant à long terme de pouvoir faire ça.

kevin39, 2020-10-06:
Bonjour,
Il est possible d'agréger les connexions uniquement mais tu fais un partage de connexion depuis les téléphones.

JeedHome44, 2020-10-06:
Ok merci mais il ne serait donc pas possible de récupérer la data utilisée directement sur le site de tel ou tel opérateur?
Avec Bouygues je peux suivre la data sur mon smartphone. Je pense donc qu'un simple petit script pourrait sûrement le faire sur OMR. Après, je ne sais pas programmer ça.

kevin39, 2020-10-06:
Techniquement tout est possible mais OMR ne fait pas ça :)

JeedHome44, 2020-10-06:
C'est dommage. Ça aurait donné une possibilité supplémentaire d'augmenter le nombre de clients internet sur OMR et booster encore plus le débit.
Juste pour tester, comment faut il faire sur OMR pour ajouter un troisième client wifi (mon iPhone en partage de connexion)?
Merci

JeedHome44, 2020-10-06:
> Techniquement tout est possible mais OMR ne fait pas ça :)

Comment configurer le wan3 svp pour un smartphone? Car ca me demande l'ip statique et la passerelle. Je dois mettre quoi?
Comme ca, je peux mettre ce wan3 en secours c'est bien ca?

Ysurac, 2020-10-06:
En DHCP pour un smartphone en tethering.

JeedHome44, 2020-10-06:
Je ne veux pas faire la même bêtise que l'autre fois.
Pour le coup la, je mets dans l'ip et la passerelle l'ip fixe de mon téléphone. C'est tout?

JeedHome44, 2020-10-06:
Comme la 5G va arriver doucement, à mon avis, les opérateurs vont sûrement passer la 4G en illimité comme quand la 4G est arrivé la 3G est passé illimité.
Ça me permettra de passer le smartphone du boulot, la tablette 4G du boulot, le smartphone de chaque enfin en wan sur OMR et additionner tous les débits... il faudra juste penser à laisser tous les appareils branché 😄

JeedHome44, 2020-10-06:
> En DHCP pour un smartphone en tethering.

J'ai mis l'ip fixe de mon iPhone dans le wan 3 pour l'ipv4 et la passerelle et ça ne marche pas

JeedHome44, 2020-10-19:
Bonjour,

Je pense avoir compris comment ajouter une connexion mobile via smartphone.

Il faut activer le partage de connexion du smartphone et connecter OMR en wifi dessus.

Par contre, il n'est donc pas possible de connecter OMR à plusieurs smartphones/tablettes en même temps pour augmenter en plus le débit?

Cette solution peut donc marcher avec un seul appareil? C'est bien ça?

closed 2021-01-22 by github-actions[bot]

babyss69, 2021-02-08:
> Bonjour,
>
> Je pense avoir compris comment ajouter une connexion mobile via smartphone.
>
> Il faut activer le partage de connexion du smartphone et connecter OMR en wifi dessus.
>
> Par contre, il n'est [...]

Bonjour, est ce que tu peux expliquer comment tu as fait en détail , merci beaucoup JeedHome44

JeedHome44, 2021-02-08:
Il suffit s'implement de mettre un Hub USB sur le Pi OMR et de connecter autant de dongle wifi que de point d'accès.
Un wifi ne peut pas se connecter à plusieurs points d'accès en même temps.

Mais au final, je ne fais pas ce partage car en mode partage de connexion, les smartphones n'utilisent plus le wifi de la maison.

Sur mes iPhone, il faut choisir entre partage de connexion et wifi.

babyss69, 2021-02-08:
Bonjour ,
Donc tu  rajoutes un hub usb sur ton pi , tu inserts des dongles wifi pas chères, tu scan dans la parties "réseau/ sans fil" les connexions en partage ,  tu t'y connectes et les rajoutes .
Mais dans la configuration "openmptcprouter/Assistant de configuration /" j'ai vu que cela rajoutais une interface "wwan"
mais dans le l'interface physique , et le protocole tu mets quoi ?

Mon objectif est d'utiliser des vieux mobiles portables 4G comme modem en activant le partage de connexion.