Ysurac/openmptcprouter #1284: Could this hardware CPU reach 1gbps upload?

openaspace, 2020-10-06:
Hi.
I need a very small hardware to install in a portable bag with CPU power to reach 1gbps upload bandwidth.

Could this mini computer reach this speed? 
https://www.amazon.it/dp/B07PS1KVD1/ref=cm_sw_r_cp_apa_fab_lScFFbK5Q136H

Thank you.

Adorfer, 2020-10-06:
in terms of CPU: certainly. 
in terms of NIC: certainly not.  

Despite beeing "full duplex", even "intel server grade eths" struggle with high UDP loads in full duplex. 
in other word's: if 950MBit/s+890MBit/s duplex is possible on TCP-loads, doing the same if (at least) 1/2 of it is UDP (GRE etc): hardly possible. 

Off course i welcome any feedback, but i would recommend hardware which is advertised for "pfsense firewalls", with several iNIC, on dedicated PCI lanes each.

openaspace, 2020-10-06:
Thank you.
For the lte router i'm looking for a device like this: https://e.huawei.com/en/products/enterprise-networking/routers/ar-agile/ar502-agile

openaspace, 2020-10-06:
https://alibaba.com/p-detail/detail-60810676302.html?__sceneInfo=%7B%22cacheTime%22%3A%221800000%22%2C%22type%22%3A%22appDetailShare%22%7D

This motherboard?

Adorfer, 2020-10-06:
This issue starts to look like a double of  #1108
https://github.com/Ysurac/openmptcprouter/issues/1108

closed 2020-10-06 by Ysurac