Ysurac/openmptcprouter #1285: Possibilité de diriger les services vers un autre wan si le 1er est hors service

JeedHome44, 2020-10-06:
Bonjour,

J'ai redirigé Netflix et Amazon vers la 4G pour que ca fonctionne.
Si je ne redirige pas Amazon, impossible de consulter les commentaires.

Serait-ce possible d'améliorer cette partie de OMR pour pouvoir faire une redirection plus complexe?

J'entends par là, dire pour Netflix, par exemple, de passer par le wan2 si tous les wan sont dispo. Mais, si wan2 coupé, passer par wan1.

Ce genre d'update est-il envisageable?
Ca éviterait de ne plus pouvoir utiliser Netflix si la 4G coupe mais pas l'ADSL. En gros, faire une règle de priorité.

Merci!
Je viens de publier une vidéo explication de OMR sur ma chaîne youtube pour parler un peu de ce super logiciel :)

Ysurac, 2020-10-07:
Actuellement quand aucun wan est définis ça utilise celui qui est définis en master et si il répond pas ça passe par un autre.

JeedHome44, 2020-10-07:
Même pour Netflix ou autre ayant une règle?

Ysurac, 2020-10-07:
oui

JeedHome44, 2020-10-07:
Je n'avais pas vu la mention dessous précisant que si rien n'est renseigné c'est le master qui est pris en premier

closed 2021-01-10 by github-actions[bot]