Ysurac/openmptcprouter #1286: Que se passe-t-il si VPS HS?

JeedHome44, 2020-10-07:
Bonjour,

Suite à ma vidéo YouTube:
https://youtu.be/-LhxArzUVXg

Une personne m'a dit qu'OMR ne marchait plus si le VPS était en maintenance.

N'y a-t-il pas moyen de mettre une règle simple qui vérifie l'état du VPS.

Si VPS Hors service, alors si WAN1 ok, alors passerelle WAN1 sinon passerelle WAN2

Comme ça, OMR envoie tout le flux vers le WAN1 si le VPS ne marche plus temporairement et si ce WAN1 est aussi hors service, vers le WAN2.

C'est un SI ALORS SINON facile à faire sur Box Domotique. Je pense qu'il peut être programmé aussi sur OMR non?

Ysurac, 2020-10-07:
C'est déjà le cas. Si le VPS ne répond pas alors par défaut on passe en load balancing sur les diverses connexions.

JeedHome44, 2020-10-07:
Super merci pour ta réponse rapide. Je vais en informer la personne qui ne doit pas le savoir! 😉

kevin39, 2020-10-07:
Et je rajouterai qu'on peut avoir un/des VPS de backup aussi :)

JeedHome44, 2020-10-07:
Après ça fait augmenter l'abonnement au moins mais c'est aussi une idée! Merci

closed 2020-10-07 by Ysurac