Ysurac/openmptcprouter #1287: Aggregation doesn't bounded

dettofatto, 2020-10-07:
## Expected Behavior
Two huawei 4g bounded, but the aggregation is limited to 1

## Specifications

  - OpenMPTCProuter version: 056b5, but also 055.3 (i've update for testing, but it is the same)
  - OpenMPTCProuter VPS version:  OpenMPTCProuter VPS 0.1018-test (before was 0.1017)
  - OpenMPTCProuter platform: pi4

omr-test-speed wan1 it is about  16.9M^C
omr-test-speed wan2 it is about 3M^C

Together is the same of wan1, but no data on wan2

Ysurac, 2020-10-07:
Can you make a screenshot of status page ?

Malaga82, 2020-10-07:
try to make MPTCP over VPN for the wan2, might be blocked by ISP.

dettofatto, 2020-10-07:
[image]

Ysurac, 2020-10-07:
What is the result of a check for both interface in Network->MPTCP and "MPTCP Support Check" tab ?

dettofatto, 2020-10-07:
[image]

[image]

Ysurac, 2020-10-07:
This seems to be ok.
What do you have in Network->MPTCP and MPTCP Fullmesh tab ?

dettofatto, 2020-10-07:
Index, Address-ID, Backup, IP-address, if-idx
IPv4, next v4-index: 7
2, 3, 0, 192.168.8.2, 17
3, 4, 0, 192.168.10.2, 18
IPv6, next v6-index: 1

Ysurac, 2020-10-07:
When you run ```omr-test-speed``` and you check MPTCP bandwidth graph, there is no traffic on wan2 ?

dettofatto, 2020-10-08:
No... only few kb...

Malaga82, 2020-10-08:
check with `omr-test-speed wan1` and `omr-test-speed wan2` if traffic is ok, then try my previous advice, make mptcp over vpn for the wan2 and check if aggregation works.

dettofatto, 2020-10-08:
No, it is the same.
If i do omr-test-speed wan2 it is ok, but omr-test-speed it use only wan1...

Malaga82, 2020-10-08:
Even with mptcp over vpn?
[image]

dettofatto, 2020-10-09:
I've seen that sometimes seems ok, with mptcp over vpn, but the speed is almost the same.

closed 2020-12-28 by dettofatto