Ysurac/openmptcprouter #1288: Bridge support for internal LAN

Mutanter, 2020-10-07:
## Expected Behavior

As internal LAN a bridge interface "br-lan" between "eth0" and "wlan0" should be displayed in the interfaces dropdown list.

## Current Behavior

Although in the LuCi overview the bridge interface is correctly configured and worked before the MPTCP endpoint was configured, the wizard only shows "eth0" or "wlan0" instead of "br-lan" interface after the first save. 

## Possible Solution

The bridge interface worked fine in previous versions, perhaps the scenario was missed during the revision of the interface mapping (at least in the current state or is intended?). (btw/ot: new Dashboard and interface mapping is great!)

## Steps to Reproduce the Problem

  1. Add wlan0 (RPI4 internal WiFi chip as AP) to LAN (bridge with eth0 - internal raspberry LAN port)
  2. Select "br-lan" in OMR Wizard as internal LAN, configure WAN1 + WAN2 and save/apply.
  3. Go back to OMR wizard and the "br-lan" internal LAN interface changed to eth0 and wlan0 is added to the dropdown. 

## Context (Environment)

I try to setup a bridge between eth0 and wlan0 as internal lan port to route LAN as well as WiFi traffic over MPTCP.
Furthermore, that allows to access and configure the OMR via WiFi as well as LAN. 

## Specifications

  - OpenMPTCProuter version: 
< openmptcprouter-v0.56beta5-8398a092-r0+14639-2a90d308c7-bcm27xx-bcm2711-rpi-4-ext4-factory.img >
  - OpenMPTCProuter VPS version: 
< OpenMPTCProuter VPS 0.1017 >
  - OpenMPTCProuter platform: 
< RPI4 >

Ysurac, 2020-10-08:
Should be fixed in latest beta.

closed 2020-10-08 by Ysurac

Mutanter, 2020-10-08:
working!!

Also referring to the interface, but more specifically to the USB tethering: 

**Steps to Reproduce the Problem**
For example, a USB tethering device is integrated into the MPTCP configuration as "eth3". 
Everything works as long as the phone remains plugged in. 
As soon as the phone is unplugged, the interface "eth3" is deleted and will be automatically replaced by "eth1" in the MPTCP configuration. This leads to a misconfiguration and the connection to the VPS endpoint is lost. 

Let me know if you want me to make a separate report for this.