Ysurac/openmptcprouter #1289: Private info scrubbed on the OpenMPTCP Status Page

sigkill, 2020-10-07:
**Is your enhancement proposal related to a problem? Please describe.**

The status page leaks private data - like a vps public host name if one is present, which would then provide a privacy concern. This data should be scrubbed like the IP address.
**Describe the solution you'd like**

Like the IP addresses, we should have at least a partial redaction of the host name from that field if possible.
**Describe alternatives you've considered**

None at the moment, although if taking screenshots, I would need to blur the hostname.
**Additional context**

kevin39, 2020-10-07:
Hi,
What is the problem ? All status page may leak data... Thats for private use.

In can you need to share your status page, there is an anonymize option at the bottom.

Adorfer, 2020-10-24:
"admin status pages showing configuration data": i would consider that as normal. 
I do NOT want that to be scrubbed. 
(We could discuss to make it an option. and i would then disable it. But without such an option, i would strongly oppose.)

sigkill, 2020-10-30:
I agree you would not want that scrubbed, however what I am speaking of is when you click the check mark on the bottom of the status page to enable the privacy on the status page - where the IP addresses get obfuscated. Unfortunately, if your VPS or VPN server has a hostname that resolves, that information is still present, so while the client info is shielded and obfuscated, the server is still present - I'm not really sure why this is a feature, unless it is used for helping resolve issues via screenshots, but my expectation would be that when you tick a box for privacy, all data would be private.

sigkill, 2020-10-30:
The "anonymize option" at the bottom is exactly what I am referring to. The hostname of the server you are using is still present, and in many cases I am certain it would be something that resolves to a DNS / IP etc.

Adorfer, 2020-10-30:
> check mark on the bottom of the status page

Not using the template produces a lot of questions. 
at least for me it's totally unclear which OMR version you are using 
Or what button you mean that does not work as expected (why didn't you put THAT in the issue itself?)

closed 2026-10-04 by Ysurac