Ysurac/openmptcprouter #1290: VoIP for incoming calls

Malaga82, 2020-10-08:
## Expected Behavior
voip should work out of the box 😄 
Both with udp and tcp transport
## Current Behavior
Unable to register with UDP transport provider.

## Possible Solution
for now switch for the box ( A510 IP ) to wan1 gateway, since i need provider DNS for register and domain resolution, but i use adguard dns for everything else.

## Steps to Reproduce the Problem

with siemens A510 IP i can't get registration from FTTC provider that give me the number.
The protocol used must be UDP, i can't use TCP.
And i have to use its dns for registration.
Sometimes it register but if i call the number it doesn't hook the call, as soon as i change the gateway to one of my wans, it register and if call the number it hooks the call fine.

## Specifications

  - OpenMPTCProuter version: v0.55.3 r0+14198-da98603597
  - OpenMPTCProuter VPS version: 0.1017 5.4.52-mptcp
  - OpenMPTCProuter platform: x86_64

Ysurac, 2020-10-09:
After many tests, this issue will be fixed in next release (and I think this will also solve some others issues).

Malaga82, 2020-10-09:
if you need some tests about this, count in me 😉

Malaga82, 2020-11-01:
i'm testing VOIP, will report if evrything is fine with UDP this week.

Malaga82, 2020-11-01:
After setting up the A510 IP gateway with openmptcp router IP it shows the VOIP connection as registred, but incoming calls are not working.
As soon as i set back gateway to wan1, then it starts to receive calls.
Connection is made to port 5060 and UDP protocol. Maybe i should try to change VPN mode?  I'm using glorytun UDP.

Ysurac, 2020-11-02:
I'm working on that. I can make call but not receive any.
This is not related to the VPN used.

Malaga82, 2020-11-02:
Ok, if you any test i'm here 👍

closed 2021-02-06 by github-actions[bot]