Ysurac/openmptcprouter #1291: "Can't get Public IP from ShadowSocks" that may cause a bad aggregation

thomasgx, 2020-10-08:
## Expected Behavior
A proper ShadowSocks status with a good aggregation is expected.

## Current Behavior

As you can see on the screenshots below, I have a mention "Can't get Public IP from ShadowSocks", but I don't know how to fix that.

It may cause an issue with aggreation, because, as you can see on the second screenshot, the sum of the 3 connexion is not good.
[image]

[image]

MPTCP config 
[image]

## Specifications

  - OpenMPTCProuter version: 0.55.3
  - OpenMPTCProuter VPS version: Version 0.1017 5.4.52-mptcp
  - OpenMPTCProuter platform: RPI2B with FirstHeberg. Same results on OVH it seems.

Ysurac, 2020-10-08:
Don't use mptcpdesync.
Check State->System log on OpenMPTCProuter.
Check that ss-server is running on the VPS: ```ps aux | grep ss```.

thomasgx, 2020-10-09:
Thanks @Ysurac 

I reverted to bbr.

I checked State->System Log, here it is for the last 12 hours, I'm not sure to see something faulty :

> Thu Oct  8 20:00:00 2020 cron.err crond[2364]: USER root pid 28201 cmd /etc/init.d/vnstat_backup backup
> Thu Oct  8 20:25:02 2020 daemon.err uhttpd[2677]: luci: accepted login on / for root from 192.168.100.155
> Thu Oct  8 20:58:41 2020 daemon.info dnsmasq-dhcp[936]: DHCPREQUEST(eth0) 192.168.100.155 b4:fb:e4:24:10:6c
> Thu Oct  8 20:58:41 2020 daemon.info dnsmasq-dhcp[936]: DHCPACK(eth0) 192.168.100.155 b4:fb:e4:24:10:6c USG
> Thu Oct  8 22:00:00 2020 cron.err crond[2364]: USER root pid 10393 cmd /etc/init.d/vnstat_backup backup
> Fri Oct  9 00:00:00 2020 cron.err crond[2364]: USER root pid 18014 cmd /etc/init.d/vnstat_backup backup
> Fri Oct  9 02:00:00 2020 cron.err crond[2364]: USER root pid 14397 cmd /etc/init.d/vnstat_backup backup
> Fri Oct  9 02:00:00 2020 cron.err crond[2364]: USER root pid 14398 cmd /etc/init.d/omr-bypass bypass_asn
> Fri Oct  9 02:29:35 2020 daemon.info dnsmasq-dhcp[936]: DHCPREQUEST(eth0) 192.168.100.155 b4:fb:e4:24:10:6c
> Fri Oct  9 02:29:35 2020 daemon.info dnsmasq-dhcp[936]: DHCPACK(eth0) 192.168.100.155 b4:fb:e4:24:10:6c USG
> Fri Oct  9 04:00:00 2020 cron.err crond[2364]: USER root pid 28821 cmd /etc/init.d/vnstat_backup backup
> Fri Oct  9 05:57:03 2020 authpriv.info dropbear[19498]: Child connection from 192.168.100.155:58745
> Fri Oct  9 05:57:04 2020 authpriv.notice dropbear[19498]: Password auth succeeded for 'root' from 192.168.100.155:58745
> Fri Oct  9 06:00:00 2020 cron.err crond[2364]: USER root pid 3860 cmd /etc/init.d/vnstat_backup backup
> Fri Oct  9 06:37:43 2020 daemon.info dnsmasq-dhcp[936]: DHCPREQUEST(eth0) 192.168.100.155 b4:fb:e4:24:10:6c
> Fri Oct  9 06:37:43 2020 daemon.info dnsmasq-dhcp[936]: DHCPACK(eth0) 192.168.100.155 b4:fb:e4:24:10:6c USG
> Fri Oct  9 07:04:37 2020 daemon.err uhttpd[2677]: luci: accepted login on /admin/network/network for root from 192.168.100.155
> Fri Oct  9 07:05:07 2020 user.notice omr-tracker: Launching...
> Fri Oct  9 07:05:07 2020 user.notice omr-tracker: Launched
> Fri Oct  9 07:05:08 2020 daemon.info glorytun: starting glorytun vpn instance vpn
> Fri Oct  9 07:05:09 2020 user.notice post-tracking-post-tracking: Tunnel up : Replace default route by 10.255.255.1 dev tun0

And finally, I ran ps aux | grep ss on the VPS
Same, I'm not sure to see something faulty here 👍 

[image]

What do you think ? 

Thanks again for your time and support :)

Ysurac, 2020-10-09:
Can you copy (without keys or/and VPS IP) the configuration available in System->OpenMPTCProuter, Show all settings tab ?

thomasgx, 2020-10-09:
Hi,
Here it is !

dhcp.@dnsmasq[0]=dnsmasq
dhcp.@dnsmasq[0].domainneeded='1'
dhcp.@dnsmasq[0].boguspriv='1'
dhcp.@dnsmasq[0].filterwin2k='0'
dhcp.@dnsmasq[0].localise_queries='1'
dhcp.@dnsmasq[0].rebind_localhost='1'
dhcp.@dnsmasq[0].local='/lan/'
dhcp.@dnsmasq[0].domain='lan'
dhcp.@dnsmasq[0].expandhosts='1'
dhcp.@dnsmasq[0].authoritative='1'
dhcp.@dnsmasq[0].readethers='1'
dhcp.@dnsmasq[0].leasefile='/tmp/dhcp.leases'
dhcp.@dnsmasq[0].resolvfile='/tmp/resolv.conf.d/resolv.conf.auto'
dhcp.@dnsmasq[0].nonwildcard='1'
dhcp.@dnsmasq[0].localservice='1'
dhcp.@dnsmasq[0].server='127.0.0.1#5353' '/lan/' '/use-application-dns.net/' '127.0.0.1#5353'
dhcp.@dnsmasq[0].noresolv='1'
dhcp.@dnsmasq[0].nonegcache='1'
dhcp.@dnsmasq[0].rebind_domain='plex.direct'
dhcp.@dnsmasq[0].rebind_protection='0'
dhcp.@dnsmasq[0].ipset='/upload.video.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/windowsupdate.com/omr_dscp-cs1,omr_dscp6-cs1' '/ttvnw.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/update.microsoft.com/omr_dscp-cs1,omr_dscp6-cs1' '/aiv-delivery.net/omr_dscp-cs4,omr_dscp6-cs4' '/googletagmanager.com/omr_dscp-cs2,omr_dscp6-cs2' '/googlevideo.com/omr_dscp-cs4,omr_dscp6-cs4' '/googleusercontent.com/omr_dscp-cs2,omr_dscp6-cs2' '/vevo.com/omr_dscp-cs4,omr_dscp6-cs4' '/google.com/omr_dscp-cs2,omr_dscp6-cs2' '/nflxvideo.net/omr_dscp-cs4,omr_dscp6-cs4' '/audio-fa.scdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/fbcdn.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs2,omr_dscp6-cs2' '/akamaihd.net/omr_dscp-cs2,omr_dscp6-cs2' '/s3.ll.dash.row.aiv-cdn.net/omr_dscp-cs4,omr_dscp6-cs4' '/whatsapp.net/omr_dscp-cs2,omr_dscp6-cs2' '/deezer.com/omr_dscp-cs4,omr_dscp6-cs4' '/whatsapp.com/omr_dscp-cs2,omr_dscp6-cs2' '/d25xi40x97liuc.cloudfront.net/omr_dscp-cs4,omr_dscp6-cs4' '/googleapis.com/omr_dscp-cs2,omr_dscp6-cs2' '/sndcdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/1e100.net/omr_dscp-cs2,omr_dscp6-cs2' '/last.fm/omr_dscp-cs4,omr_dscp6-cs4' '/hwcdn.net/omr_dscp-cs2,omr_dscp6-cs2' '/v.redd.it/omr_dscp-cs4,omr_dscp6-cs4' '/download.qq.com/omr_dscp-cs1,omr_dscp6-cs1' '/steamcontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gs2.ww.prod.dl.playstation.net/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropboxstatic.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox-dns.com/omr_dscp-cs1,omr_dscp6-cs1' '/log.getdropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive-thirdparty.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gvt1.com/omr_dscp-cs1,omr_dscp6-cs1' '/mmg-fna.whatsapp.net/omr_dscp-cs1,omr_dscp6-cs1' '/upload.youtube.com/omr_dscp-cs1,omr_dscp6-cs1'
dhcp.lan=dhcp
dhcp.lan.interface='lan'
dhcp.lan.start='100'
dhcp.lan.limit='150'
dhcp.lan.leasetime='12h'
dhcp.lan.force='1'
dhcp.lan.ra_slaac='1'
dhcp.lan.ra_flags='managed-config' 'other-config'
dhcp.wan=dhcp
dhcp.wan.interface='wan'
dhcp.wan.ignore='1'
dhcp.odhcpd=odhcpd
dhcp.odhcpd.maindhcp='0'
dhcp.odhcpd.leasefile='/tmp/hosts/odhcpd'
dhcp.odhcpd.leasetrigger='/usr/sbin/odhcpd-update'
dhcp.odhcpd.loglevel='4'
dropbear.@dropbear[0]=dropbear
dropbear.@dropbear[0].PasswordAuth='on'
dropbear.@dropbear[0].RootPasswordAuth='on'
dropbear.@dropbear[0].Port='22'
dscp.@classify[0]=classify
dscp.@classify[0].direction='both'
dscp.@classify[0].proto='icmp'
dscp.@classify[0].class='cs5'
dscp.@classify[0].comment='ICMP'
dscp.@classify[1]=classify
dscp.@classify[1].direction='both'
dscp.@classify[1].proto='udp'
dscp.@classify[1].class='cs5'
dscp.@classify[1].src_port='53,123,5353'
dscp.@classify[1].comment='DNS udp and NTP'
dscp.@classify[2]=classify
dscp.@classify[2].direction='both'
dscp.@classify[2].proto='tcp'
dscp.@classify[2].class='cs5'
dscp.@classify[2].src_port='53,5353'
dscp.@classify[2].comment='DNS tcp'
dscp.@classify[3]=classify
dscp.@classify[3].direction='both'
dscp.@classify[3].proto='tcp'
dscp.@classify[3].class='cs6'
dscp.@classify[3].dest_port='65001,65301,65011'
dscp.@classify[3].comment='OMR vpn'
dscp.@classify[4]=classify
dscp.@classify[4].direction='both'
dscp.@classify[4].proto='udp'
[... 1470 lines cut ...]
upnpd.@perm_rule[0].int_addr='0.0.0.0/0'
upnpd.@perm_rule[0].int_ports='1024-65535'
upnpd.@perm_rule[0].comment='Allow high ports'
upnpd.@perm_rule[1]=perm_rule
upnpd.@perm_rule[1].action='deny'
upnpd.@perm_rule[1].ext_ports='0-65535'
upnpd.@perm_rule[1].int_addr='0.0.0.0/0'
upnpd.@perm_rule[1].int_ports='0-65535'
upnpd.@perm_rule[1].comment='Default deny'
vnstat.@vnstat[0]=vnstat
vnstat.@vnstat[0].interface='wan1' 'wan2' 'wan3'
vnstat.@vnstat[0].backup='0'
wireless.radio0=wifi-device
wireless.radio0.type='mac80211'
wireless.radio0.channel='11'
wireless.radio0.hwmode='11g'
wireless.radio0.path='platform/soc/3f980000.usb/usb1/1-1/1-1.3/1-1.3:1.0'
wireless.radio0.htmode='HT20'
wireless.radio0.disabled='1'
wireless.default_radio0=wifi-iface
wireless.default_radio0.device='radio0'
wireless.default_radio0.network='lan'
wireless.default_radio0.mode='ap'
wireless.default_radio0.ssid='OpenWrt'
wireless.default_radio0.encryption='none'

Ysurac, 2020-10-09:
Can you check if the key in "shadowsocks-libev.sss0.key" that you removed above is the same in ```/etc/shadowsocks-libev/manager.json```on the VPS ?

thomasgx, 2020-10-09:
Indeed, it's the same, and indeed the "/" is replace by "_" on the VPS.

I don't have any clue, It was working previously, and now this message.
Also I restarsted several time the local router and the VPS. It didn't fix.

Last option : I can do a full reinstall on the local router and on the VPS ? Unless you have other ideas ?

h45h3ncryp73d, 2020-10-21:
I've just updated to latest RC1 on both OMR and VPS ...
When I opened /etc/shadowsocks-libev/manager.json on VPS on which I got "Can't get Public IP from ShadowSocks" error I've seen two things
1st) There was a double entry ie.
        "65101": "<KEY>",
        "65101": "<KEY>",
and 2nd)
        '+' was replaced by '-'  BUT '_' was replaced by '/' (reversed of what Ysurac wrote) in /etc/shadowsocks-libev/manager.json

I've checked my other VPS which is working fine and have seen that it has the same <KEY> string in /etc/shadowsocks-libev/manager.json and System -> OpenMPTCProuter -> Show all setting -> "shadowsocks-libev.sss0.key" with no replacements.

As soon as I deleted second entry and copied <KEY> as it was under System -> OpenMPTCProuter -> Show all setting -> "shadowsocks-libev.sss0.key" and rebooted VPS it started working just fine ...

thomasgx, 2020-10-28:
Thanks for your feedback. Unfortunately, I have only one entry.
And when I copy the same string in /etc/shadowsocks-libev/manager.json and System -> OpenMPTCProuter -> Show all setting -> "shadowsocks-libev.sss0.key" with no replacements, it doesn't work at all.

thomasgx, 2020-12-10:
Hi,

Today I updated VPS and Router to last version, and now, I don't have any message any more about Shadow Socks.

Still, the aggregation is bad....

georgedorn, 2021-01-19:
Just fought with this error myself.  What is baffling is that it works for a short while, then this error comes up.  Then I replaced the `-` with `+` in the json on the server, then _something_ restarted `ss-manager` (not me) and it suddenly started working again.  For now.

closed 2021-04-25 by github-actions[bot]