Ysurac/openmptcprouter #1292: VPN tunnel down and loss conection when speed test

jpossetti, 2020-10-08:
## Current Behavior

When ran speed test (speedtest.net or fast.com) after a few secconds of starts VPN goes down (VPN tunnel down), loss connection to VPS and some connections goes down too (cant ping gateway). Iam ussing three conections via ethernet concentred on a single switch on eth1 and my lan on eth0. After of this, router take about 1 minute to get up all conections and VPN tunnel again. Using Glorytun TCP and wizzards default settings. 

## Specifications

  - OpenMPTCProuter version: v0.55.3 r0+14198-da98603597
  - OpenMPTCProuter VPS version: 0.1017 5.4.52-mptcp
  - OpenMPTCProuter platform: RPI3 x86_64

Mutanter, 2020-10-09:
you should use a single gateway (eth / lan port) for each interface not a switch to "combine" them behind one line

jpossetti, 2020-10-09:
I don’t think that cause any problems I have dhcp disabled And diferents ip’s and gateways in each conn. By the way same problem using two different wan phisical interfaces

Ysurac, 2020-10-09:
You can try to enable SQM in the wizard and set max upload/download speed.
You can also check VPS and router CPU load average (using top for example).

Malaga82, 2020-10-10:
i found it this too, on some speedtests the tunnel went down.
I have 4G as secondary connection, don't know if it's a prevention from BTS in that specific band or something else.
vps side the cpu usage is low, near 20%, and so is the memory, near 400/500 of 2gigs, so it's not that.

Ysurac, 2020-10-10:
Same answer. Enable SQM.

Malaga82, 2020-10-10:
i enabled SQM but it doesn't seem to avoid this "issue".
i don't know OP, but i have openvpn over mptcp for one wan, maybe is that?

bobbykrin, 2020-12-07:
Me facing same issue.Sometimes it corrects automatically and sometimes need to restart the pi4.Enabling and disabling SQM
 does not help either.

bobbykrin, 2020-12-07:
[image]

bobbykrin, 2020-12-07:
It also changes the Server IP.

closed 2021-03-12 by github-actions[bot]