Ysurac/openmptcprouter #1294: can´t contact Server Admin Script (No result - No API script Answer)

rnunes77, 2020-10-09:
## Expected Behavior

Hello, when starting mptcprouter all arrows turn green, with the exception of Direct Output, which has a triangle with an exclamation mark, stating the following problem: can´t contact Server Admin Script (No result - No API script Answer)
## Current Behavior

The router is unable to contact the server
## Specifications

  - OpenMPTCProuter version: 0.55.1
  - OpenMPTCProuter VPS version: 0.1017
  - OpenMPTCProuter platform: Raspberry Pi 3 model B+

rnunes77, 2020-10-09:
I took a look at the logs and I have received the following report: user.notice OMR-VPS: Can't get token, try later (can ping server, no API server answer)

Ysurac, 2020-10-10:
The VPS was rebooted after script install ?

rnunes77, 2020-10-10:
Yes, I restarted the VPS. Only the ports 65101, 65001 and ICMP are open, Do I need to open any more ports? I'm using Glorytun and I just put the keys for that service. Do I need to put the MLVPN and OpenVPN keys?

Ysurac, 2020-10-10:
Port 65000 to 65535 needs to be open.
You can find all ports in https://github.com/Ysurac/openmptcprouter/wiki/Install-or-update-the-VPS
You need to put only API key, and all keys are retrieved from the API (when the API can be called).

rnunes77, 2020-10-10:
Perfect! Ysurac.. thank you. After the changes, everthing works fine!

closed 2020-10-10 by rnunes77

mstrolo, 2022-03-31:
can this be also related??
[image]

i get this all the rime

timbrookers, 2022-04-08:
> can this be also related?? [image]
> 
> i get this all the rime

mstrolo was your question answered, or did you find the solution?