Ysurac/openmptcprouter #1295: Error maj VPS Temporary failure resolving 'deb.debian.org'

DavidB33, 2020-10-10:
Bonjour et désolé mais mon anglais est trop mauvais pour expliquer mon problème, et mes compétences en Linux pas beaucoup mieux.

J'ai tardé à faire une mise à jour du VPS et lorsque que j’exécute le script pour mettre à jour : 
`wget -O - https://www.openmptcprouter.com/server/debian10-x86_64.sh | sh`

j'ai plusieurs messages indiquant un problème de résolution de domaine :
`Temporary failure resolving 'prod.debian.map.fastly.net' Temporary failure resolving 'deb.debian.org'`

voici mon source.list : 
`deb http://deb.debian.org/debian buster main contrib non-free`
`deb-src http://deb.debian.org/debian buster main contrib non-free`
`deb http://security.debian.org/ buster/updates main contrib non-free`
`deb-src http://security.debian.org/ buster/updates main contrib non-free`
`deb http://deb.debian.org/debian buster-updates main contrib non-free`
`deb-src http://deb.debian.org/debian buster-updates main contrib non-free`

Je reste donc bloqué à la version 0.1008 sur le VPS
Quelle pourrait être la raison de ce problème ?
J'ai testé l'opération hier et aujourd'hui mais rien y fait.
Seul OMR tourne sur ce VPS et je n'ai fait aucune manipulation autre que les maj via le script depuis sa création il y a plus de 6 mois.
D'avance merci pour votre aide.

## Specifications

  - OpenMPTCProuter version: 0.55.3
  - OpenMPTCProuter VPS version: 0.1008
  - OpenMPTCProuter platform: Debian 10.2 (Milkywan)

Ysurac, 2020-10-10:
Un soucis de DNS sur le VPS. Si c'est un soucis temporaire relancer le script, sinon il faut changer le DNS sur le VPS mais c'est indépendant d'OMR.

DavidB33, 2020-10-10:
Merci Ysurac, c’était bien ça. Je préférais demander ici avant de faire des bêtises et risquer de tout casser !

closed 2020-10-10 by DavidB33