Ysurac/openmptcprouter #1296: Resolving fastpipe.io VPS problems

vistman, 2020-10-10:
Hi,

I've purchased fastpipe.io VPS and of course only to later figure out it really does have some problem with OMR installation and it has been disqualified from the OMR compatibility list - as far as I could notice it is about not being able to properly set -mptcp kernel and after reboot it rejects SSH connections.
I've tried all possible solutions and installing Debian and Ubuntu but always with the same - unable to login result.
But since fastpipe.io has really good ping (at least for me) I'd like to contact them and try to make them fix the issue if possible, so if anyone (Ysurac) could pinpoint what should I exactly ask them to fix.
Thanks.

Ysurac, 2020-10-10:
Good question, I have no idea at all. When I tested them they are really too slow to make anything good. Sometimes it work but sometimes not.

vistman, 2020-10-10:
Thanks. If anyone has any idea let me know in the next 25 days while my subscription is valid and I'll contact them ...

desmando, 2020-10-27:
I don't know if this will fix your issue, but I was testing with a VPS that allows console access and went down the rabbit hole. What I ended up doing on a Debian 9 machine was adding "blacklist gsmi" to /etc/modprobe.d/blacklist.conf

h45h3ncryp73d, 2020-10-28:
Unfortuantely this didn't help me out ... Everything seems to be fine ... on Status page everything is checked but when I try to access any page connections get refused like fastpipe is blocking certain ports ...

$ ss -tulw
Netid  State   Recv-Q  Send-Q    Local Address:Port          Peer Address:Port
icmp6  UNCONN  0       0              *%mlvpn0:ipv6-icmp                *:*
udp    UNCONN  0       0             127.0.0.1:323                0.0.0.0:*
udp    UNCONN  0       0             127.0.0.1:8839               0.0.0.0:*
udp    UNCONN  0       0               0.0.0.0:65001              0.0.0.0:*
udp    UNCONN  0       0               0.0.0.0:65101              0.0.0.0:*
udp    UNCONN  0       0               0.0.0.0:65201              0.0.0.0:*
udp    UNCONN  0       0               0.0.0.0:65202              0.0.0.0:*
udp    UNCONN  0       0               0.0.0.0:65203              0.0.0.0:*
udp    UNCONN  0       0               0.0.0.0:65204              0.0.0.0:*
udp    UNCONN  0       0               0.0.0.0:65205              0.0.0.0:*
udp    UNCONN  0       0               0.0.0.0:65206              0.0.0.0:*
udp    UNCONN  0       0               0.0.0.0:65207              0.0.0.0:*
udp    UNCONN  0       0               0.0.0.0:65208              0.0.0.0:*
udp    UNCONN  0       0               0.0.0.0:65301              0.0.0.0:*
udp    UNCONN  0       0        0.0.0.0%mlvpn0:bootps             0.0.0.0:*
udp    UNCONN  0       0               0.0.0.0:bootpc             0.0.0.0:*
udp    UNCONN  0       0                 [::1]:323                   [::]:*
udp    UNCONN  0       0                  [::]:65101                 [::]:*
tcp    LISTEN  0       0               0.0.0.0:65222              0.0.0.0:*
tcp    LISTEN  0       0               0.0.0.0:65001              0.0.0.0:*
tcp    LISTEN  0       0               0.0.0.0:65101              0.0.0.0:*
tcp    LISTEN  0       0               0.0.0.0:65301              0.0.0.0:*
tcp    LISTEN  0       0               0.0.0.0:65401              0.0.0.0:*
tcp    LISTEN  0       0               0.0.0.0:65500              0.0.0.0:*
tcp    LISTEN  0       0                     *:65222                    *:*
tcp    LISTEN  0       0                     *:65101                    *:*
tcp    LISTEN  0       0                     *:65400                    *:*

closed 2021-01-31 by github-actions[bot]