Ysurac/openmptcprouter #1297: Kernel header / source needed to compile wifi drivers

ndaptardar, 2020-10-10:
I am fairly new to driver compilation, so pl bare with me. I have RPI4 and using 0.55.3 on it. I want to bond 3 wifi connections. I have 4 different USB wifi adapters but none of them are supported by openmptcprouter/openwrt. Pl suggest if I can download and use drivers from somewhere. The wifi adapters are TL-WN722N, Tenda W311MI, TL-WN725N, TL-WN822N.

Since I couldnt find readymade drivers, I am trying to compile them, but stuck at kernel headers/source. Where can I download source for 5.4.58 which is used in 0.55.3? 

Since couldnt find the headers/source, I tried to compile openmptcprouter with OMR-TARGET=rpi4. I was able to finish it with success, but then it chose latest .56beta9 and the opkg update is not working with this tag. Then I downloaded the .ipk from my build but it seems online packages are more than this build (e.g. git-http is not there in my build, but its available OMR package website 0.55.3)

Then I tried git clone --branch 0.55.3. But at the end, the img.gz says its 0.56beta9. Not sure if its just naming issue or it really made build with .56beta9

Could someone pl help?

Mutanter, 2020-10-11:
I can't help you compile the drivers / kernel, but I can tell you at least two USB WLAN adapters that are ready to work out of the box. Try your luck on: Alfa Network AWUS036NHA (Atheros) or MediaTek RT5370N (Ralink) Adapters. Also keep in mind that you need an isolated gateway for each WLAN connection.

ndaptardar, 2020-10-11:
> I can't help you compile the drivers / kernel, but I can tell you at least two USB WLAN adapters that are ready to work out of the box. Try your luck on: Alfa Network AWUS036NHA (Atheros) or [...]

Thanks!! I checked this in my region, too costly for me. Let me try compilation first. Thanks again.

Ysurac, 2020-10-11:
By default git is on develop branch, if you want current release you need to use master branch and use ```OMR_FEED_SRC="master" ./build.sh``` when compiling.
I think some of your adapters should work with kmod-rtl8xxxu but it's possible that only firmwares are missing from my image.

ndaptardar, 2020-10-11:
> By default git is on develop branch, if you want current release you need to use master branch and use `OMR_FEED_SRC="master" ./build.sh` when compiling.
> I think some of your adapters should work with kmod-rtl8xxxu but it's possible that only firmwares are missing from my image.

OK. Can I somehow get firmware for 0.55.3? Also, let me try with "master" and get back.

Mutanter, 2020-10-11:
Does the "short" working adapter authenticate and keep 1Mbits as RX/TX line in your WLAN tab?
If so, you can try a clean factory installation with only this adapter installed to the Pi.
While switching through different wlan interfaces I also observed strange phenomena where I don't know if the error is due to the adapters, OpenWRT or OpenMPTCP. A clean OS reinstallation (which probably resets the wlan interfaces completely) has often helped me with adapters that are actually recognized correctly but not working as expected (dmesg).

ndaptardar, 2020-10-11:
> Does the "short" working adapter authenticate and keep 1Mbits as RX/TX line in your WLAN tab?
> If so, you can try a clean factory installation with only this adapter installed to the Pi.
> While [...]

Actually when I tested those 4 adapters, I didnt do any changes at all ... all was factory installed. One of them worked very erratically.

Has anyone got any experience on compiling modules on RPI with openmptcprouter?

Ysurac, 2020-10-11:
Please, tell exactly what you want else it's not possible to help you.
I think you want to use some particular drivers, and to make a real support you will need to make real packages for them as any Linux distribution.

ndaptardar, 2020-10-12:
> Please, tell exactly what you want else it's not possible to help you.
> I think you want to use some particular drivers, and to make a real support you will need to make real packages for them as any Linux distribution.

Sure, here are the adapters and their chipsets.
TPLink TL-WN822N - RTL8192EU
TPLink TL-WN725N - RTL8188EUS
TPLink TL-WN722N - AR9271?
Tenda W311MI - RTL8188GU
Since they are not working out of box, am trying to download source and compile them. For example,
 $ git clone https://github.com/ZeeRooo/RTL8192EU.git
 $ cd RTL8192EU/
 $ make clean
 $ make <-- fails at this
 $ make install
 $ modprobe 8192eu

But for every such make, I get this error as I dont have kernel source/headers. And "opkg install linux-headers (with/without uname -r)" doesnt work since it doesnt find the package.
make[1]: *** /lib/modules/5.4.58/build: No such file or directory

Please help!

ndaptardar, 2020-10-15:
> Please, tell exactly what you want else it's not possible to help you.
> I think you want to use some particular drivers, and to make a real support you will need to make real packages for them as any Linux distribution.

Hi Ysurac, is it possible for you to give me quick pointers on how to build wifi driver/module? This is showstopper for me. Any help you provide greatly appreciated.

ndaptardar, 2020-11-20:
Finally, I bought Leoxsys LEO-HG150N 150Mbps Wireless USB Wi-Fi Adapter and it works great since it has RT5370. Thanks for all your support.

closed 2020-11-20 by ndaptardar

Adorfer, 2020-11-20:
> Hi Ysurac, is it possible for you to give me quick pointers on how to build wifi driver/module? This is showstopper for me. Any help you provide greatly appreciated.

this is really not an OMR-issue, but a general Openwrt question. 
I would consider asking in the forum.openwrt.org

mattodinaturas, 2022-08-18:
> The wifi adapters are TL-WN722N

FYI now for me this adapter works out of the box. OMR v0.58.5-r0+16336-b36068d35d