Ysurac/openmptcprouter #1298: How latency for WAN interfaces is calculated and gateway ping?

Malaga82, 2020-10-10:
I have 2 WAN'S, one of them is 4G and must be over vpn for aggregation to work.
In this interface i saw latency over 100ms usally, while ping from the wan i have something like 15/16ms.
from omr:
[image]
from lte h-monitor ping:
[image]
And sometimes it also shows "can't ping gateway", while ping from lte h-monitor is pinging correctly.
[image]
[image]
Just curiosity.

Ysurac, 2020-10-10:
Nothing special here. It's only an ICMP ping done for status page and only an information.

Malaga82, 2020-10-10:
The "No server ping" doesn't affect the bonding, right?

Ysurac, 2020-10-10:
Not at all.

closed 2020-10-10 by Ysurac