Ysurac/openmptcprouter #1300: Cannot login to GUI after update to OMR v0.56beta9

h45h3ncryp73d, 2020-10-12:
## Expected Behavior
Should be able to login into GUI with existing username/password combination.

## Current Behavior
I've just updated OMR to 0.56beta9 for 0.55.3 and after reboot I cannot login to GUI anymore ... I don't get error about wrong password it simply redirects me to login form again and again. SSH seems to be intact

## Possible Solution
Don't know. Could try to change root password via SSH ...

## Steps to Reproduce the Problem

  1. Upgrade from v0.55.3
  2. Try to log in.

## Specifications

  - OpenMPTCProuter version: 0.56beta9
  - OpenMPTCProuter VPS version: latest - can't tell as I cannot log in ...
  - OpenMPTCProuter platform: RPI4 x86_64

Ysurac, 2020-10-12:
Try to refresh browser cache.

h45h3ncryp73d, 2020-10-12:
> 
> 
> Try to refresh browser cache.

Funny ... That helped ... I tried in Chromium and Firefox and I had the same problem in both browsers ... So obviously clearing cache is necessary with this upgrade ... Thanks.

closed 2020-10-12 by h45h3ncryp73d

Adorfer, 2020-10-14:
BTW: sometimes even refresh does not help: 
in this case: open "private browser window". It happens with several RouterUI-Pages, but Openwrt and it's forks are not exception.