Ysurac/openmptcprouter #1301: Config wizard breaks omr-tracker config file.

pquan, 2020-10-13:
## Expected Behavior

Correct config file for omr-tracker (/etc/config/omr-tracker) when using the openmptcp wizard multiple times with different proxy settings.

## Current Behavior

The openmptcpwizard generates incorrect /etc/config/omr-tracker configuration. This prevents luci from displaying the omr tracker settings page and also omr-tracker will not start.

Diagnosis after breakage:
```
root@OpenMPTCProuter:/etc/config# uci show omr-tracker
uci: Parse error (section of different type overwrites prior section with same name) at line 54, byte 27
```
And the generated omr-tracker file:
```
config defaults 'defaults'
        option enabled '1'
        list hosts '4.2.2.1'
        list hosts '8.8.8.8'
        list hosts '8.8.4.4'
        list hosts '9.9.9.9'
        list hosts '1.1.1.1'
        list hosts '1.0.0.1'
        list hosts '1.2.4.8'
        list hosts '80.67.169.12'
        list hosts '80.67.169.40'
        list hosts '114.114.114.114'
        list hosts '114.114.115.115'
        option tries '2'
        option interval '2'
        option interval_tries '1'
        option type 'ping'
        option mail_alert '0'
        option timeout '10'

config proxy 'proxy'
        option enabled '1'
        list hosts '1.1.1.1'
        list hosts '1.0.0.1'
        list hosts '212.27.48.10'
        list hosts '198.27.92.1'
        list hosts '151.101.129.164'
        list hosts '77.88.55.77'
        list hosts '74.82.42.42'
        list hosts '176.103.130.130'
        option timeout '10'
        option tries '3'
        option interval_tries '1'
        option interval '5'

config server 'server'
        option enabled '1'
        option tries '3'
        option timeout '10'
        option interval '5'

config interface 'omrvpn'
        option type 'none'
        option tries '2'
        option interval '4'
        option mail_alert '0'
        option enabled '1'
        list hosts '4.2.2.1'
        list hosts '8.8.8.8'
        list hosts '1.1.1.1'
        option timeout '10'

config shadowsocks 'proxy'
        option enabled '0'
        option timeout '1'
        option tries '4'
        option interval '2'
        list hosts 'bing.com'
        list hosts 'google.com'
```

## Possible Solution

## Steps to Reproduce the Problem

  1. Fire up a clean openmptcp router
  2. Use the wizard to configure a router with "none" as the proxy
  3. Use the wizard to configure a router with "v2ray" as the proxy
  4. Use the wizard to configure a router with "shadowsocks" as the proxy
  5. check omr-tracker luci page and "uci show omr-tracker"

## Context (Environment)

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.56beta12-a8346e37 r0+14653
  - OpenMPTCProuter VPS version: testing. not relevant.
  - OpenMPTCProuter platform: rpi4

Ysurac, 2020-10-13:
The wizard doesn't generate or set any omr-tracker settings.
The problem in your config is this part: ```config shadowsocks 'proxy'``` but I have no idea where this is coming from, I never used domain names in the tracker.

pquan, 2020-10-13:
If this was "C" code, I'd say it came from usage of uninitialized strings. Not sure if the same applies to LUA (or whatever scripting language is used here).

Perhaps, some strings get written to the config, but they are not always set to valid values, or "zeroed" when their scope does not apply, like when there's no config/proxy set.

The issue is very reproducible after a few config wizard iterations.

Ysurac, 2020-10-13:
It was related to omr-tracker interface, this will be fixed in next beta.

closed 2020-10-13 by Ysurac