Ysurac/openmptcprouter #1302: VPN over OMR

ccmks, 2020-10-14:
I noticed that when I am using VPN behind my OMR, it doesn't do aggregation, such as OpenVPN, L2TP, SSTP, Softether, IPSec, and many more. It only pick one of active WAN connection to establish VPN connection. Do you know why it doesn't aggregate against VPN connection?

Thank you

ccmks, 2020-10-16:
Anyone? Any answer?

Ysurac, 2020-10-16:
This should work. If the VPN is over UDP, it use the VPN else it use shadowsocks. You can try another VPN.
You can also try snapshot release and try v2ray proxy.

ccmks, 2020-10-25:
Where can I download the snapshot? I am using Hyper-V based router. Then how to use v2ray proxy and how does it work?

Ysurac, 2020-10-25:
It's on the pinned issue: https://github.com/Ysurac/openmptcprouter/issues/1201
Then in wizard, advanced settings checkbox choose v2ray proxy.

Adorfer, 2020-10-25:
>  VPN behind my OMR, it doesn't do aggregation

i can not confirme. here it still works on 2 OMRs. 

> Anyone? Any answer?

Issues not using the templates makes answering feel like shooting in the fog. perhaps the reason why so little replies yet

closed 2021-01-30 by github-actions[bot]