Ysurac/openmptcprouter #1303: save & apply

ascomputer, 2020-10-16:
hello, is it possible to automatically execute the button procedure (save & apply in wizard) at the start on raspberri p4?
the reason is that if I change internet lines it doesn't work even if I restart raspberri p4 while if I click in the wizard save & apply everything works perfectly.
Thank you

Mutanter, 2020-10-16:
Does not sound like normal behavior. In the current beta, the interface mapping has been substantially revised and improved. In general OMR starts the last saved configuration and there are usually no problems at boot time. An automatic activation can also lead to problems if not all interfaces are physically connected (e.g. when the USB tethering device is unplugged). Furthermore, the file system of the Raspberry is not always writable, which is currently no problem as long as no configuration changes are forced.

ascomputer, 2020-10-16:
well it is obvious that before turning on the rasp with all the connections unfortunately it happens to me that I have to run save & apply in the wizard. in fact even the (backpacks) example (liveu only) before turning it on you need to insert the 4g dongles ... it's okay if it is not possible to do an automatic script after starting the rasp I do it manually thanks anyway

Mutanter, 2020-10-16:
The decision is not mine, I just wanted to prevent new problematics, which I already see in the idea proposal.
The router has to be configured once anyway, especially for the MPTCP entpoint (IP/Key). Afterwards I just start the device and everything connects automatically. That's why I said, I can't understand the behaviour you describe.

Ysurac, 2020-10-16:
Save&Apply in the wizard only restart network and all applications (mptcp config,shadowsocks, vpn,...), and it's already done at RPI4 start.
What you mean by "If I change internet line" ?
It's possible that a delay is needed at boot before running all scripts for dongle initialization ?

ascomputer, 2020-10-16:
> What you mean by "If I change internet line" ?

that if I have 2 dongles inserted and put the third dongle even if I restart it does not work if I then save & apply in the wizard all 3 work

> It's possible that a delay is needed at boot before running all scripts for dongle initialization ?
I can't tell you this I can tell you that I waited even 5 minutes but without success.
as i click on save & apply in wizard everything works perfectly

Ysurac, 2020-10-16:
You can check in state->system log what happens when you insert it.

ascomputer, 2020-10-16:
problem solved ... I re-installed omr on rasp pi4 and everything works if I restart it works fine if I turn off it works fine everything is ok bye and thanks

ascomputer, 2020-10-16:
what this message means
  ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err = -512
ieee80211 phy0: brcmf_fill_bss_param: Failed to get bss info (-512)
I see it in the system log
I ask this question because the wifi mestra was working it stopped and orm is continuing with other wan
thank you

closed 2021-01-19 by github-actions[bot]