Ysurac/openmptcprouter #1304: Bypassing the "lowest RTT selector" ?

Ch4rlus, 2020-10-17:
Hey ! 

First of all, thanks a lot for this project. I just started using it, and the all-in-one wizard made the whole configuration very easy :) 

I know that in the default configuration, OMR selects the connection with the lowest latency, and uses it until it is at its maximum capacity (then it aggregates with the others). 

In my case, I only use two LTE connections. The first one is unlimited, with 35ms average latency, and 60mbps download. The second one is much better (19ms latency, 250mbps download, ..) but limited to 500GB/month. 

Obviously, the default configuration selects in priority the best connection (LTE2), and therefore uses the other one only when I exceed 250mbps download speed (so quite rarely...).

Is there any way to force the connection to use "first" (and therefore bypass the "lowest RTT selector") ? I'd like OMR to use my LTE1 as long as I'm consuming less than 60mbps, and only start aggregating LTE2 if I need more throughput.

I've seen a lot of settings related to the "master" configuration, but I can't really understand if this has anything to do with what I'm looking for.

Thanks !

ascomputer, 2020-10-17:
hello in network ----> mptcp ---> setting try to put in the field Multipath TCP scheduler round-robin should use the 2 lte simultaneously.

Ysurac, 2020-10-17:
It's not possible to bypass the "lower RTT selector".
Master is only the default gateway or the gateway that is used in priority in balancing mode for initial connection only.

The only solution would be to add artificial latency to LTE2 using ```tc``` and this may use it after LTE1.

Ch4rlus, 2020-10-17:
I was thinking about tricking the system by using tc. That's a kind of dirty hack, but that should do the trick. 

Thanks a lot for the reply & the clarification :)

closed 2021-01-20 by github-actions[bot]