Ysurac/openmptcprouter #1305: OMR et Pi-Hole

JeedHome44, 2020-10-17:
Bonjour,

Savez vous s'il est possible d'utiliser Pi-Hole avec OMR ou s'il peut être intégré dans OMR à l'avenir?

Merci

Ysurac, 2020-10-17:
https://github.com/Ysurac/openmptcprouter/wiki/Pi-hole

JeedHome44, 2020-10-17:
Merci pour ta réponse ultra rapide!

JeedHome44, 2020-10-17:
Si je comprends bien, il suffit de lancer le script sur le VPS en SSH et openmptcprouteur fera le reste?
Je préfère être sur de ce que je fais et ne faire aucune bêtise...

Ysurac, 2020-10-17:
normalement oui

JeedHome44, 2020-10-17:
Par contre, une personne m'a parlé de Pi-Hole en me disant que s'il est chez lui en wifi et qu'il interroge son DNS, ca passe par l'IP locale. Par contre, s'il coupe le wifi de son smartphone et qu'il consulte le même DNS, ca passe par le DNS directement.
On peut utiliser cette fonction en l'installant sur le VPS?

Ysurac, 2020-10-17:
oui c'est le fonctionnement normal du smartphone, pas lié à OpenMPTCProuter.

JeedHome44, 2020-10-17:
Ok merci pour tes réponses rapide.
Donc pour résumer, j'ai juste à ouvrir une connexion SSH avec le VPS. Je lance le script de ton lien dessus. Je mets ok sur les étapes d'interface et IP vu que OMR s'en occupera. J'active l'interface de gestion web du Pi-hole et ensuite sur OMR je fais juste sauvegarder et appliquer sur la page Assistant. C'est bien ca?

Ysurac, 2020-10-17:
oui, comme c'est mis dans la doc et lors du lancement du script.

JeedHome44, 2020-10-17:
Mais c'est marqué "after VPS script" quel script VPS?

JeedHome44, 2020-10-17:
Et comment ne pas appliquer les règles de parefeu Pi-hole? directement quand on l'installe?

Ysurac, 2020-10-17:
Il n'y a qu'un script pour la partie VPS.
Lors de l'installation Pi-Hole demande, faut refuser simplement.

JeedHome44, 2020-10-17:
ah tu veux dire par là qu'il faut débord installer le script d'OMR sur le VPS avant de lancer ce script pour Pi-Hole?
Vu que le VPS marche déjà avec OMR, j'ai donc juste à faire le script de ton lien c'est ca?
Désolé pour toutes les questions mais je préfère savoir ou je vais avant de me lancer.
J'aime pas partir tête baisser.

Ysurac, 2020-10-17:
oui.
Mais toute façon si ça fonctionne pas, tu fais une réinstallation depuis 0 du VPS, y'a peu de risque...

JeedHome44, 2020-10-17:
ok si le VPS passe hors ligne ca fera du load balancing c'est ca? comme tu m'avais déjà dit dans un autre sujet.

JeedHome44, 2020-10-17:
voilà c'est fait. pendant l'installation je n'ai rien vu sur le parefeu par contre. le vps à l'air de marcher. je vais vérifier sur l'interface web de Pi-hole pour voir si le parefeu est bien désactivé

JeedHome44, 2020-10-17:
j'ai accès à l'interface PiHole avec 10.255.255.1 sur le réseau local. Mais comment faire une redirection de port pour y avoir accès avec l'ip publique du VPS et ainsi, avec accès à l'interface depuis l'extérieure? je mets une redirection de port sur le LAN mais ca ne marche pas. Ca doit être particulier je présume.

Ysurac, 2020-10-18:
Il faut sur le VPS ouvrir le port dans shorewall, donc éditer ```/etc/shorewall/rules``` copier une régle "ACCEPT" certainement déjà présente et mettre le bon port (normalement 80). Ensuite relancer shorewall: ```/etc/init.d/shorewall reload```

JeedHome44, 2020-10-18:
bonjour,
je suis dans /etc/shorewall/rules mais je ne vois pas ce qu'il faut copier dedans.
est-ce possible de me le dire?
mon but est d'ouvrir le port 10100 par exemple.
merci d'avance

P;S: par contre, Pi-hole ne filtre rien du tout. j'ai pourtant rien touché dans les configurations.

JeedHome44, 2020-10-18:
Je dois copier ca et remplacer le port 65101 par 80 c'est ca?
ACCEPT          net             $FW             tcp     65101   # OMR openmptcprouter open shadowsocks port tcp
ACCEPT          net             $FW             udp     65101   # OMR openmptcprouter open shadowsocks port udp

Ysurac, 2020-10-18:
Juste ```ACCEPT net $FW tcp 80``` pas besoin d'UDP.

JeedHome44, 2020-10-18:
super ca marche en mettant juste l'ip publique du vps dans le navigateur. je tombe sur Pi-hole.
Reste plus qu'à trouver pourquoi il affiche 0 filtre... j'ai pourtant testé d'aller sur plusieurs sites avec plusieurs appareils du réseau qui utilisent bien le VPS

Ysurac, 2020-10-18:
Je n'ai pas testé plus que ça. Mais il faut prendre en compte que la seule IP que Pi-Hole peut voir est l'IP du VPN côté OpenMPTCProuter, donc 10.255.255.1 si Glorytun TCP est utilisé. Et il ne verra pas toute les requêtes car le routeur a un cache DNS.

JeedHome44, 2020-10-18:
Pi-hole est en écoute sur tun0.
Chest peut être ça le problème?

JeedHome44, 2020-10-18:
Question hors sujet:
Serait-il possible que vous mettiez une icône pour OMR quand on ajoute le lien sur l'écran d'accueil iPhone?
Pour Jeedom, Home Assistant, Octoprint,Pi-Hole... ça affiche une belle icône mais pas OMR.
Je pense que c'est dans le code de la page que ça se trouve non?
Merci pour tout!

Ysurac, 2020-10-18:
C'est normal que Pi-Hole écoute sur tun0.
Je ne connais pas l'iPhone, donc je peux pas dire mais je vais regarder si je trouve quelque chose sur ça.

JeedHome44, 2020-10-18:
En fait c'est pas forcément iPhone. C'est Android aussi.
Un ami m'a envoyé ça, je pense que ça pourrait t'aider:

Il faut paramétrer une favicon dans le code. + d'info ici : https://www.alsacreations.com/astuce/lire/1080-apple-touch-icon-iphone-ipad-ipod-icone.html

Ysurac, 2020-10-18:
oui je viens d'ajouter ça dans le thème, ça sera donc la prochaine version.

JeedHome44, 2020-10-18:
Super!
Un grand merci pour ta réactivité!

JeedHome44, 2020-10-24:
Salut,

J'ai fait pas mal de tests sur Pi-hole et ça ne bloque rien du tout sur le VPS. Je ne vois donc pas l'intérêt 😞

Ysurac, 2020-10-24:
Si Pi-Hole est bien configuré, il fonctionne...
Il faut vérifier dans Réseau->DHCP&DNS que ça redirige bien vers le VPS.
Après Pi-Hole ne bloque rien du tout, il s'occupe que de la résolution DNS.

JeedHome44, 2020-10-24:
Quel configuration dois-je vérifier dans Réseau/DHCP?
Merci

Ysurac, 2020-10-24:
Si il y a l'IP de la partie serveur du VPN dans transmissions DNS, donc 10.255.255.1 si c'est glorytun TCP en VPN.

JeedHome44, 2020-10-24:
Oui il y a 10.255.255.1#53

Ysurac, 2020-10-24:
donc les requêtes DNS sont bien redirigés vers pi-hole (si il n'y a pas d'autre IP), à partir du moment où l'IP d'OpenMPTCProuter est utilisé en tant que DNS sur les postes client.

JeedHome44, 2020-10-24:
Avant il y en a 3 autres:

127.0.0.1#5353
/lan/
/use-application-dsn.net/
10.255.255.1#53

Ysurac, 2020-10-24:
vire le 127.0.0.1#5353

JeedHome44, 2020-10-24:
Ça ne va pas mettre le bordel sur OMR? Il est là le problème?

Ysurac, 2020-10-24:
ça enverra toute les requêtes vers pi-hole sur le VPS et c'est ce que tu souhaites.

JeedHome44, 2020-10-24:
Je viens de le virer et je ne peux plus accéder à aucun site web sur mon ordinateur

Ysurac, 2020-10-24:
vérifie si pi-hole est bien en route sur le VPS donc.

JeedHome44, 2020-10-24:
Si je remets 127.0.0.1#5353 les sites remarchent

Ysurac, 2020-10-24:
ça parait logique

JeedHome44, 2020-10-24:
Oui je suis dessus en même temps et il est actif. Il y a donc un truc qui bloque on dirait c'est pour ça que je n'ai rien sur Pi hole, rien ne passe dessus si je comprends bien.

JeedHome44, 2020-10-24:
J'y ai bien accès en faisant 10.255.255.1/admin/

JeedHome44, 2020-10-24:
J'ai choisi Quad9 (filtré,DNSSEC) sur Pi Home en Serveur DNS en amont.
Est-ce ça le problème?

Ysurac, 2020-10-24:
non

JeedHome44, 2020-10-24:
Le comportement de l'écoute de Pi-hole il faut le mettre sur quel niveau? La je l'ai mis sur "Écouter uniquement sur l'interface tun0"

Ysurac, 2020-10-24:
Si ça a été modifié après l'installation et que ça a donc modifié la config, c'est pas bon.

JeedHome44, 2020-10-24:
Vu que ça ne marcherait pas, je me souviens avoir essayé de changer ça mais je ne sais plus sur quelle option c'était.

JeedHome44, 2020-10-24:
Ah je viens de mettre "Écouter sur toutes les interfaces" et la Ca affiche enfin des requêtes sur Pi Hole. Il faut donc mettre sur toutes les interfaces. 😉

JeedHome44, 2020-10-24:
Pi-Hole détecte que ça vient du client 10.255.255.2
C'est donc normal?

J'ai enfin compris ce qui n'allait pas!

Encore merci pour tout! C'est donc fonctionnel maintenant 😉

Ysurac, 2020-10-24:
oui normal.

JeedHome44, 2020-10-24:
Encore merci et bonne soirée

closed 2021-01-29 by github-actions[bot]