Ysurac/openmptcprouter #1306: Video Playback

ByteArrayException, 2020-10-18:
## Expected Behavior
I would like to stream my rtmp stream through omr to get more bandwidth via different uplinks.

## Current Behavior
The stream stops or freezes almost every second, so that no proper playback is possible.

Anyone else expected this issue/behavior? Or have already an workaround/fix for it?

## Specifications

  - OpenMPTCProuter version: v0.55.3
  - OpenMPTCProuter VPS version: 0.1017
  - OpenMPTCProuter platform: x86_64

Mutanter, 2020-10-18:
Sounds like a misconfiguration. 
Can you post the last 10-20 entries of your system-log (excluding any public IPs)?

ByteArrayException, 2020-10-18:
At the last test, I unfortunately did not make a log. I will test it again soon, and then attach a log here.

ByteArrayException, 2020-11-03:
I can't reproduce the bug at the moment, otherwise I'll open a new issue (with log excerpt). Thanks for the help anyway.

closed 2020-11-03 by ByteArrayException