Ysurac/openmptcprouter #1307: port forward to Wan instead of vpn

brazzo97, 2020-10-18:
[image]
[image]
i need to open a port on the wan not using the vpn , but i am not able to , what i am doing wrong ?

Ysurac, 2020-10-18:
This seems correct, you can try to use tcpdump to check if you see any traffic.

brazzo97, 2020-10-18:
mmmh can you tell me how to ?

Ysurac, 2020-10-18:
```tcpdump -i <interface> -vvv``` on OpenMPTCProuter via ssh.

cayenne17, 2021-01-04:
I have the same problem. To work around the problem, I have created OMR bypass rules on the device where I have NAT/PAT rules on a WAN interface without going through the VPS

The problem comes from the fact that OMR sends all packets to internet destination in VPN (even return packets coming from WAN).

cayenne17, 2021-04-11:
I have just tested and the problem is alway present on version 0.58beta5

gsxryan, 2021-06-24:
This does not help solve the problem, but may i recommend not forwarding RDP to your NAS?  This is extremely risky and puts your data at great risk against ransomware.  Obfuscating the port can help against some common attacks, but it is not safe.  Please check into forwarding SSH, or a VPN tunnel instead.

cayenne17, 2021-09-23:
problème toujours d'actualité

fverryden, 2021-11-01:
Pb Toujours d'actualité

cayenne17, 2022-02-04:
Pb Toujours d'actualité