Ysurac/openmptcprouter #1310: By-Pass Amazon Prime Video

JeedHome44, 2020-10-22:
Bonjour,

J'ai trouvé un post pour mettre les règles pour que Netflix passe par un wan specifique.
Ce soir, je souhaite regarder Amazon Prime Vidéo et ça me dit: "Internet Connectivity Problem" mais je peux regarder Netflix ou autre.
C'est donc un problème de règle.

Dans le service By-Pass, j'ai mis tout en bas Prime Video mais ça n'a pas l'air de suffire.

J'ai donc mis aussi le site primevideo.com et primevideo.fr en règle et ça ne marche pas non plus.

Que dois-je faire comme règle pour qu'Amazon Prime video marche avec OMR?

Merci

Specifications

  - OpenMPTCProuter version: 0.55.3
  - OpenMPTCProuter VPS version: 0.1017 5.4.52-mptcp
  - OpenMPTCProuter platform: Pi 4B 2 Go

kevin39, 2020-10-22:
Salut,
J'ai pas la réponse exacte, il doit y avoir plusieurs tranches d'IP, c'est potentiellement compliqué.

Prime Video, tu y accèdes depuis où ? Ta TV ? Le plus simple je pense serait de faire un bypass par l'IP du device concerné (la tv), au moins t'es sûr que ça sortira sur le WAN désiré (mais tous les services de ta TV seront concernés évidemment).

Ysurac, 2020-10-22:
Chez qui est le VPS ? Amazon Prime est rarement bloqué.

JeedHome44, 2020-10-22:
Salut! Oui pas bête c'est une bonne idée de mettre la TV en by-pass et sans renseigner le wan, ça va donc diriger vers le maître et si maître HS, vers le second. Mais j'aimerai aussi faire ça pour les tablettes, smartphones et pc de la maison comme j'ai fait pour Netflix et Disney+.

JeedHome44, 2020-10-22:
> Chez qui est le VPS ? Amazon Prime est rarement bloqué.

Chez OVH.
Ça charge longtemps et ensuite ça dit que la connexion n'est pas bonne.
Pourtant Netflix marche.

Quels noms de domaine dois-je mettre sur le by-pass pour Amazon Prime Vidéo?

Je n'ai trouvé aucun sujet dans les sujets clos pour ça. J'ai par contre trouvé pour Disney+ et Netflix.

Ysurac, 2020-10-22:
Il y a un soucis dans le bypass protocol de la release actuelle. Ca sera corrigé dans la prochaine version qui sera dispo lundi normalement.

Sinon les domaines normalement : 
s3.ll.dash.row.aiv-cdn.net
s3-dub.cf.dash.row.aiv-cdn.net
dmqdd6hw24ucf.cloudfront.net
d25xi40x97liuc.cloudfront.net
.aiv-delivery.net
.aiv-cdn.net
1s3.lvlt.dash.us.aiv-cdn.net
c.footprint.net
.s.loris.llnwd.net
atv-ext.amazon.com
c.media-amazon.com

JeedHome44, 2020-10-22:
Tout ça pour Amazon Prime vidéo? Ça en fait un paquet. En fait, les services à la fin de la page de by-pass regroupent tous ces domain c'est ça?
Je vais tester de tous les ajouter pour voir!
Ah lundi mise à jour? J'ai jamais fait. Ça se fait tout seul ou pas?

rjhollan, 2020-10-27:
Try including these, then doing an "/etc/init.d/omr-bypass reload" via ssh.  I've found that reloading the config via SSH works when sometimes the GUI doesn't.

https://www.netify.ai/resources/applications/amazon-video

Ysurac, 2020-10-27:
Protocol section in OMR-ByPass is working on new 0.56 release.
It's possible that another domain is needed, in this case I can add it.

JeedHome44, 2020-10-27:
Bonjour,

Mise à jour faites à l'instant!
Déjà l'icône iOS marche très bien!

Merci!

Je vais tester cet après midi pour Amazon Prime 😉

rjhollan, 2020-10-27:
@Ysurac is there an easy way to see the changelog from 55.3 to 56?  I couldn't find it poking around GIT.  Trying to determine if it's worth rebuilding on my unsupported platform.  :)

Thanks,
Rich

Ysurac, 2020-10-27:
On https://www.openmptcprouter.com/ also on pinned 0.56 issue, else complete changes are in git commit history.

closed 2021-01-31 by github-actions[bot]