Ysurac/openmptcprouter #1311: Add new Interface

bobbykrin, 2020-10-23:
## Expected Behavior

## Current Behavior

## Specifications

  - OpenMPTCProuter version: 
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter platform: 

**Hi all
How can i add a new Interface as i am already using 2 4g USB dongle and one Wifi.I want to add 3rd or 4th USB dongle.Is there any way ?**

Ysurac, 2020-10-23:
You can add one in Network->Interfaces and click on "Add new interface"

bobbykrin, 2020-10-23:
I tried to do as I add others but no success.I am not able to add more then 2 USB interfaces.

Ysurac, 2020-10-23:
What is the error you have ?

JeedHome44, 2020-10-29:
if I understand, to add a wifi interface like the connection sharing of a smartphone, so i need a wifi dongle on OMR by smartphone?

bobbykrin, 2020-11-02:
> What is the error you have ?

when i try to add more then 2 USB Dongles the power of all dongles blinked and all of them restart constantly.

Ysurac, 2020-11-02:
This looks like a power problem. Try to use a powered USB hub.

bobbykrin, 2020-11-11:
Hi
Thanks for your guide.I use powered USB hub and its working great.4 Usb hub
and 1 hotspot.Brilliant.
Can I take WiFi output of Bonded network instead of using Ethernet cable
from Raspberry Pi?

closed 2021-02-15 by github-actions[bot]