Ysurac/openmptcprouter #1312: Can I use USB 3g modems not ethernet based

eltsov-da, 2020-10-24:
Dear Sirs, first of all thankyou great work.
But I can't understand if I can use old 3g modems such as Huawei e173. 
As far as I know they work as /dev/usbttyX device.
I was able to connect such modem to internet from openmptcprouter.
Bit I can't understand how to add them as WAN interface.

Ysurac, 2020-10-24:
If they can access to internet, you only have to set enable Multipath on the device in advanced parameters or in Network->MPTCP

eltsov-da, 2020-10-24:
And what about fix ip address? Of course this device got address from provider.

Ysurac, 2020-10-24:
You can do what you want, but I think I don't understand your issue.

Mutanter, 2020-10-24:
My guess is that he wants to use a pure lte modem (ppp) without an isolated LAN interface / static internal IP.

eltsov-da, 2020-10-25:
Thanks to Mutanter. Yes I want to use pure 3G modem (ppp).

Ysurac, 2020-10-25:
Choose PPP protocol and use the device added by the dongle. You can check on OpenWrt doc for more info.

stephanebhiri, 2020-11-13:
I don't if that may help, but we are using 5 Huawei 8372 usb dongle. Each dongle are kept in DHCP mode but with different IP such as : 192.168.11.1,  192.168.12.1, 192.168.13.1 etc.
In OMR they appears as eth1,eth,2,eth3... we choose normal/DHCP for each.

eltsov-da, 2021-02-12:
Still don't understand how to use such modems but let's close this issue.

closed 2021-02-12 by eltsov-da