Ysurac/openmptcprouter #1313: why switched wan3 off?

ascomputer, 2020-10-24:
Sat Oct 24 14:29:05 2020 user.notice post-tracking-post-tracking: wan3 (eth3) switched off because ping from 192.168.8.100 error (80.67.169.12,80.67.169.40)
Sat Oct 24 14:29:06 2020 user.notice post-tracking-post-tracking: Delete default route via 192.168.8.1 dev eth3
Sat Oct 24 14:29:06 2020 user.notice post-tracking-post-tracking: Set server Server1 (95.141.32.80) default route 95.141.32.80 nexthop via 192.168.43.1 dev wlan0 weight 1 nexthop via 192.168.11.1 dev eth4 weight 1

why switched wan3 off?

why do I report this error in the log and all of a sudden the wen 3 doesn't work for me anymore?
can someone help me understand? Thank you

Adorfer, 2020-10-24:
looks like the OMR assumes that WAN3 can not reach the VPS (is offline, or provider has peering issues)

Ysurac, 2020-10-24:
```wan3 (eth3) switched off because ping from 192.168.8.100 error (80.67.169.12,80.67.169.40)```
ICMP ping from wan3, that is eth3 and have ip 192.168.8.100, can't reach test ip 80.67.169.12 and 80.67.169.40. So it seems to be down.

ascomputer, 2020-10-24:
Ok, but I don't know these test ip's can you remove and put different ip's?  if so where can i find them?  thank you for answering me

Ysurac, 2020-10-24:
In Services->OMR-Tracker. But the problem was not the IPs, but the connection.

ascomputer, 2020-10-24:
Yes, I agree with you that it is the connection ... I think that operator in this case has problems pinning to that ip and while on a different ip example google no ... I have to try ... thanks

Ysurac, 2020-10-24:
Because you don't think at all that the connection had a problem ?

ascomputer, 2020-10-24:
yes i think the connection had a problem.

Adorfer, 2020-10-24:
To answer the initial question```

if the WAN3 connection had a problem 
  -> OMRtracker can not ping 'world' through WAN3 
  -> WAN3 is turned off for aggregation
OMR-tracker: keeps trying and if it detects world-connection on WAN3
- > reenable WAN3 for aggregation
```

ascomputer, 2020-10-25:
Hi adorfer, I was doing a live streaming on Facebook and I had 3 4g connections (wan1 wan2 wan3) the wan3 as you saw in the log file is turned off.  Streaming on Facebook went on thanks to the other 2 wans.  but I think ... after an hour the wan3 did not restart even if the dongle was on and the public ip of wan3 was active ... I ask now It could be that the wan3 operator at that moment was having problems pinging that ip  control located in the omr-tracker?

Adorfer, 2020-10-25:
It was turned off since the tracker considered the WAN3 having "not enough" connectivity. 
That's the function of the OMR-tracker. 

nevertheless: if the question is was it a) a false positive or b) a problem in the wan3/dialup-script, not returning online after a (short) problem: we can't say. Try to get more logs (pinging 8.8.8.8 via WAN3) next time it happens.

ascomputer, 2020-10-26:
ok Adorfer, if it happens again I will try to get more info from log. thank you

ascomputer, 2020-10-26:
however I want to say openmtptcp is currently with this version 0.55.3 r0 is very stable for what I do ... if we consider that when one connection stops and the others continue to work and a live stream does not drop. so honestly I have to say THANK YOU to those who designed all this ... then the details of perfecting it can be obtained slowly in the future. Thanks again to the designers.

closed 2021-01-30 by github-actions[bot]