Ysurac/openmptcprouter #1314: Use wiregard instead of openvpn for mptcp over vpn

brazzo97, 2020-10-24:
i found out using secondary modem with wireguard that using wireguard instead of openvpn helps a lot the reability of the connection , connection is more stable and vpn tunnel never goes down, this is needed by me because all the connection i have are 4g with high latency. if there is a way to do it with current version , or maybe could it be implemented ? Thanks

Ysurac, 2020-10-24:
This need to bind to an interface, it's not easy with wireguard.

brazzo97, 2020-10-24:
ok so cant be done ?

Ysurac, 2020-10-24:
No idea. By default wireguard can't be binded to an interface, I don't know if there is another way to do that. Maybe...

Adorfer, 2020-10-24:
this is a duplicate of https://github.com/Ysurac/openmptcprouter/issues/1252

Malaga82, 2020-10-25:
it will be good, not to improve performance but for not reduce performance due to vpn ( with a vpn ping and bandwitdth are affected, since openvpn is not that light ).

closed 2021-01-29 by github-actions[bot]