Ysurac/openmptcprouter #1315: Adding custom FFmpeg packet including librtmp module

Mutanter, 2020-10-24:
**Is your feature request related to a problem? Please describe.**

My goal is to merge various RTMP inputs and use ffmpeg as mixer.
The two currently available FFmpeg versions in OMR are not compiled with the appropriate rtmp module (--enable-librtmp) and therefore not able to process RTMP inputs as expected. (Operation not permitted)

**Describe the solution you'd like**

There is a [prebuilt](https://github.com/openwrt/packages/pull/9475) FFmpeg openwrt config that contains the appropriate librtmp module.
May it be possible to include the package in the standard repository of OMR?

Thanks in advanced!

Ysurac, 2020-10-26:
I'm not sure this is really needed on a router.
This may be added, but I think I will add a new feeds for packets not directly related to OpenWrt.

Mutanter, 2020-10-27:
Did you already started this mentioned and optional packet feed as i can't find the nginx nor the ffmpeg package in the current 56r0 anymore? Could also relate to the following error I get when retrieving the packet update:

Collected errors:
 * opkg_download: Failed to download http://packages.openmptcprouter.com/v0.56/aarch64_cortex-a72/luci/Packages.gz, wget returned 8.
 * opkg_download: Failed to download http://packages.openmptcprouter.com/v0.56/aarch64_cortex-a72/packages/Packages.gz, wget returned 8.
 * opkg_download: Failed to download http://packages.openmptcprouter.com/v0.56/aarch64_cortex-a72/base/Packages.gz, wget returned 8.
 * opkg_download: Failed to download http://packages.openmptcprouter.com/v0.56/aarch64_cortex-a72/routing/Packages.gz, wget returned 8.
 * opkg_download: Failed to download http://packages.openmptcprouter.com/v0.56/aarch64_cortex-a72/telephony/Packages.gz, wget returned 8.

Apart from the missing packages I rly like the 56.r0 so far!

Ysurac, 2020-10-27:
I didn't yet made the optional feed, I will made some doc for new 0.56 features and I will work on this.
I fixed error on package update.

closed 2021-01-31 by github-actions[bot]