Ysurac/openmptcprouter #1317: Server side Block RPi

videoPi4, 2020-10-25:
How can I block the RPi to access the server?

I'm using **server/debian10-x86_64.sh** script on the server-side.

Thanks!

Ysurac, 2020-10-25:
If it's the RPI with OpenMPTCProuter on it then why are you using OpenMPTCProuter ?
Else read https://github.com/Ysurac/openmptcprouter/wiki/Block-client-Internet-access

Adorfer, 2020-10-25:
i am not sure what "access" means. 
But if you do not want the OMR to use the VPS: 
on the vps modifiy the keys in /etc/openmptcprouter-vps-admin/omr-admin-config.json and reboot the server.

videoPi4, 2020-10-25:
@Ysurac Here's what I want:

I've installed **server/debian10-x86_64.sh** script on the server and it provided me some configuration that I've to add in RPI.

I used the provided config and added it in multiple RPI (R1 & R2).

Now I want to block all the requests from the R1 (RPI) but not from R2.

Ysurac, 2020-10-25:
Remove key on RPI R1, no real choices.

Adorfer, 2020-10-25:
since this not an issue linked to the architecture of the OMR (as far as i understand): Can the issue be renamed to to something like "blocking phased out OMR on the VPS"?

videoPi4, 2020-10-25:
Thank you for your support.

Can you please help me out with a few more questions as I still have some doubts?
I want to block it from the server-side (VPS), as I can't access the RPI.

Ysurac, 2020-10-25:
It's not possible. The RPI have the key, it can access the VPS.
You can reinstall the VPS from an initial Debian install.

Adorfer, 2020-10-25:
> You can reinstall the VPS from an initial Debian install.

there is no easy way to initiate the VPS to generate a new key? (or to change in manually in some config file(s)?)

Ysurac, 2020-10-26:
By using UPDATE=no parameter of VPS install script but this also reset all OpenMPTCProuter VPS configuration.

closed 2021-01-30 by github-actions[bot]