Ysurac/openmptcprouter #1318: Cloud.it to add to supported VPS providers

Corrado56, 2020-10-25:
Hi Ysurac,

I think we could add cloud.it (owned by aruba) to the list of supported cloud VPS providers.
it's one of the italy biggest providers with datacenters in Italy, Czech Republic, Germany, France, UK and Poland. You can select the location when You create the VPS.
Debian 10: The "small" sizing is enough to run Openmptcp. I found that the 2 cpu offer (Large size) is more responsive. 
This is the link to the shared VPS offer page: https://www.cloud.it/vps/vps-hosting.aspx

Malaga82, 2020-10-25:
aruba cloud is not that geart "ping side".
i tried it and often had over 35ms ping, while with other i have 10 ( but way less bandwidth ).
Other than that, i used the smallest tier and worked out of the box.

Ysurac, 2020-10-26:
Added

closed 2020-10-26 by Ysurac